openwrt/packages #24186: rust: musl libc - cannot compile rust for x86_64

vortexilation, 2024-05-18:
Maintainer: @lu-zero
Environment: x86_64, musl libc

Description:

Getting the following when trying to compile rust for x86_64 (musl libc)
```
configure: processing command line
configure: 
configure: build.configure-args := ['--build=x86_64-unknown-linux-gnu', '--target ...
configure: build.build          := x86_64-unknown-linux-gnu
configure: build.target         := ['x86_64-unknown-linux-musl', 'x86_64-unknown- ...
configure: build.host           := ['x86_64-unknown-linux-gnu']
configure: install.prefix       := /home/user/works/openwrt/staging_dir/target ...
configure: install.bindir       := /home/user/works/openwrt/staging_dir/target ...
configure: install.libdir       := /home/user/works/openwrt/staging_dir/target ...
configure: install.sysconfdir   := /home/user/works/openwrt/staging_dir/target ...
configure: install.datadir      := /home/user/works/openwrt/staging_dir/target ...
configure: install.mandir       := /home/user/works/openwrt/staging_dir/target ...
configure: dist.compression-formats := ['gz']
configure: build.sanitizers     := False
configure: rust.channel         := stable
configure: build.cargo-native-static := True
configure: build.bootstrap-cache-path := /home/user/works/openwrt/dl/rustc
configure: llvm.download-ci-llvm := True
configure: target.x86_64-unknown-linux-musl.ar := x86_64-openwrt-linux-musl-gcc-a ...
configure: target.x86_64-unknown-linux-musl.cc := x86_64-openwrt-linux-musl-gcc
configure: target.x86_64-unknown-linux-musl.cxx := x86_64-openwrt-linux-musl-g++
configure: target.x86_64-unknown-linux-musl.linker := x86_64-openwrt-linux-musl-g ...
configure: target.x86_64-unknown-linux-musl.ranlib := x86_64-openwrt-linux-musl-g ...
configure: target.x86_64-unknown-linux-musl.crt-static := False
configure: target.x86_64-unknown-linux-musl.musl-root := /home/user/works/open ...
configure: profile              := dist
configure: 
configure: writing `config.toml` in current directory
configure: 
configure: run `python /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/x.py --help`
touch /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/.configured
CARGO_HOME=/home/user/works/openwrt/dl/cargo TARGET_CFLAGS="-march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/user/works/openwrt/build_dir/target-x86_64_musl_custom/rust-1.78.0=rust-1.78.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" python3 /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/x.py --build-dir /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/build --config /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
extracting /home/user/works/openwrt/dl/rustc/2024-03-21/rust-std-1.77.0-x86_64-unknown-linux-gnu.tar.xz
extracting /home/user/works/openwrt/dl/rustc/2024-03-21/rustc-1.77.0-x86_64-unknown-linux-gnu.tar.xz
extracting /home/user/works/openwrt/dl/rustc/2024-03-21/cargo-1.77.0-x86_64-unknown-linux-gnu.tar.xz
Building bootstrap
   Compiling proc-macro2 v1.0.76
   Compiling cfg-if v1.0.0
   Compiling unicode-ident v1.0.12
   Compiling version_check v0.9.4
   Compiling typenum v1.17.0
   Compiling memchr v2.7.1
   Compiling libc v0.2.151
   Compiling rustix v0.38.28
   Compiling serde v1.0.195
   Compiling cc v1.0.73
   Compiling regex-syntax v0.8.2
   Compiling crossbeam-utils v0.8.18
   Compiling bitflags v2.4.1
   Compiling linux-raw-sys v0.4.12
   Compiling pkg-config v0.3.28
   Compiling anstyle v1.0.4
   Compiling clap_lex v0.6.0
   Compiling heck v0.4.1
error: linker `x86_64-openwrt-linux-gnu-gcc` not found
  |
  = note: No such file or directory (os error 2)

   Compiling clap_builder v4.4.12
error: could not compile `proc-macro2` (build script) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: could not compile `crossbeam-utils` (build script) due to 1 previous error
error: could not compile `rustix` (build script) due to 1 previous error
error: could not compile `libc` (build script) due to 1 previous error
error: could not compile `serde` (build script) due to 1 previous error
error: could not compile `typenum` (build script) due to 1 previous error
failed to run: /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/build/x86_64-unknown-linux-gnu/stage0/bin/cargo build --manifest-path /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/src/bootstrap/Cargo.toml
Build completed unsuccessfully in 0:00:13
make[2]: *** [Makefile:109: /home/user/works/openwrt/build_dir/target-x86_64_musl_custom/host/rustc-1.78.0-src/.built] Error 1
make[2]: Leaving directory '/home/user/works/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#23.99#21.04#29.89
    ERROR: package/feeds/packages/rust [host] failed to build.
make[1]: *** [package/Makefile:127: package/feeds/packages/rust/host/compile] Error 1
make[1]: Leaving directory '/home/user/works/openwrt'
make: *** [/home/user/works/openwrt/include/toplevel.mk:233: package/rust/host/compile] Error 2
```

vortexilation, 2024-05-18:
Possible fixes, after applied the following in rust-values.mk (line 14) : 
```
CARGO_HOME:=$(STAGING_DIR)/host/cargo
```
Compilation going again.

lu-zero, 2024-05-19:
I wonder when that variable went away...

1715173329, 2024-05-19:
> target-x86_64_musl_custom

OpenWrt does not have such "custom" target, so I think this is the main cause.

> I wonder when that variable went away...

FYI 853c9c9e8625c54a2c3214b2ef770ffc76bd3495

lu-zero, 2024-05-19:
That should still work, the cargo home should be empty when you start...

vortexilation, 2024-05-19:
@1715173329 
It's just "CONFIG_BUILD_SUFFIX" in OpenWrt config

With "CARGO_HOME:=$(DL_DIR)/cargo" above error happened.

1715173329, 2024-05-21:
> could it be because it was "$(DL_DIR)/cargo" previously used for "glibc" and now used for "musl"?

That would make sense. The log above showed rust failed to find glibc toolchain.

vortexilation, 2024-05-25:
So the temporary solution would be deleting $(DL_DIR)/cargo if you switch from glibc to musl or vice versa.

lu-zero, 2024-05-25:
Looks like so