openwrt/packages #24203: ocserv 1.3 crashes and does not start

buboleck, 2024-05-22:
Maintainer: @nmav
Environment: (MVEBU, WRT1900ACSv1, OpenWrt 23.05.02, 23.05.03 etc)

Description:

Wed May 22 17:17:17 2024 daemon.err uhttpd[2071]: error connecting to ocserv socket '/var/run/occtl.socket': No such file or directory
Wed May 22 17:17:19 2024 daemon.err uhttpd[2071]: error connecting to ocserv socket '/var/run/occtl.socket': No such file or directory
Wed May 22 17:17:21 2024 daemon.err uhttpd[2071]: error connecting to ocserv socket '/var/run/occtl.socket': No such file or directory
Wed May 22 17:17:35 2024 daemon.err uhttpd[2071]: error connecting to ocserv socket '/var/run/occtl.socket': No such file or directory
Wed May 22 17:17:42 2024 daemon.err uhttpd[2071]: error connecting to ocserv socket '/var/run/occtl.socket': No such file or directory
Wed May 22 17:18:00 2024 daemon.info procd: Instance ocserv::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

The crash is reproducible with the following combinations:

OpenWrt 23.05.02/Ocserv 1.3/luci-app-ocserv git-20.110.55046-74da73b
OpenWrt 23.05.03/Ocserv 1.3/luci-app-ocserv git-20.110.55046-74da73b
OpenWrt SNAPSHOT r26300 (Linux 6.6.30)/Ocserv 1.3/luci-app-ocserv  23.267.29502~3e7fe9f

Working combinations for me are OpenWrt 23.05.02/Ocserv 1.1.x/1.2.x/luci-app-ocserv git-20.110.55046-74da73b
OpenWrt SNAPSHOT r26300 (Linux 6.6.30)/Ocserv 1.1.6-2/luci-app-ocserv  23.267.29502~3e7fe9f

The 1.1.6-2 I got from older build as 1.2.x was replaced by 1.3. I also tried on fresh install without transferring configs but Ocserv 1.3 still crashes.

OpenWrt version information from system /etc/openwrt_release

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r26300-da0cd9d764'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r26300-da0cd9d764'
DISTRIB_TAINTS=''

Please let me know what other info I can provide.

nmav, 2024-05-26:
Can you try this? https://github.com/openwrt/packages/pull/24231

buboleck, 2024-05-26:
Yes, I can try it, but how can I get the package?

buboleck, 2024-05-27:
Hello, installed ocserv 1.3.0-r1 and edited the init script by hand as suggested, I can confirm this fixed the startup issue.

PR #24231 (2024-05-27): ocserv: use a more neutral character for sed
This resolves a startup issue.

Resolves: #24203

Maintainer: me
Compile tested: mvebu/cortexa9
Run tested: mvebu/cortexa9
commit bd1464db (2024-05-27): ocserv: use a more neutral character for sed

This resolves a startup issue.

Resolves: #24203
commit 57d3da50 (2024-05-27): ocserv: use a more neutral character for sed

This resolves a startup issue.

Resolves: #24203