openwrt/packages #24224: ax25-tools ax25-apps

parabyte, 2024-05-25:
Can someone please make these packages work, they are very important for packet radio and aprs.

https://github.com/ve7fet/linuxax25

official repo site does not seem to work for me so i gave the above url

brada4, 2024-05-25:
That someone should be you.

parabyte, 2024-05-25:
starting to think that,  only problem is i am not familiar with the openwrt build system yet, 

hence why i thought i would ask

esaaprillia, 2024-06-22:
@parabyte 

I tried to build it

```
root@OpenWrt:/usr/lib# ldd libax25.so.1.0.1
        ldd (0x7fb7d99000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fb7d47000)
        libc.so => ldd (0x7fb7d99000)
root@OpenWrt:/usr/lib#
root@OpenWrt:/usr/lib#
root@OpenWrt:/usr/lib#
root@OpenWrt:/usr/lib#
root@OpenWrt:/usr/lib# ldd libax25io.so.1.0.0
        ldd (0x7f9ff9a000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7f9ff47000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f9ff16000)
        libc.so => ldd (0x7f9ff9a000)
root@OpenWrt:/usr/lib# ldd libax25.so
        ldd (0x7fba890000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fba83e000)
        libc.so => ldd (0x7fba890000)
root@OpenWrt:/usr/lib#
root@OpenWrt:/usr/lib#
root@OpenWrt:/usr/lib# ldd libax25io.so
        ldd (0x7fb920b000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7fb91b8000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fb9187000)
        libc.so => ldd (0x7fb920b000)
root@OpenWrt:/usr/lib# cd /usr/bin
root@OpenWrt:/usr/bin# ldd call
        /lib/ld-musl-aarch64.so.1 (0x7f872cb000)
        libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x7f8725a000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7f87239000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f87208000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f872cb000)
root@OpenWrt:/usr/bin# ldd listen
        /lib/ld-musl-aarch64.so.1 (0x7fb0d33000)
        libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x7fb0cc2000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7fb0ca1000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fb0c70000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7fb0d33000)
root@OpenWrt:/usr/bin# cd /usr/sbin
root@OpenWrt:/usr/sbin# ldd ax25ipd
        /lib/ld-musl-aarch64.so.1 (0x7f9c379000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7f9c358000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f9c327000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f9c379000)
root@OpenWrt:/usr/sbin# ldd ax25mond
        /lib/ld-musl-aarch64.so.1 (0x7fad8b0000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7fad88f000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fad85e000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7fad8b0000)
root@OpenWrt:/usr/sbin# ldd ax25rtctl
        /lib/ld-musl-aarch64.so.1 (0x7f8c0a9000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f8c078000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f8c0a9000)
root@OpenWrt:/usr/sbin# ldd ax25rtd
        /lib/ld-musl-aarch64.so.1 (0x7f896c6000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7f896a5000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f89674000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f896c6000)
root@OpenWrt:/usr/sbin# cd /usr/bin
root@OpenWrt:/usr/bin# ldd bget
        /lib/ld-musl-aarch64.so.1 (0x7fa5d2a000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fa5cf9000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7fa5d2a000)
root@OpenWrt:/usr/bin# ldd mheard
        /lib/ld-musl-aarch64.so.1 (0x7f86e5b000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7f86e3a000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f86e09000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f86e5b000)
root@OpenWrt:/usr/bin# ldd sethdlc
        /lib/ld-musl-aarch64.so.1 (0x7fb3fc7000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fb3f96000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7fb3fc7000)
root@OpenWrt:/usr/bin# ldd smmixer
        /lib/ld-musl-aarch64.so.1 (0x7fb5789000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fb5758000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7fb5789000)
root@OpenWrt:/usr/bin# cd /usr/sbin
root@OpenWrt:/usr/sbin# ldd ttylinkd
        /lib/ld-musl-aarch64.so.1 (0x7f91f8c000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f91f5b000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f91f8c000)
root@OpenWrt:/usr/sbin# ldd mkiss
        /lib/ld-musl-aarch64.so.1 (0x7fb1f60000)
        libax25.so.1 => /usr/lib/libax25.so.1 (0x7fb1f3f000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fb1f0e000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7fb1f60000)
root@OpenWrt:/usr/sbin#
```

brada4, 2024-06-22:
Good work ;-) Pull request and it is golden (category net)

parabyte, 2024-06-22:
thanks esaaprillia, i was using debian chroot as my solution, but this is waaay better thanks for taking your time to figure this out, i was on it for days and just around in circles as i am not a openwrt dev,

esaaprillia, 2024-06-24:
@brada4 thanks

@parabyte 

maybe you want to try it

I built using openwrt-23.05.3 aarch64-generic

[[xcxx](https://github.com/esaaprillia/RADIO/raw/main/RADIO.rar)](url)

parabyte, 2024-07-08:
im having zero issues, and building on various different platforms including x86-64,

thanks for making this a reality, have been reading up on the build system hopefully can return the favour one day :)

closed 2024-07-08 by parabyte