openwrt/packages #24225: travelmate: allows 1 radio with both STA and AP mode?

Sawtaytoes, 2024-05-25:
# Issue template
Maintainer: @dibdot
Environment: OpenWrt 23.05.3 - Raspberry Pi 4 Model B

From the docs:
> A logical **combination of AP+STA mode on one physical radio** allows most of OpenWrt supported router devices to connect to a wireless hotspot/station (STA) and provide a wireless access point (AP) from that hotspot at the same time.

Description:
I setup `travelmate` and have another interface configured as an "Access Point", but when I go to use both an AP and STA on a single radio, it only connects as an STA, not also an AP.

[image]

It shows the Access Point one as "disabled" even though it shows enabled in its settings:

[image]

[image]

[image]

[image]

brada4, 2024-05-26:
Set country code, otherwise you catch such confusing displays while dfs clears for transmission.

Sawtaytoes, 2024-05-26:
I changed the country code to US, but nothing else changed. Are you sure it's supposed to be able to do all this on a single radio?

brada4, 2024-05-28:
No idea chan 11 80mhz looks nonsensical

brada4, 2024-05-28:
Normally passive roaming scan is possible, but AP needs to roam along for  real connection.

dibdot, 2024-05-29:
@Sawtaytoes This can work very well, for example with a GL-MT300N-V2 (ATH79 arch). Broadcom chipsets will most probably fail (from my experiences with an older raspberry PI) ...

brada4, 2024-05-29:
You need 2 APs in different bands to elegantly roam out of sinister channel jump when it happens.

dibdot, 2025-10-19:
nothing todo - close this outdated ticket.