openwrt/packages #24229: curl: gcc 14 compatibilities - Failed to find size of size_t

vortexilation, 2024-05-26:
Maintainer: 
Environment: gcc 14.1.0, x86_64 musl, Debian SID x86_64 glibc host/build.

Description:

When trying to compile for/targeting OpenWrt x86_64 musl, I am getting the following : 
```
checking for struct timeval... no
checking size of size_t... configure: error: Failed to find size of size_t
```

```
configure:34850: ccache x86_64-openwrt-linux-musl-gcc -c -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fhonour-copts -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include -fmacro-prefix-map=/home/user/works/openwrt/build_dir/target-x86_64_musl/curl-8.7.1=curl-8.7.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -D_FORTIFY_SOURCE=0 -Werror-implicit-function-declaration -Wno-system-headers -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/usr/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include conftest.c >&5
<command-line>: warning: "_FORTIFY_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
conftest.c:71:16: error: duplicate 'unsigned'
   71 | #define size_t unsigned int
      |                ^~~~~~~~
In file included from /usr/include/bits/endian.h:35,
                 from /usr/include/endian.h:24,
                 from /usr/include/sys/types.h:176,
                 from conftest.c:76:
/usr/include/bits/endianness.h:9:9: warning: "__BYTE_ORDER" redefined
    9 | #define __BYTE_ORDER __LITTLE_ENDIAN
      |         ^~~~~~~~~~~~
In file included from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/stddef.h:19,
                 from /usr/include/sys/types.h:144:
/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/bits/alltypes.h:5:9: note: this is the location of the previous definition
    5 | #define __BYTE_ORDER 1234
      |         ^~~~~~~~~~~~
In file included from /usr/include/sys/select.h:37,
                 from ../../../staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify/sys/select.h:22,
                 from /usr/include/sys/types.h:179:
/usr/include/bits/types/struct_timeval.h:11:3: error: unknown type name '__time64_t'; did you mean '__timer_t'?
   11 |   __time64_t tv_sec;            /* Seconds.  */
      |   ^~~~~~~~~~
      |   __timer_t
In file included from /usr/include/sys/select.h:39:
/usr/include/bits/types/struct_timespec.h:14:3: error: unknown type name '__time64_t'; did you mean '__timer_t'?
   14 |   __time64_t tv_sec;            /* Seconds.  */
      |   ^~~~~~~~~~
      |   __timer_t
conftest.c: In function 'main':
conftest.c:83:8: error: duplicate case value
   83 |        case (sizeof(size_t) == 1):;
      |        ^~~~
conftest.c:82:8: note: previously used here
   82 |        case 0:
      |        ^~~~
configure:34850: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "curl"
| #define PACKAGE_TARNAME "curl"
| #define PACKAGE_VERSION "-"
| #define PACKAGE_STRING "curl -"
| #define PACKAGE_BUGREPORT "a suitable curl mailing list: https://curl.se/mail/"
| #define PACKAGE_URL ""
| #define HAVE_STDIO_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_UNISTD_H 1
| #define STDC_HEADERS 1
| #define HAVE_STDATOMIC_H 1
| #define HAVE_ATOMIC 1
| #define PACKAGE "curl"
| #define VERSION "-"
| #define OS "x86_64-openwrt-linux-gnu"
| #define HAVE_DLFCN_H 1
| #define LT_OBJDIR ".libs/"
| #define HAVE_LDAP_SSL 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_TIME_H 1
| #define HAVE_CLOCK_GETTIME_MONOTONIC 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_TIME_H 1
| #define HAVE_CLOCK_GETTIME_MONOTONIC_RAW 1
| #define HAVE_LIBZ 1
| #define HAVE_LIBZSTD 1
| #define HAVE_ZSTD_H 1
| #define HAVE_ZSTD 1
| #define HAVE_LBER_H 1
| #define HAVE_LDAP_H 1
| #define HAVE_LDAP_URL_PARSE 1
| #define HAVE_LDAP_INIT_FD 1
| #define USE_OPENLDAP 1
| #define ENABLE_IPV6 1
| #define HAVE_SOCKADDR_IN6_SIN6_SCOPE_ID 1
| #define HAVE_LIBSSL 1
| #define HAVE_OPENSSL_X509_H 1
| #define USE_OPENSSL 1
| #define HAVE_OPENSSL_RSA_H 1
| #define USE_OPENSSL 1
| #define HAVE_OPENSSL_CRYPTO_H 1
| #define USE_OPENSSL 1
| #define HAVE_OPENSSL_PEM_H 1
| #define USE_OPENSSL 1
| #define HAVE_OPENSSL_SSL_H 1
| #define USE_OPENSSL 1
| #define HAVE_OPENSSL_ERR_H 1
| #define USE_OPENSSL 1
| #define HAVE_OPENSSL3 1
| #define HAVE_SSL_SET_QUIC_USE_LEGACY_CODEPOINT 1
| #define HAVE_OPENSSL_SRP 1
| #define CURL_CA_BUNDLE "/etc/ssl/certs/ca-certificates.crt"
| #define HAVE_LIBSSH2 1
| #define USE_LIBSSH2 1
| #define HAVE_IDN2_H 1
| #define HAVE_LIBIDN2 1
| #define HAVE_NGHTTP2_NGHTTP2_H 1
| #define USE_NGHTTP2 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_TIME_H 1
| #define HAVE_UNISTD_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_SYS_STAT_H 1
| #define size_t unsigned int
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_TIME_H 1
| /* end confdefs.h.  */
| 
| #include <sys/types.h>
| 
| 
| int main (void)
| {
| switch(0) {
|        case 0:
|        case (sizeof(size_t) == 1):;
|      }
| 
|  ;
|  return 0;
| }
configure:34864: error: Failed to find size of size_t
```

Full config.log : 
[config.log](https://github.com/openwrt/packages/files/15447073/config.log)

brada4, 2024-05-26:
Disable ccache?

vortexilation, 2024-05-26:
The same error with disabled ccache

brada4, 2024-05-26:
Something is wrong with debian gcc-14 packaging, just use current gcc(-13) to compile cross-compilers.

vortexilation, 2024-05-26:
I think you have misread my title and yes my description also wasn't clear enough it's regarding compiling curl with gcc 14, the OpenWrt gcc 14.1.0 musl toolchain are already compiled using host/debian gcc 13 compiler, the problem is when compiling curl with the OpenWrt gcc 14.1.0 musl toolchain.

brada4, 2024-05-26:
use stable debian?

vortexilation, 2024-05-27:
I don't see the correlation switching to stable variant as I have using SID since a long time for compiling OpenWrt.

The OpenWrt gcc 14 musl toolchain is already compiled, i don't see much different for switching. And this gcc 14 toolchain is already compiled 660 OpenWrt packages. 

Have you test compiling curl yourself recently with gcc 14?

vortexilation, 2024-05-28:
diffconfig : 
[diffconfig.txt](https://github.com/openwrt/packages/files/15463879/diffconfig.txt)

@neheb 
Can you please confirm, you don't have the same problem as this report?.

brada4, 2024-05-28:
Absolutely mot, cpu is zen though...

brada4, 2024-05-28:
Your diffconfig says you are crosscompiling for non-x86 tsrget with obvious march=native x86 flags i.e. non functional compiler. How do you think package snapshots are brought to life?

vortexilation, 2024-05-29:
What do you mean?, I am targeting x86, the custom cflags?, tigerlake is x86_64 target, it's only code [gcc generation optimization](https://wiki.gentoo.org/wiki/GCC_optimization), the same config is working fine for glibc, the OpenWrt image is bootable and working flawlessly. This is not a new config, it's working one since a long time for me.

The compiler is functional, it's already compiled 660 OpenWrt packages before stopped at curl.

Also if you take a look at config.log above, this is mostly about header problem.

brada4, 2024-05-29:
You>CONFIG_EXTRA_OPTIMIZATION="-fno-caller-saves -fno-plt -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize"
Gentoo> Clearly it is not safe to compile whole system with -O3

vortexilation, 2024-05-29:
With either -O2 or -O3 doesn't make such compilation errors as above, I am reporting issue about first post not about gcc optimization level. Also optimization level problems happened at only runtime not compilation time.

brada4, 2024-05-29:
If you compile compiler with -O3 it cannot compute sizeof(size_t) - it is just broken by unsafe optimisations.
Build compiler with normal default parameters so that compilee/configure/cmake does not bail out at random.

vortexilation, 2024-05-29:
There is no differences, I have just tried with only "-mtune=tigerlake -march=tigerlake -O2 -pipe" cflags, compiling the OpenWrt toolchain, kernel, and packages, still stopped at curl as error above.

vortexilation, 2024-05-29:
Also tried switching to debian stable, still the same.

vortexilation, 2024-05-30:
Starting over .config doesn't have this problem, I have tried without any cflags, etc but seems cannot pinpoint which config variable does these.

But I am able to fixes it : after diving through the config.log , found out that -I/usr/include is being added after libngtcp2 check being run : 
```
configure:31302: checking for libngtcp2 options with pkg-config
configure:31316: result: found
configure:31330: -l is -lngtcp2
configure:31339: -I is -I/usr/include
configure:31349: -L is -L/usr/lib
configure:31359: checking for ngtcp2_conn_client_new_versioned in -lngtcp2
configure:31382: ccache x86_64-openwrt-linux-musl-gcc -o conftest -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -fhonour-copts -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include -fmacro-prefix-map=/home/user/works/openwrt/build_dir/target-x86_64_musl/curl-8.8.0=curl-8.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -DHAVE_ULONG -Werror-implicit-function-declaration -Wno-system-headers -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/usr/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/usr/include -L/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/usr/lib -L/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/lib -fuse-ld=bfd -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/lib -Wl,-rpath-link=/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/lib -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/lib -Wl,-rpath-link=/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/lib -znow -zrelro -Wl,--gc-sections  -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/usr/lib conftest.c -lngtcp2  -lngtcp2 -lnghttp2 -lidn2 -lssh2 -lssh2 -lssl -lcrypto -lssl -lcrypto -lldap -llber -lzstd -lzstd -lz  >&5
configure:31382: $? = 0
configure:31392: result: yes
configure:31399: checking for ngtcp2/ngtcp2.h
configure:31399: ccache x86_64-openwrt-linux-musl-gcc -c -march=tigerlake -mtune=tigerlake -O3 -funsafe-math-optimizations -ffp-contract=fast -ftree-vectorize -fno-caller-saves -fno-plt -fhonour-copts -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include -fmacro-prefix-map=/home/user/works/openwrt/build_dir/target-x86_64_musl/curl-8.8.0=curl-8.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -DHAVE_ULONG -Werror-implicit-function-declaration -Wno-system-headers -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/usr/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include -I/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/user/works/openwrt/staging_dir/target-x86_64_musl/usr/include -I/usr/include conftest.c >&5
In file included from ../../../staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify/stdio.h:22,
                 from conftest.c:69:
/usr/include/stdio.h:52:9: error: unknown type name '__gnuc_va_list'
   52 | typedef __gnuc_va_list va_list;
      |         ^~~~~~~~~~~~~~
/usr/include/stdio.h:52:24: error: conflicting types for 'va_list'; have 'int'
   52 | typedef __gnuc_va_list va_list;
      |                        ^~~~~~~
```

so the solution is to remove -I/usr/include from libngtcp2.pc (pkgconfig)
```
$(SED) 's| -I$$$${includedir}||' $(1)/usr/lib/pkgconfig/libngtcp2.pc
```

full packages/libs/ngtcp2/Makefile : 
```
include $(TOPDIR)/rules.mk

PKG_NAME:=ngtcp2
[... cut ...]
discussed in IETF QUICWG for its standardization.
endef

CMAKE_OPTIONS += -DENABLE_LIB_ONLY=ON

define Build/InstallDev
	$(INSTALL_DIR) $(1)/usr/lib/pkgconfig
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/pkgconfig/libngtcp2.pc $(1)/usr/lib/pkgconfig/libngtcp2.pc
	$(SED) 's| -I$$$${includedir}||' $(1)/usr/lib/pkgconfig/libngtcp2.pc
endef

define Package/libngtcp2/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(INSTALL_DATA) $(PKG_INSTALL_DIR)/usr/lib/libngtcp2*.so* $(1)/usr/lib
endef

$(eval $(call BuildPackage,libngtcp2))
```

neheb, 2024-05-31:
https://github.com/openwrt/packages/commit/2fd1e9d6e0ad7f6d0b3d150229750fb0a03c3907

closed 2024-05-31 by neheb