openwrt/packages #24230: uhttpd: Invalid Lua prefix error when restarting uhttpd

vochong, 2024-05-26:
# Issue template (remove lines from top till here)

Maintainer: @nbd168
Environment: 

{
	"kernel": "6.6.30",
	"hostname": "Home",
	"system": "ARMv7 Processor rev 0 (v7l)",
	"model": "Netgear Nighthawk X4S R7800",
	"board_name": "netgear,r7800",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r26434-0a861a0c0f",
		"target": "ipq806x/generic",
		"description": "OpenWrt SNAPSHOT r26434-0a861a0c0f"
	}
}

Description:

# /etc/init.d/uhttpd stop
# /etc/init.d/uhttpd start
Skipping invalid Lua prefix "/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua"

brada4, 2024-05-27:
It is a warning, lua wsgi is not installed and is skipped.

vochong, 2024-05-28:
Thank you @brada4  I will close it.

closed 2024-05-28 by vochong

NeroBlackstone, 2025-01-28:
Same problem..

Which package does “lua wsgi” refer to?

NeroBlackstone, 2025-01-28:
be sure `uhttpd` and `uhttpd-mod-lua` at same version
```
root@OpenWrt:/tmp# opkg list-installed | grep uhttpd
uhttpd - 2021-03-21-15346de8-2
uhttpd-mod-lua - 2021-03-21-15346de8-2
```