openwrt/packages #24264: openssh: missing from openwrt 22.03.4 arm_cortex-a8_vfpv3

kutaycinar, 2024-05-29:
Maintainer: @tripolar 
Environment:
>  OpenWRT Version: 22.03.4
    Architecture: arm_cortex-a8_vfpv3
    Package: openssh (openssh-sftp-server)

### Description:

The OpenSSH packages are missing from the arm_cortex-a8_vfpv3 repository for OpenWRT 22.03.4, which prevents the installation of openssh-sftp-server. To reproduce the issue, navigate to the OpenWRT package repository for 22.03.4 and search for the openssh-sftp-server package; you will find it absent, with fail logs indicating a problem.

#### Package repository:
https://downloads.openwrt.org/releases/22.03.4/packages/arm_cortex-a8_vfpv3/packages

#### Fail logs:
https://downloads.openwrt.org/releases/faillogs-22.03/arm_cortex-a8_vfpv3/packages/

#### Console output:
``` 
Unknown package 'openssh-sftp-server'.
...
Collected errors:
 * opkg_install_cmd: Cannot install package openssh-sftp-server.
make[2]: *** [Makefile:176: package_install] Error 255
make[1]: *** [Makefile:125: _call_image] Error 2
make: *** [Makefile:249: image] Error 2
```

iExad, 2024-06-05:
same for 22.03.6, package openssh-sftp-server is absent in repo. tl-wr842n v5

castillofrancodamian, 2024-06-07:
Also in master.

bowserjklol, 2024-07-07:
Check now - should be resolve resultant of https://github.com/openwrt/packages/pull/24494

kutaycinar, 2024-07-17:
Thanks! Seem's resolved.

closed 2024-07-17 by kutaycinar