openwrt/packages #24268: openssh: build failure on mips_24kc with -fzero-call-used-regs

bowserjklol, 2024-05-30:
Maintainer: @neheb  @SibrenVasse 
Environment: mips_24kc, 22.03

Description: OpenSSH is currently failing to build for mips_24kc on 22.03.  Package is unavailable resulting in, at least for the openssh-client, `opkg_install_cmd: Cannon install package openssh-client`

Fail logs are available [here](https://downloads.openwrt.org/releases/faillogs-22.03/mips_24kc/packages/openssh/)

```
bsd-closefrom.c
bsd-closefrom.c: In function 'closefrom':
bsd-closefrom.c:150:1: sorry, unimplemented: '-fzero-call-used-regs' not supported on this target
  150 | }
      | ^
make[5]: *** [Makefile:106: bsd-closefrom.o] Error 1
```

castillofrancodamian, 2024-06-07:
```
(cd /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-mips_24kc/iperf3 && find . -type f,l -printf "/%P\n" > /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-mips_24kc/iperf3/lib/apk/packages/iperf3.list)
if [ -f /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/apk-mips_24kc/iperf3/conffiles ]; then mv /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/apk-mips_24kc/iperf3/conffiles /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-mips_24kc/iperf3/lib/apk/packages/iperf3.conffiles; fi;
/media/hdd320gb/openwrt/staging_dir/host/bin/fakeroot /media/hdd320gb/openwrt/staging_dir/host/bin/apk mkpkg --info "name:iperf3" --info "version:3.17.1-r1" --info "description:" --info "arch:mips_24kc" --info "license:BSD-3-Clause" --info "origin:feeds/packages/net/iperf3" --info "provides:" --script "post-install:/media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/apk-mips_24kc/iperf3/post-install" --script "pre-deinstall:/media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/apk-mips_24kc/iperf3/pre-deinstall" --info "depends:libc libiperf3" --files "/media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-mips_24kc/iperf3" --output "/media/hdd320gb/openwrt/bin/packages/mips_24kc/packages/iperf3-3.17.1-r1.apk" --sign "/media/hdd320gb/openwrt/private-key.pem"
echo "iperf3" >> /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/iperf3.nossl.install
rm -rf /media/hdd320gb/openwrt/tmp/stage-iperf3
mkdir -p /media/hdd320gb/openwrt/tmp/stage-iperf3/host /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/packages
install -d -m0755 /media/hdd320gb/openwrt/tmp/stage-iperf3/usr/lib
install -d -m0755 /media/hdd320gb/openwrt/tmp/stage-iperf3/usr/include
cp -fpR /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-install/usr/lib/libiperf.* /media/hdd320gb/openwrt/tmp/stage-iperf3/usr/lib/
cp -fpR /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-install/usr/include/* /media/hdd320gb/openwrt/tmp/stage-iperf3/usr/include/
find /media/hdd320gb/openwrt/tmp/stage-iperf3 -name '*.la' | xargs -r rm -f; 
if [ -f /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/packages/iperf3.nossl.list ]; then /media/hdd320gb/openwrt/scripts/clean-package.sh "/media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/packages/iperf3.nossl.list" "/media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl"; fi
if [ -d /media/hdd320gb/openwrt/tmp/stage-iperf3 ]; then (cd /media/hdd320gb/openwrt/tmp/stage-iperf3; find ./ > /media/hdd320gb/openwrt/tmp/stage-iperf3.files);      SHELL= flock /media/hdd320gb/openwrt/tmp/.staging-dir.flock -c ' mv /media/hdd320gb/openwrt/tmp/stage-iperf3.files /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/packages/iperf3.nossl.list && cp -fpR /media/hdd320gb/openwrt/tmp/stage-iperf3/* /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/; '; fi
rm -rf /media/hdd320gb/openwrt/tmp/stage-iperf3
touch /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/stamp/.iperf3.nossl_installed
rm -rf /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3.installed /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3
mkdir -p /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3
install -d -m0755 /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3/usr/lib
cp -fpR /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/ipkg-install/usr/lib/libiperf.so.* /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3/usr/lib
touch /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3.installed
mkdir -p /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/stamp
SHELL= flock /media/hdd320gb/openwrt/tmp/.root-copy.flock -c 'cp -fpR /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/iperf-nossl/iperf-3.17.1/.pkgdir/libiperf3/. /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/'
touch /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/stamp/.libiperf3_installed
echo "libiperf3" >> /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/pkginfo/iperf3.nossl.install
make[3]: Leaving directory '/media/hdd320gb/openwrt/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/nossl/compile#0.82#0.31#1.38
make[3]: Entering directory '/media/hdd320gb/openwrt/feeds/packages/libs/libpam'
make[3]: Leaving directory '/media/hdd320gb/openwrt/feeds/packages/libs/libpam'
time: package/feeds/packages/libpam/compile#0.34#0.15#0.81
make[3]: Entering directory '/media/hdd320gb/openwrt/feeds/packages/net/openssh'
rm -rf /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/ipkg-mips_24kc/openssh-sftp-server
mkdir -p /media/hdd320gb/openwrt/bin/targets/ath79/generic/packages /media/hdd320gb/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/ipkg-mips_24kc/openssh-sftp-server /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/pkginfo
[... cut ...]
    ERROR: package/feeds/packages/openssh failed to build (build variant: without-pam).
make[2]: *** [package/Makefile:173: package/feeds/packages/openssh/compile] Error 1
make[2]: Leaving directory '/media/hdd320gb/openwrt'
make[1]: *** [package/Makefile:167: /media/hdd320gb/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/hdd320gb/openwrt'
make: *** [/media/hdd320gb/openwrt/include/toplevel.mk:233: world] Error 2

real    4m25,904s
user    2m1,994s
sys     0m39,874s

```

neheb, 2024-06-08:
@bowserjklol backported version update. Should fix things.

@castillofrancodamian that's related to apk, not openssh.

bowserjklol, 2024-06-12:
Thanks @neheb.  Saw the commit in the 22.03 branch.  Apologies as I'm not fully aware of the process here - when will the CI/build system/github actions (whatever is used) re-attempt compilation of the openssh package so it becomes available in the 22.03 package feed?  Thanks again!

bowserjklol, 2024-06-20:
Bumping the thread.  Package feed shows compilation attempts again on 12/16/20 Jun and still fails with the same error.  Thanks!

neheb, 2024-06-20:
hrm maybe https://github.com/openssh/openssh-portable/commit/f9193f03db0029fc9c31fbdb5c66a2737446bd8f needs backporting. Can anyone test?

bowserjklol, 2024-06-21:
Using the 22.03 toolchain, reproduced the current error:

```
bsd-closefrom.c: In function 'closefrom':
bsd-closefrom.c:150:1: sorry, unimplemented: '-fzero-call-used-regs' not supported on this target
  150 | }
      | ^
make[4]: *** [Makefile:106: bsd-closefrom.o] Error 1
make[4]: Leaving directory '/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/openbsd-compat'
make[3]: *** [Makefile:199: openbsd-compat/libopenbsd-compat.a] Error 2
make[3]: Leaving directory '/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1'
make[2]: *** [Makefile:270: /root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/.built] Error 2
make[2]: Leaving directory '/root/openwrt/feeds/packages/net/openssh'
time: package/feeds/packages/openssh/without-pam/compile#18.58#11.71#28.31
    ERROR: package/feeds/packages/openssh failed to build (build variant: without-pam).
make[1]: *** [package/Makefile:120: package/feeds/packages/openssh/compile] Error 1

```

Applied patch:

```
root@c92e96655184:~/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1# cat ~/openwrt/901-zero-call.patch 
diff --git m4/openssh.m4 m4/openssh.m4
index 033df501c..176a8d1c9 100644
--- m4/openssh.m4
+++ m4/openssh.m4
@@ -20,7 +20,10 @@ char *f2(char *s, ...) {
 	va_end(args);
 	return strdup(ret);
 }
+int i;
+double d;
 const char *f3(int s) {
+	i = (int)d;
 	return s ? "good" : "gooder";
 }
 int main(int argc, char **argv) {
root@c92e96655184:~/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1# patch m4/openssh.m4 < ~/openwrt/901-zero-call.patch 
patching file m4/openssh.m4
```

Reattempt and error:

```
root@c92e96655184:~/openwrt# make package/openssh/compile -j1 V=s
[... snip ...]
make[2]: Entering directory '/root/openwrt/feeds/packages/net/openssh'
rm -f /root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/.built
touch /root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/.built_check
make -C /root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1 DESTDIR="/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/ipkg-install" STRIP_OPT="" all install
make[3]: Entering directory '/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1'
(cd openbsd-compat && make)
make[4]: Entering directory '/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/openbsd-compat'
mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1=openssh-9.7p1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -pipe -Wno-error=format-truncation -Wall -Wextra -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-parameter -Wno-unused-result -Wimplicit-fallthrough -Wmisleading-indentation -fno-strict-aliasing -ftrapv -fzero-call-used-regs=used -fno-builtin-memset -fstack-protector-strong   -fPIC -I. -I.. -I. -I./.. -I/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/include -I/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include/fortify -I/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include  -D_XOPEN_SOURCE=600 -D_BSD_SOURCE -D_DEFAULT_SOURCE -D_GNU_SOURCE -I/root/openwrt/staging_dir/target-mips_24kc_musl/usr/include  -DHAVE_CONFIG_H -c bsd-closefrom.c
bsd-closefrom.c: In function 'closefrom':
bsd-closefrom.c:150:1: sorry, unimplemented: '-fzero-call-used-regs' not supported on this target
  150 | }
      | ^
make[4]: *** [Makefile:106: bsd-closefrom.o] Error 1
make[4]: Leaving directory '/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/openbsd-compat'
make[3]: *** [Makefile:199: openbsd-compat/libopenbsd-compat.a] Error 2
make[3]: Leaving directory '/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1'
make[2]: *** [Makefile:270: /root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/.built] Error 2
make[2]: Leaving directory '/root/openwrt/feeds/packages/net/openssh'
time: package/feeds/packages/openssh/without-pam/compile#0.17#0.10#0.24
    ERROR: package/feeds/packages/openssh failed to build (build variant: without-pam).
make[1]: *** [package/Makefile:120: package/feeds/packages/openssh/compile] Error 1
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:230: package/openssh/compile] Error 2
```

cat patched file to make sure...

```
root@c92e96655184:~/openwrt# cat build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1/m4/openssh.m4 
dnl OpenSSH-specific autoconf macros
dnl

dnl The test program that is used to try to trigger various compiler
dnl behaviours.
AC_DEFUN([OSSH_COMPILER_FLAG_TEST_PROGRAM],
	[AC_LANG_SOURCE([[
#include <stdlib.h>
#include <stdarg.h>
#include <stdio.h>
#include <string.h>
#include <unistd.h>
/* Trivial function to help test for -fzero-call-used-regs */
int f(int n) {return rand() % n;}
char *f2(char *s, ...) {
	char ret[64];
	va_list args;
	va_start(args, s);
	vsnprintf(ret, sizeof(ret), s, args);
	va_end(args);
	return strdup(ret);
}
int i;
double d;
const char *f3(int s) {
	i = (int)d;
	return s ? "good" : "gooder";
}
int main(int argc, char **argv) {

```

neheb, 2024-06-21:
what does config.log show around -fzero-call-used-regs?

bowserjklol, 2024-06-21:
Sanity checked to make sure all the autoreconf stuff worked in PKG_FIXUP and it _looks_ like the configure script is indeed picking up the patch

```
        { $as_echo "$as_me:${as_lineno-$LINENO}: checking if $CC supports compile flag -fzero-call-used-regs=used and linking succeeds" >&5
$as_echo_n "checking if $CC supports compile flag -fzero-call-used-regs=used and linking succeeds... " >&6; }
        saved_CFLAGS="$CFLAGS"
        CFLAGS="$CFLAGS $WERROR -fzero-call-used-regs=used"
        _define_flag=""
        test "x$_define_flag" = "x" && _define_flag="-fzero-call-used-regs=used"
        cat confdefs.h - <<_ACEOF >conftest.$ac_ext
/* end confdefs.h.  */

#include <stdlib.h>
#include <stdarg.h>
#include <stdio.h>
#include <string.h>
#include <unistd.h>
/* Trivial function to help test for -fzero-call-used-regs */
int f(int n) {return rand() % n;}
char *f2(char *s, ...) {
        char ret[64];
        va_list args;
        va_start(args, s);
        vsnprintf(ret, sizeof(ret), s, args);
        va_end(args);
        return strdup(ret);
}
int i;
double d;
const char *f3(int s) {
        i = (int)d;

```

Over in config.log, it seems like the test is (incorrectly) passing

```
configure:9410: checking if mips-openwrt-linux-musl-gcc supports compile flag -fzero-call-used-regs=used and linking succeeds
configure:9471: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/root/openwrt/build_dir/target-mips_24kc_musl/openssh-without-pam/openssh-9.7p1=openssh-9.7p1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -pipe -Wno-error=format-truncation -Wall -Wextra -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-parameter -Wno-unused-result -Wimplicit-fallthrough -Wmisleading-indentation -fno-strict-aliasing -ftrapv -Werror -fzero-call-used-regs=used -I/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/include -I/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include/fortify -I/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/include  -L/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/usr/lib -L/root/openwrt/staging_dir/toolchain-mips_24kc_gcc-11.2.0_musl/lib -znow -zrelro  -Wl,-z,relro -Wl,-z,now -Wl,-z,noexecstack conftest.c  >&5
configure:9471: $? = 0
configure:9480: result: yes
```

bowserjklol, 2024-06-21:
Interesting that gcc11 on mips (the exact problematic combo we're dealing with here) was attempted to be addressed in [this commit](https://github.com/openssh/openssh-portable/commit/99a2df5e1994cdcb44ba2187b5f34d0e9190be91).  Not sure if we should move the conversation upstream as there have been several patches for -fzero-call-used-regs tests (as you've pointed out @neheb). 

Alternatively, we could include a configure.ac patch in the openwrt openssh package to skip the test on mips, roughly here in the fall through portion of this case statement...

```
	# clang 15 seems to have a bug in -fzero-call-used-regs=all.  See
	# https://bugzilla.mindrot.org/show_bug.cgi?id=3475 and
	# https://github.com/llvm/llvm-project/issues/59242
	# clang 17 has a different bug that causes an ICE when using this
	# flag at all (https://bugzilla.mindrot.org/show_bug.cgi?id=3629)
	case "$CLANG_VER" in
	apple-15*) OSSH_CHECK_CFLAG_LINK([-fzero-call-used-regs=used]) ;;
	17*)	;;
	*)	OSSH_CHECK_CFLAG_LINK([-fzero-call-used-regs=used]) ;;
	esac
```

neheb, 2024-06-21:
The simplest solution is a configure var override

bowserjklol, 2024-06-21:
Sure.  If we were to do this in the 22.03 branch, this resolves the issue locally for me.

```
diff --git a/net/openssh/Makefile b/net/openssh/Makefile
index dd5c17939..c1f3116c8 100644
--- a/net/openssh/Makefile
+++ b/net/openssh/Makefile
@@ -181,7 +181,12 @@ CONFIGURE_ARGS += \
        --with-stackprotect \
        --with$(if $(CONFIG_OPENSSL_ENGINE),,out)-ssl-engine \
        --with$(if $(CONFIG_OPENSSH_LIBFIDO2),,out)-security-key-builtin
-       
+
+ifeq ($(ARCH),mips)
+CONFIGURE_ARGS += \
+       --with-cflags-after="-fzero-call-used-regs=skip"
+endif
+
 ifeq ($(BUILD_VARIANT),with-pam)
 CONFIGURE_ARGS += \
        --with-pam
```

neheb, 2024-06-21:
I think this is failing because fortify-headers is preventing the check from failing.

Unfortunately, I don't see a way to disable fortify-headers from being used.

neheb, 2024-06-22:
It's not only failing under mips. https://downloads.openwrt.org/releases/faillogs-22.03/arm_cortex-a7_neon-vfpv4/packages/openssh/with-pam/compile.txt

neheb, 2024-07-04:
backported.

bowserjklol, 2024-07-05:
Confirmed and thanks @neheb!

closed 2024-07-05 by bowserjklol