openwrt/packages #24278: No Guide on how to actually create and test new packages exists

fkonradmain, 2024-05-31:
So, I have been trying to write a new package for this repository. And I have failed to find a guide on how to actually add packages, without rewriting the whole openwrt build chain. Fyi, the package is the [victoriametrics vmagent](https://github.com/VictoriaMetrics/VictoriaMetrics/tree/master/app/vmagent).

Description:

What I did to write my code was, following the [actually great manual](https://openwrt.org/docs/guide-developer/package):
1. clone the openwrt/openwrt repository
2. run `./scripts/feeds update -a && ./scripts/feeds install -a`
3. Create a new directory `feeds/packages/util/vmagent`
4. Created my package files.

When working myself through the manual, I found a part called [Testing a package Makefile](https://openwrt.org/docs/guide-developer/packages#testing_a_package_makefile), that was actually the part, I needed. So I ran `make package/utils/vmagent` from the root of the openwrt repo - as the guide tells me to do.

Nothing happened. it did not even find the package. I just ended up in the openwrt build selection, as we all know it.

```log
openwrt/include/toplevel.mk:231: target 'package/utils/vmagent' does not exist
```

I also tried running `make packages/utils/vmagent` and `make feeds/packages/utils/vmagent`. The former also leads to the openwrt build selection menu. The latter leads to the warning `make: 'feeds/packages/utils/vmagent' is up to date.`. Adding the words `download` or `check` don't work either. Why would they - There aren't even make targets called "download" or "check".

At that point, I found out, that make targets in "package/.../.../" actually point to "feeds/base/.../.../" and not to "feeds/packages/.../.../", which was very confusing.

Another idea was looking at gitlab pull requests and the CI pipelines. Yet there is actually no pipeline, that runs make. --> actually A Dockerfile might help with testing.

I also tried [adding the SDK](https://openwrt.org/docs/guide-developer/toolchain/using_the_sdk#obtain_the_sdk). Yet, I could not find the actual download link in the documentation. https://openwrt.org/docs/techref/targets/ipq40xx has no sdk downloads. Later, I actually discovered it on the download server: https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/

At that point, I had already stopped my development. It does not make sense to write Code when you have no idea how to test it. And getting the SDK to work, would probably lead to it not finding the target either.

That's why I would like to raise an issue to the community. We need a guide that actually does "make it incredibly easy to port software to OpenWrt". I am willing to participate in that process.

brada4, 2024-05-31:
And what change are you trying to achieve in this world?

stangri, 2024-05-31:
> When working myself through the manual, I found a part called [Testing a package Makefile](https://openwrt.org/docs/guide-developer/packages#testing_a_package_makefile), that was actually the part, [...]

Actually, the guide you linked explicitly says: 

```
To check your package:

make package/nlbwmon/download
make package/nlbwmon/check V=s
To automatically attempt to fix the Makefile:

make package/nlbwmon/check V=s FIXUP=1
Note: despite the similar name, this has nothing to do with the PKG_FIXUP variable presented below.
```

So in your case location/command would be `package/vmagnet`. Maybe you should try that.

Also, the Using SDK page has the section called "Obtain the SDK", which contains what looks to me like pretty good instructions on that, so I'm not sure why you were looking for the SDK links at the tech reference for a platform page.

fkonradmain, 2024-06-01:
Hi @stangri,

thank you for your response.

```
make package/nlbwmon/download
make package/nlbwmon/check V=s
To automatically attempt to fix the Makefile:
```

It does not explain in which path we are supposed to run the command. The repo root path does not work in my case and the subpaths do not work either. If someone does not know the correct path, the commands do not provide enough help to fix this own your own.

About the second part of your anser: I agree, yet it is a little misleading. You actually have to click on the first hyperlink and not on the second one. Although the first one points to the second one.

stangri, 2024-06-02:
> The repo root path does not work in my case and the subpaths do not work either.

You should really copy-paste the path/command you ran and the error if you expect a meaningful response.

I didn't have an opportunity to look deep into the package you're trying to make, but I'd start with the downloadable SDK (maybe even release version for stability) instead of building from sources.