openwrt/packages #24279: xtables-addons ERROR

kufei326, 2024-06-01:
make  all-recursive
make[5]: Entering directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
Making all in extensions
make[6]: Entering directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions'
Xtables-addons 3.24 - Linux 6.6.32
if [ -n "/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32" ]; then make -C /Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32 M=/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions modules; fi;
make[7]: Entering directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32'
  CC [M]  /Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ip.o
In file included from ../xtables-addons-3.24/extensions/LUA/prot_buf_ip.c:20:
./include/net/checksum.h: In function 'csum_shift':
./include/net/checksum.h:90:40: error: implicit declaration of function 'ror32' [-Werror=implicit-function-declaration]
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
In file included from ./include/linux/kernel.h:22,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/tcp.h:17,
                 from ./include/net/tcp.h:20,
                 from ../xtables-addons-3.24/extensions/LUA/prot_buf_ip.c:21:
./include/linux/bitops.h: At top level:
./include/linux/bitops.h:134:21: error: conflicting types for 'ror32'; have '__u32(__u32,  unsigned int)' {aka 'unsigned int(unsigned int,  unsigned int)'}
  134 | static inline __u32 ror32(__u32 word, unsigned int shift)
      |                     ^~~~~
./include/net/checksum.h:90:40: note: previous implicit declaration of 'ror32' with type 'int()'
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
cc1: all warnings being treated as errors
make[10]: *** [scripts/Makefile.build:243: /Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA/prot_buf_ip.o] Error 1
make[9]: *** [scripts/Makefile.build:480: /Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/LUA] Error 2
make[8]: *** [/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32/Makefile:1913: /Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions] Error 2
make[7]: *** [Makefile:234: __sub-make] Error 2
make[7]: Leaving directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32'
make[6]: *** [Makefile:468: modules] Error 2
make[6]: Leaving directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions'
make[5]: *** [Makefile:625: all-recursive] Error 1
make[5]: Leaving directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make[4]: *** [Makefile:397: all] Error 2
make[4]: Leaving directory '/Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make[3]: *** [Makefile:203: /Openwrt_lxc/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/.built] Error 2
make[3]: Leaving directory '/Openwrt_lxc/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#1.33#0.30#1.61
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:173: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/Openwrt_lxc/openwrt'
make[1]: *** [package/Makefile:167: /Openwrt_lxc/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Openwrt_lxc/openwrt'
make: *** [/Openwrt_lxc/openwrt/include/toplevel.mk:233: world] Error 2

brada4, 2024-06-01:
DUP: https://github.com/openwrt/packages/issues/24026 (potential fix via lede in latest post)

qingtian110, 2024-06-18:
https://github.com/sbwml/r4s_build_script/blob/master/openwrt/patch/packages-patches/xtables-addons/202-fix-lua-packetscript-for-linux-6.6.patch

httpstorm, 2024-07-11:
Fixed in the first commit here https://github.com/openwrt/packages/pull/24535

closed 2025-05-23 by BKPepe