openwrt/packages #24282: libwebsockets-4.3.2 - compiling error "ws-genrsa.h:51:9: error: unknown type name RSA *rsa;  Openwrt Release 23

desigabri, 2024-06-01:
Maintainer: andy@warmcat.com

Environment: Qualcomm Atheros IPQ806X , TP-Link Archer C2600 v1, openwrt-23.05

Description:

Make stops compiling firmware with the error:  "compiling error "/libwebsockets/ws-genrsa.h:51:9: error: unknown type name RSA *rsa;"

desigabri, 2024-06-01:
I tryed differents configuration files,  Starting with DEFAULT .config files, the compiling finishes correctly,

brada4, 2024-06-01:
Format your diffconfig please,. Can you collect erro$s compiling just one package on top of default config?

desigabri, 2024-06-02:
I could try...which packet have to try? I don't know who calls libwebsockets, I imagine it's called by more packages...what are you asking for please? entering the directory package and tring to compile it alone from there?

desigabri, 2024-06-02:
what does it means "format your deffconfig"?  I gitpulled the sources from scratch,

desigabri, 2024-06-02:
Ok found that package ttyd wants libwebsockets, so **EXCLUDING luci-app-ttyd** and **ttyd**, the compiling works fine.

closed 2024-06-02 by desigabri