openwrt/packages #24284: wsdd2 fails to compile with GCC14

sppmasterspp, 2024-06-01:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (qualcommax, mvebu OpenWrt main snapshot build)

Description:
Wsdd2 compilation fails with GCC14.

I've tried it myself and got this error.
```
sed 's|VERSION|017|g;s|@usbids@|/usr/share/hwdata/usb.ids|g' lsusb.py.in >lsusb.py
chmod 755 lsusb.py
sed 's|VERSION|017|g' lsusb.8.in >lsusb.8
sed 's|VERSION|017|g' usbhid-dump.8.in >usbhid-dump.8
sed 's|VERSION|017|g' usb-devices.1.in >usb-devices.1
make[6]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
make[5]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
make[4]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017=usbutils-017 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/usr/include -I/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/include -I/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/include/fortify " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017=usbutils-017 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/usr/include -I/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/include -I/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/include/fortify " LDFLAGS="-L/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/usr/lib -L/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-14.1.0_musl/lib -fuse-ld=bfd -znow -zrelro " make -C /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017=usbutils-017 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="aarch64-openwrt-linux-musl-ld.bfd" NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" DESTDIR="/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install"  install;
make[4]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
Making install in usbhid-dump
make[5]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump'
Making install in src
make[6]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump/src'
make[7]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump/src'
 /usr/bin/mkdir -p '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin'
  /home/debian/openwrt/staging_dir/host/bin/install -c usbhid-dump '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin'
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump/src'
make[6]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump/src'
make[6]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump'
make[7]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump'
make[7]: Nothing to be done for 'install-exec-am'.
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump'
make[6]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump'
make[5]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbhid-dump'
make[5]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
make[6]: Entering directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
 /usr/bin/mkdir -p '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin'
  /home/debian/openwrt/staging_dir/host/bin/install -c lsusb '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin'
 /usr/bin/mkdir -p '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin'
 /home/debian/openwrt/staging_dir/host/bin/install -c usb-devices lsusb.py '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin'
 /usr/bin/mkdir -p '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/man/man1'
 /home/debian/openwrt/staging_dir/host/bin/install -c -m 644 usb-devices.1 '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/man/man1'
 /usr/bin/mkdir -p '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/man/man8'
 /home/debian/openwrt/staging_dir/host/bin/install -c -m 644 lsusb.8 usbhid-dump.8 '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/man/man8'
 /usr/bin/mkdir -p '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/lib/pkgconfig'
 /home/debian/openwrt/staging_dir/host/bin/install -c -m 644 usbutils.pc '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/lib/pkgconfig'
make[6]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
make[5]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
make[4]: Leaving directory '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017'
touch /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.built
rm -rf /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils.installed /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils
mkdir -p /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils
install -d -m0755 /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils/usr/bin
install -m0755 /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/ipkg-install/usr/bin/lsusb /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils/usr/bin/
install -m0755 /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/usbreset /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils/usr/bin/
touch /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils.installed
mkdir -p /home/debian/openwrt/staging_dir/target-aarch64_cortex-a72_musl/root-mvebu/stamp
SHELL= flock /home/debian/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/usbutils-017/.pkgdir/usbutils/. /home/debian/openwrt/staging_dir/target-aarch64_cortex-a72_musl/root-mvebu/'
touch /home/debian/openwrt/staging_dir/target-aarch64_cortex-a72_musl/root-mvebu/stamp/.usbutils_installed
[... cut ...]
    ERROR: package/feeds/packages/wsdd2 failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/wsdd2/compile] Error 1
make[2]: Leaving directory '/home/debian/openwrt'
make[1]: *** [package/Makefile:123: /home/debian/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/openwrt'
make: *** [/home/debian/openwrt/include/toplevel.mk:233: world] Error 2

```
There are more reports on openwrt forum about this.
Ping @Lucky1openwrt @neheb

brada4, 2024-06-01:
Add the cast? Like (int)NULL

neheb, 2024-06-01:
musl has padding in the struct for 64-bit architectures. No idea why. Anyway, proper fix:

```
struct msghdr msg = {
    .msg_name = (void *)&sa,
    .msg_namelen = (socklen_t)sizeof(sa),
    .msg_iov = &iov,
    .msg_iovlen = 1,
    .msg_control = NULL,
    .msg_controllen = 0,
    .msg_flags = 0
};
```

sppmasterspp, 2024-06-01:
> musl has padding in the struct for 64-bit architectures. No idea why. Anyway, proper fix:
> 
> ```
> struct msghdr msg = {
>     .msg_name = &sa,
>     .msg_namelen = sizeof(sa),
>     .msg_iov = &iov,
>     .msg_iovlen = 1,
>     .msg_control = NULL,
>     .msg_controllen = 0,
>     .msg_flags = 0
> };
> ```

How do I apply this?

vortexilation, 2024-06-02:
```
(int)NULL
```
It will error out on glibc, for me it needs just "0" for both working in glibc and musl :  
```
diff -Naur a/wsdd2.c b/wsdd2.c
--- a/wsdd2.c
+++ b/wsdd2.c
@@ -543,7 +543,7 @@
 	char buf[PAGE_SIZE];
 	struct sockaddr_nl sa;
 	struct iovec iov = { buf, sizeof buf };
-	struct msghdr msg = { &sa, sizeof sa, &iov, 1, NULL, 0, 0 };
+	struct msghdr msg = { &sa, sizeof sa, &iov, 1, 0, 0, 0 };
 	ssize_t msglen = recvmsg(ep->sock, &msg, 0);
 
 	DEBUG(2, W, "%s: %zd bytes", __func__, msglen);
```

neheb, 2024-06-02:
Proper fix pushed.

closed 2024-06-06 by neheb