openwrt/packages #24325: nano: ucode syntax highlighting breaks when encountering "}}"

eric-j-ason, 2024-06-04:
Maintainer: @hnyman
Environment: master

Syntax highlighting of ucode stops working when the string `}}` is encountered in the source code.

```ucode
#! /usr/bin/env ucode

({a: {}});

// Everyting from here on gets the same color.
print("Hello!\n");
0 + 1 + 2 + 3 + 4 + 5;
return null;
```

[image]

dave14305, 2024-06-05:
It’s working, but the use of a double `}}` triggers the template rule. If you separate them by a space, it should snap back to normal, no?

https://github.com/openwrt/packages/blob/4ef2b7e0fc3d0cb714e7522b8c9fc57185681560/utils/nano/files/ucode.nanorc#L44-L45

eric-j-ason, 2024-06-05:
> It’s working, but the use of a double `}}` triggers the template rule. If you separate them by a space, it should snap back to normal, no?

Indeed, that's my current workaround, but it's ugly and I don't want to have to do that! :) I'm not using templates at all.