openwrt/packages #24332: zabbix 7.0 support

fidelix, 2024-06-06:
Can we please get zabbix 7?

It is backward-compatible with zabbix 6.4, and improves performance considerably.
While at it, it would be great to get libcurl support for zabbix too, it isn't very useful without it.

diasdmhub, 2024-06-07:
Unfortunately, Openwrt Zabbix packages are not updated often and are sometimes limited, as seen in my request #15419.

tbsky, 2024-06-18:
Hi:
   I want to said some thing about zabbix-agent.  old zabbix server can not support new zabbix agent. so currently 23.05 and snapshot which include zabbix agent 6.4 can only supported by zabbix 6.4/7.0 server.  but zabbix server version 6.2/6.4 is standard version and has much shorter life than lts versions (6.0/7.0). I  think many people will use lts zabbix server since it is easier to maintain.

  since zabbix 7.0 lts is released, I hope openwrt zabbix-agent can maintain a lts version in the future.  maybe openwrt should just pick up the lts version and ignore the standard version. or maybe use different package names like other distribution. (for example: zabbix70-agent, zabbix72-agent)

troffasky, 2024-07-09:
> old zabbix server can not support new zabbix agent

This simply is not true. I have a Zabbix 4.4 server with hundreds of agents from version 4.x all the way up to 7.0.
Obviously the version 4 agent will not support an item key only added in a later version of Zabbix, but that is a very long way from not being usable - Zabbix agent has had all the useful metrics since 10+ years so you are unlikely to hit a showstopper due to this.

I agree that targeting LTS releases makes more sense that just using whatever was current at the time.

fidelix, 2024-07-09:
Indeed, the Zabbix agent can be of a different version.
It's the zabbix proxy that has to be of the same major version.

Makes no difference from the standpoint that this issue is proposing

tbsky, 2024-07-10:
> > old zabbix server can not support new zabbix agent
> 
> This simply is not true. I have a Zabbix 4.4 server with hundreds of agents from version 4.x all the way up to 7.0. 

this maybe a working configuration, but I think it is not the "supported" configuration.  document already said it like this (https://www.zabbix.com/documentation/6.0/en/manual/appendix/compatibility):

Supported agents
To be compatible with Zabbix 6.0, Zabbix agent must not be older than version 1.4 and must not be newer than 6.0.

santiagobiali, 2024-07-10:
I believe we should follow Zabbix's EOL [Full support cycle](https://www.zabbix.com/life_cycle_and_release_policy). As Zabbix 7 brings a lot in performance to the table, to me it's a no-brainer and openwrt should use it as it's default package.

EOL Zabbix 6.0 LTS: 28/02/2025
EOL Zabbix 6.4 (non-LTS): 30/06/2024 (not supported anymore)
EOL Zabbix 7.0 LTS: 30/06/2027
EOL Zabbix 7.xx (non-LTS): TBD

fidelix, 2025-01-10:
@GrumpyMeow but the packages in openwrt are not.
agentd is currently at 6.4.7-1

diasdmhub, 2025-01-10:
> I think this issue should be closed as Zabbix is currently at v7.0.0.

Actually, the latest versions at the moment are

- 7.0.8
- 7.2.2

GrumpyMeow, 2025-01-10:
@fidelix Indeed the latest Zabbix version is v6.4.7-1 for OpenWRT v23.05. Openwrt v24.10 should bring you Zabbix v7.

tbsky, 2025-01-11:
> I think this issue should be closed as Zabbix is currently at v7.0.0.

hope we can prevent any .0.0 version if possible. but I hadn't use that version, maybe it is stable.

danielfdickinson, 2026-01-19:
24.10: https://github.com/openwrt/packages/pull/28150
25.12: https://github.com/openwrt/packages/pull/28122

closed 2026-01-19 by BKPepe