openwrt/packages #24365: curl: (35) ssl_handshake returned

lengfwong, 2024-06-10:
Maintainer: @krant 
Environment: x86/64 OpenWrt SNAPSHOT r26581-33db914607 / LuCI Master 24.158.03388~a6f8361
curl: 8.8.0
libmbedtls21: 3.6.0
libustream-mbedtls20201210: 2024.04.19~524a76e5

Description:

```
curl --connect-timeout 5 -m 120 --ipv4 -vkfSLo "./apple-cn.txt" "https://github.com/Loyalsoldier/v2ray-rules-dat/relea

ses/latest/download/apple-cn.txt"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* ssl_handshake returned - mbedTLS: (-0x2700) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (35) ssl_handshake returned - mbedTLS: (-0x2700) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
```

Ra2-IFV, 2024-06-10:
Did you install `ca-bundle` or `ca-certificates` package on your system?

lengfwong, 2024-06-10:
> Did you install `ca-bundle` or `ca-certificates` package on your system?

yes, have installed:
ca-bundle	20240203-r1
ca-certificates	20240203-r1

Ra2-IFV, 2024-06-10:
I guess a proxy software handled your request and responded with self-signed certificate, or it's a MITM attack (not likely but possible)

lengfwong, 2024-06-10:
> openssl s_client -connect github.com:443

Thanks.
```
root@OpenWrt:~# curl -V
curl 8.8.0 (x86_64-openwrt-linux-gnu) libcurl/8.8.0 mbedTLS/3.6.0 nghttp2/1.62.1
Release-Date: 2024-05-22
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets

root@OpenWrt:~# openssl s_client -connect github.com:443
CONNECTED(00000003)
depth=2 C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust ECC Certification Authority
verify return:1
depth=1 C = GB, ST = Greater Manchester, L = Salford, O = Sectigo Limited, CN = Sectigo ECC Domain Validation Secure Server CA
verify return:1
depth=0 CN = github.com
verify return:1
---
Certificate chain
 0 s:CN = github.com
   i:C = GB, ST = Greater Manchester, L = Salford, O = Sectigo Limited, CN = Sectigo ECC Domain Validation Secure Server CA
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: ecdsa-with-SHA256
   v:NotBefore: Mar  7 00:00:00 2024 GMT; NotAfter: Mar  7 23:59:59 2025 GMT
 1 s:C = GB, ST = Greater Manchester, L = Salford, O = Sectigo Limited, CN = Sectigo ECC Domain Validation Secure Server CA
   i:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust ECC Certification Authority
   a:PKEY: id-ecPublicKey, 256 (bit); sigalg: ecdsa-with-SHA384
   v:NotBefore: Nov  2 00:00:00 2018 GMT; NotAfter: Dec 31 23:59:59 2030 GMT
 2 s:C = US, ST = New Jersey, L = Jersey City, O = The USERTRUST Network, CN = USERTrust ECC Certification Authority
   i:C = GB, ST = Greater Manchester, L = Salford, O = Comodo CA Limited, CN = AAA Certificate Services
   a:PKEY: id-ecPublicKey, 384 (bit); sigalg: RSA-SHA384
   v:NotBefore: Mar 12 00:00:00 2019 GMT; NotAfter: Dec 31 23:59:59 2028 GMT
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIEozCCBEmgAwIBAgIQTij3hrZsGjuULNLEDrdCpTAKBggqhkjOPQQDAjCBjzEL
MAkGA1UEBhMCR0IxGzAZBgNVBAgTEkdyZWF0ZXIgTWFuY2hlc3RlcjEQMA4GA1UE
BxMHU2FsZm9yZDEYMBYGA1UEChMPU2VjdGlnbyBMaW1pdGVkMTcwNQYDVQQDEy5T
ZWN0aWdvIEVDQyBEb21haW4gVmFsaWRhdGlvbiBTZWN1cmUgU2VydmVyIENBMB4X
DTI0MDMwNzAwMDAwMFoXDTI1MDMwNzIzNTk1OVowFTETMBEGA1UEAxMKZ2l0aHVi
LmNvbTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABARO/Ho9XdkY1qh9mAgjOUkW
mXTb05jgRulKciMVBuKB3ZHexvCdyoiCRHEMBfFXoZhWkQVMogNLo/lW215X3pGj
ggL+MIIC+jAfBgNVHSMEGDAWgBT2hQo7EYbhBH0Oqgss0u7MZHt7rjAdBgNVHQ4E
FgQUO2g/NDr1RzTK76ZOPZq9Xm56zJ8wDgYDVR0PAQH/BAQDAgeAMAwGA1UdEwEB
/wQCMAAwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMEkGA1UdIARCMEAw
NAYLKwYBBAGyMQECAgcwJTAjBggrBgEFBQcCARYXaHR0cHM6Ly9zZWN0aWdvLmNv
bS9DUFMwCAYGZ4EMAQIBMIGEBggrBgEFBQcBAQR4MHYwTwYIKwYBBQUHMAKGQ2h0
dHA6Ly9jcnQuc2VjdGlnby5jb20vU2VjdGlnb0VDQ0RvbWFpblZhbGlkYXRpb25T
ZWN1cmVTZXJ2ZXJDQS5jcnQwIwYIKwYBBQUHMAGGF2h0dHA6Ly9vY3NwLnNlY3Rp
Z28uY29tMIIBgAYKKwYBBAHWeQIEAgSCAXAEggFsAWoAdwDPEVbu1S58r/OHW9lp
LpvpGnFnSrAX7KwB0lt3zsw7CAAAAY4WOvAZAAAEAwBIMEYCIQD7oNz/2oO8VGaW
WrqrsBQBzQH0hRhMLm11oeMpg1fNawIhAKWc0q7Z+mxDVYV/6ov7f/i0H/aAcHSC
Ii/QJcECraOpAHYAouMK5EXvva2bfjjtR2d3U9eCW4SU1yteGyzEuVCkR+cAAAGO
Fjrv+AAABAMARzBFAiEAyupEIVAMk0c8BVVpF0QbisfoEwy5xJQKQOe8EvMU4W8C
IGAIIuzjxBFlHpkqcsa7UZy24y/B6xZnktUw/Ne5q5hCAHcATnWjJ1yaEMM4W2zU
3z9S6x3w4I4bjWnAsfpksWKaOd8AAAGOFjrv9wAABAMASDBGAiEA+8OvQzpgRf31
uLBsCE8ktCUfvsiRT7zWSqeXliA09TUCIQDcB7Xn97aEDMBKXIbdm5KZ9GjvRyoF
9skD5/4GneoMWzAlBgNVHREEHjAcggpnaXRodWIuY29tgg53d3cuZ2l0aHViLmNv
bTAKBggqhkjOPQQDAgNIADBFAiEAru2McPr0eNwcWNuDEY0a/rGzXRfRrm+6XfZe
SzhYZewCIBq4TUEBCgapv7xvAtRKdVdi/b4m36Uyej1ggyJsiesA
-----END CERTIFICATE-----
subject=CN = github.com
issuer=C = GB, ST = Greater Manchester, L = Salford, O = Sectigo Limited, CN = Sectigo ECC Domain Validation Secure Server CA
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 3483 bytes and written 380 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 256 bit
Secure Renegotiation IS NOT supported
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
---
Post-Handshake New Session Ticket arrived:
SSL-Session:
    Protocol  : TLSv1.3
    Cipher    : TLS_AES_128_GCM_SHA256
    Session-ID: EBCCCA8C1873E5DDD93C8A04B57BFA7921921DB19871683DDB2F0B92F97A7F66
    Session-ID-ctx:
    Resumption PSK: 5E0192708CDD0FDCEC0B65DFF7B49D2AB29D827134ABB6E4E30A2FED5BE86203
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - d7 8a 6c dd ef dc de 7a-45 ef 4e 28 2d 30 8e 06   ..l....zE.N(-0..
    0010 - 96 1d 29 be b9 6e 23 20-99 39 c8 57 ee 20 67 87   ..)..n# .9.W. g.

    Start Time: 1718037760
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
---
Post-Handshake New Session Ticket arrived:
SSL-Session:
    Protocol  : TLSv1.3
    Cipher    : TLS_AES_128_GCM_SHA256
    Session-ID: 1D15459B2B4F086309899F4BDBFB2ED3A3F59919A0566F6410A675BFFCF090E3
    Session-ID-ctx:
    Resumption PSK: 80DDA40707127C017A38D771594DDFEEC5A6D98B587CA58D8F6498C1BBBA6EFF
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - ad ef 29 b9 cf 7c dd d0-40 12 66 f5 00 06 c3 e2   ..)..|..@.f.....
    0010 - 36 c1 78 17 1c 77 54 d3-4d 92 f9 62 39 93 97 a7   6.x..wT.M..b9...

    Start Time: 1718037760
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
closed

```

gili-gili, 2024-06-10:
yeah, I see a successful handshake.  
Then how about `curl --ipv4 -vkSL "https://github.com/Loyalsoldier/v2ray-rules-dat/releases/latest/download/apple-cn.txt"`?

lengfwong, 2024-06-10:
> yeah, I see a successful handshake. Then how about `curl --ipv4 -vkSL "https://github.com/Loyalsoldier/v2ray-rules-dat/releases/latest/download/apple-cn.txt"`?

Same:
```
root@OpenWrt:~# curl --ipv4 -vkSL "https://github.com/
Loyalsoldier/v2ray-rules-dat/releases/latest/download/
apple-cn.txt"
* ssl_handshake returned - mbedTLS: (-0x2700) X509 - C
ertificate verification failed, e.g. CRL, CA or signat
ure check failed
curl: (35) ssl_handshake returned - mbedTLS: (-0x2700)
 X509 - Certificate verification failed, e.g. CRL, CA
or signature check failed
```

Ra2-IFV, 2024-06-13:
Oh fuck I am having the same error, on other devices curl returns correctly, and on openwrt it just outputs "error" without additional information. Since we already tested the connection with openssl and it works fine, mbedtls is very sus

Ra2-IFV, 2024-06-13:
You are using a custom build right? Build Libraries-libcurl with libopenssl and try again

Ra2-IFV, 2024-06-13:
I re-built libcurl and curl and its working now. The default configuration for libmbedtls must dropped something important feature so that it failed to read /etc/ssl. Pending investigation.

Catfriend1, 2024-06-13:
Happens on my OpenWrt CURL too. I've had "curl --insecure ..." calls in my scripts and they failed with that error above. Removing "--insecure" fixed it, e.g. api.telegram.org is connecting fine now. But one of my servers presenting a valid LetsEncrypt certificate still leads to this error.

@Ra2-IFV  So should I wait for an update to the curl package for OpenWrt? Is there a way to disable the check temporarily until its fixed by a command line switch? I've already tried --ssl-revoke-best-effort and --ssl-no-revoke plus --insecure but it throws the error.

Ra2-IFV, 2024-06-14:
If you are too lazy to build, here are the ipk files  
[build.zip](https://github.com/user-attachments/files/15829951/build.zip)  
```text
opkg remove --autoremove --force-depends libopenssl3 libcurl4 curl
opkg install /path/to/ipk
```
or you can use web interface to upload new packages. Good luck.

lengfwong, 2024-06-14:
> If you are too lazy to build, here are the ipk files [build.zip](https://github.com/user-attachments/files/15829951/build.zip)
> 
> ```
> opkg remove --autoremove --force-depends libopenssl3 libcurl4 curl
> opkg install /path/to/ipk
> ```
> 
> or you can use web interface to upload new packages. Good luck.

Thanks. 
My firmware already contains the libopenssl3_3.0.14-r1, it seems useless. Someone on the Internet said that the Libraries-libusteam-openssl could be used instead of libusteam-mbedtls, I rebuild but no help. 
Now works well by dropping commit [49fc257(](https://github.com/openwrt/packages/commit/49fc257b45c68ddc8496a9087fe57cc371bd8f86) and converting to curl8.7.1.

Ra2-IFV, 2024-06-14:
no, the problem is libcurl. it was linked with mbedtls. just reinstall it

krant, 2024-06-16:
wget is broken too:
```
# wget https://raw.githubusercontent.com
Downloading 'https://raw.githubusercontent.com'
Connecting to 185.199.111.133:443
Redirected to / on github.com
SSL error: SSL - Bad input parameters to function
Connection error: Connection failed
```
While this works:
```
# wget https://github.com
Downloading 'https://github.com'
Connecting to 140.82.121.3:443
Writing to 'index.html'

Download completed (237596 bytes)
```

bassopt, 2024-06-18:
DynamicDNS is also broke due to this issue.
Same error.
`curl: (35) ssl_handshake returned - mbedTLS: (-0x2700) X509 - Certificate verification failed, e.g. CRL, CA or signature check faile`

Ra2-IFV, 2024-06-18:
in short: In `Libraries>libcurl`, select openssl. mbedtls is broken

bassopt, 2024-06-18:
Yes, that works, but it means compiling everytime a a new image is created using image builder, which is annoying.

Ra2-IFV, 2024-06-18:
just wait for the fix...  
I tried selecting everything in mbedtls menuconfig but it still returned a error, without additional information. It should be a compatibility issue

Ra2-IFV, 2024-06-18:
https://github.com/Mbed-TLS/mbedtls/issues/9210

PR #24414 (2024-06-20): curl: upstream backports for mbedtls
Maintainer: @krant
Compile tested: aarch64 531b3f667c40405581a398a9648a0e6c27909a87
Run tested: aarch64 531b3f667c40405581a398a9648a0e6c27909a87, just simple connection test

Description:
tlsv1.3 support is broken in curl 8.8.0 with mbedtls 3.6.0.  
See https://github.com/curl/curl/issues/13653 and https://github.com/Mbed-TLS/mbedtls/issues/9210 for more details.  
A workaround was implemented in upsteam codes, see https://github.com/curl/curl/commit/0c4b4c1e93c8e869af230090f32346fdfd548f21 and https://github.com/curl/curl/commit/5f9017d4e28096b4589c4d5ee4f [... cut]
commit d97d07c6 (2024-06-20): curl: upstream backports for mbedtls

tlsv1.3 support is broken in curl 8.8.0 with mbedtls 3.6.0.
See curl/curl#13653 and Mbed-TLS/mbedtls#9210 for more details.
A workaround was implemented in upsteam code, see curl/curl@0c4b4c1 and curl/curl@5f9017d
This commit includes patches generated from upstream commits.

fix #24365 #24386
closed 2024-06-20 by 1715173329