openwrt/packages #24366: git: build fails with LTO enabled

Ra2-IFV, 2024-06-10:
# Issue template (remove lines from top till here)

Maintainer: @neheb  
Environment: aarch64 generic 2b01007dd47b438222332bd64b75304ea5ed5ff6

Description:
Enabling LTO for `git` will cause build to fail, the error log is here  
```text
/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h: In function 'process_curl_messages':
/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h:90:8: error: inlining failed in call to 'always_inline' 'fwrite': function body can be overwritten at link time
   90 | size_t fwrite(const void *__restrict, size_t, size_t, FILE *__restrict);
      |        ^
http.c:355:33: note: called from here
  355 |                                 fprintf(stderr, "Received DONE message for unknown request!\n");
      |                                 ^
In function 'fetch_using_walker',
    inlined from 'cmd_main' at http-fetch.c:167:9:
/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h:90:8: error: inlining failed in call to 'always_inline' 'fwrite': function body can be overwritten at link time
   90 | size_t fwrite(const void *__restrict, size_t, size_t, FILE *__restrict);
      |        ^
http-fetch.c:39:17: note: called from here
   39 |                 fprintf(stderr,
      |                 ^
make[4]: *** [/openwrt/tmp/cc7XwYlE.mk:2: /openwrt/tmp/ccvHfD60.ltrans0.ltrans.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In function 'create_ref_symlink',
    inlined from 'create_symref_locked' at refs/files-backend.c:1927:30,
    inlined from 'files_create_symref' at refs/files-backend.c:1963:8:
/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h:90:8: error: inlining failed in call to 'always_inline' 'fwrite': function body can be overwritten at link time
   90 | size_t fwrite(const void *__restrict, size_t, size_t, FILE *__restrict);
      |        ^
refs/files-backend.c:1901:17: note: called from here
 1901 |                 fprintf(stderr, "no symlink - falling back to symbolic ref\n");
      |                 ^
make[4]: *** [/openwrt/tmp/cc7XwYlE.mk:23: /openwrt/tmp/ccvHfD60.ltrans7.ltrans.o] Error 1
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
mold: fatal: lto-wrapper failed
```

brada4, 2024-06-11:
Could you share diffconfig?

Ra2-IFV, 2024-06-11:
What should I compare my config with?

Ra2-IFV, 2024-06-14:
it should has nothing to do with custom config file. just LTO problems.  
My suggestion is simply append `no-lto` to `BUILD_PKG_FLAGS` in Makefile as a workaround, since it's not important at all, and it may take developers a long time to make git compatible with LTO.

brada4, 2024-06-14:
Default is gcc13, it is imperative that you show diffconfig that leads to error, or alternatively submit a PR.

Ra2-IFV, 2024-06-14:
yeah, I'm using GCC14 for testing purpose, enabled `-O3` and LTO in `make menuconfig`  
~~So may I ask, how to run diffconfig? It's not a standard command.~~ It's in the scripts dir, sorry  
GCC13, LTO enabled, diffconfig ouput:  
```
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_EXTRA_OPTIMIZATION="-Ofast -fno-caller-saves -fno-plt -pipe"
# CONFIG_GDB is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_git=m
CONFIG_PACKAGE_git-http=m
CONFIG_PACKAGE_libcurl=m
CONFIG_PACKAGE_libnghttp2=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_zlib=m
CONFIG_TARGET_OPTIMIZATION="-Os -pipe -march=armv8-a -mtune=cortex-a72.cortex-a53"
CONFIG_TARGET_OPTIONS=y
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
CONFIG_USE_MOLD=y
```
`make package/feeds/packages/git/{clean,prepare,compile} -j1 V=s` output: [git.log](https://github.com/user-attachments/files/15839239/git.log)

brada4, 2024-06-14:
> /openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h

Yeah, gcc13, maybe even gcc12....

Ra2-IFV, 2024-06-14:
wtf I did ran `make dirclean` before...

brada4, 2024-06-14:
Make coinciding error and diffconfig. Without -ofast which is guaranteed to miscompile

Ra2-IFV, 2024-06-15:
diffconfig:  
```
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_EXTRA_OPTIMIZATION="-fno-caller-saves -fno-plt -pipe"
# CONFIG_GDB is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_git=m
CONFIG_PACKAGE_git-http=m
CONFIG_PACKAGE_libcurl=m
CONFIG_PACKAGE_libnghttp2=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_zlib=m
CONFIG_TARGET_OPTIMIZATION="-pipe -march=armv8-a -mtune=cortex-a72.cortex-a53 -O3"
CONFIG_TARGET_OPTIONS=y
# CONFIG_TARGET_ROOTFS_EXT4FS is not set
CONFIG_USE_GC_SECTIONS=y
CONFIG_USE_LTO=y
CONFIG_USE_MOLD=y
```
build log: [git.log](https://github.com/user-attachments/files/15845514/git.log)

Must both use (`-O3` or `-Ofast`) and LTO to reproduce this error. Test passed with `-O2`.

Ra2-IFV, 2024-06-15:
Close this issue?

brada4, 2024-06-15:
Yep, O3/Ofast is excessive optimization . Like `1.0*(2+3)/4`  can evaluate in any order ignoring parens

Ra2-IFV, 2024-11-13:
New update on this:
This is a reproducible bug, reported in this mailing list https://lore.kernel.org/git/CAHfWF5mjquES-nocQaK+CAEsqWgdy-_OYdGtN82heYs0eJP3eQ@mail.gmail.com/T/#t
And I was wrong, it's caused by fortify headers.

PR #25334 (2025-01-22): git: Set PKG_FORTIFY_SOURCE to 0 when lto enabled
If PKG_FORTIFY_SOURCE is not 0 and lto is enabled for package git, git will fail to compile with the following example error:

/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h: In function 'process_curl_messages': /openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h:90:8: error: inlining failed in call to 'always_inline' 'fwrite': function body can be overwritten at link time
   90 | size_t fwrite(const void *__restrict, size_t, size_t, FILE *__restrict);
      |        ^
http.c:355:33: no [... cut]
commit b7bbd425 (2025-01-22): git: Set PKG_FORTIFY_SOURCE to 0 when lto enabled

If PKG_FORTIFY_SOURCE is not 0 and lto is enabled for package git, it
will fail to compile with the following example error:

/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h: In function 'process_curl_messages': /openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/stdio.h:90:8: error: inlining failed in call to 'always_inline' 'fwrite': function body can be overwritten at link time
   90 | size_t fwrite(const void *__restrict, size_t, size_t, FILE *__restrict);
      |        ^
http.c:355:33: no [... cut]
closed 2025-01-22 by neheb