openwrt/packages #24375: rust: need a way to pass ldflags to rustc?

vortexilation, 2024-06-13:
Maintainer: @lu-zero @1715173329 
Environment: x86_64, glibc

Description:

I wanted to compile [einat](https://github.com/muink/openwrt-einat-ebpf) , but somehow rustc cannot found libintl at the end.

I have tried everything 
```
TARGET_CFLAGS += -Xlinker -rpath=$(STAGING_DIR)/usr/lib/libintl-full/lib
or
TARGET_LDFLAGS += -L$(STAGING_DIR)/usr/lib/libintl-full/lib -lintl
or
export LD_LIBRARY_PATH=$(STAGING_DIR)/usr/lib/libintl-full/lib
```
but rustc still didn't picked it up.
```
   Compiling einat v0.1.2 (/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3)
     Running `rustc --crate-name build_script_build --edition=2021 build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off -C overflow-checks=on --cfg 'feature="default"' --cfg 'feature="ipv6"' -C metadata=13037c25982734ec -C extra-filename=-13037c25982734ec --out-dir /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec -C linker=x86_64-openwrt-linux-gnu-gcc -C strip=symbols -L dependency=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps --extern libbpf_cargo=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-fc7173ecf7072a22.rlib -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release -L native=/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out`
error: linking with `x86_64-openwrt-linux-gnu-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/bin:/home/user/works/openwrt/staging_dir/hostpkg/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/host/bin:/home/user/works/openwrt/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" VSLANG="1033" "x86_64-openwrt-linux-gnu-gcc" "-m64" "/home/user/works/openwrt/tmp/rustca9rJSt/symbols.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec/build_script_build-13037c25982734ec.build_script_build.900026c213c08482-cgu.0.rcgu.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec/build_script_build-13037c25982734ec.2kuwii56gzittnew.rcgu.o" "-Wl,--as-needed" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-fc7173ecf7072a22.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_metadata-77b698b469ccad23.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libserde_json-d18e1952940c9d6c.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libitoa-c99bf860a1243733.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libryu-42733ab1749368f9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libthiserror-aa231274481bae73.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_platform-c20f7cad56b30019.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcamino-a1fd712e042bd596.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libmemmap2-73fca818d05ca4c9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_rs-12af648727bedc4e.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libvsprintf-52f5d1afc74d566c.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibc-a50be116079b6680.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_sys-6ce4fe8ed806fc57.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libsemver-d9af4a6bbaa40b09.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libserde-73abfa3b10e52ca3.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libregex-662c0e78adcebbf3.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libregex_automata-e8e6e085a55c8ab3.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libregex_syntax-f1d457fe75bcbfd6.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libtempfile-e4833a2ad6c4f429.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libfastrand-906c084e5ef6039e.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/librustix-4d3b20d2716f8231.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libbitflags-3a495cf5bbfede0b.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblinux_raw_sys-ba02cfb0a81adfba.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcfg_if-1c8851ce38e8a7c1.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libanyhow-0deb8dd3c263e7fc.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libstd-18d1c84127d04dc5.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libpanic_unwind-bc53c6e5d5f4111f.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libobject-a79bc22e6ae9ee83.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libmemchr-9c256084533d6dad.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libaddr2line-e53b44d64d10bb35.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libgimli-9d0b3274a5b15e82.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_demangle-b0fb619b58a6601a.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_ [... cut ...] .1.3/target/release/build/vsprintf-f6a1be3b8a629995/out` (exit status: 1)
warning: build failed, waiting for other jobs to finish...
error: failed to compile `einat v0.1.2 (/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3)`, intermediate artifacts can be found at `/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

lu-zero, 2024-06-13:
Doesn't look right. Where is `libintl.so.8` and why it is needed?

lu-zero, 2024-06-13:
The build.rs for libbpf uses `LD_LIBRARY_PATH` https://github.com/libbpf/libbpf-sys/blob/master/build.rs#L188C36-L188C51

vortexilation, 2024-06-13:
> Doesn't look right. Where is `libintl.so.8` and why it is needed?

Because my OpenWrt build is NLS enabled, libintl.so.8 is located in here : 
```
/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libintl-full/lib
or
$(STAGING_DIR)/usr/lib/libintl-full/lib
```

> The build.rs for libbpf uses `LD_LIBRARY_PATH` https://github.com/libbpf/libbpf-sys/blob/master/build.rs#L188C36-L188C51

The compilation is failing at einat not libbpf

```
root@buildsystem:/home/user/works/openwrt# ldd /home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libelf.so
        linux-vdso.so.1 (0x00007ffd965f1000)
        libintl.so.8 => not found
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f146c828000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f146c7fb000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f146c616000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f146c872000)
```

lu-zero, 2024-06-13:
you may try your luck using `RUSTFLAGS` for that purpose. I guess we could add those search paths in a central config.toml if it solves your problem.

vortexilation, 2024-06-14:
@lu-zero 
I thought that it was libelf issue, after i have tried to patchelf the libelf with the location of libintl.so.8 but still errror out, this might be OpenWrt's rust build process issue ?.

I have tried to include the following to einat build.rs and "rpath = true" to einat's Cargo.toml : 
```
println!("cargo:rustc-link-lib=dylib=libintl.so.8");
println!("cargo:rustc-link-search=native=/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libintl-full/lib");
println!("cargo:rustc-link-arg=-Wl,-rpath,/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/libintl-full/lib");
```

But still rustc seems to ignore it

lu-zero, 2024-06-14:
The order should be the opposite, I think. 

Check what's the rustc final invocation passing -v to cargo

vortexilation, 2024-06-14:
@lu-zero 

Here is cargo very verbose log : 
```
   Compiling einat v0.1.2 (/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3)
     Running `CARGO=/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/bin/cargo CARGO_CRATE_NAME=build_script_build CARGO_MANIFEST_DIR=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3 CARGO_PKG_AUTHORS='Huang-Huang Bao <i@eh5.me>' CARGO_PKG_DESCRIPTION='' CARGO_PKG_HOMEPAGE='' CARGO_PKG_LICENSE=GPL-2.0 CARGO_PKG_LICENSE_FILE='' CARGO_PKG_NAME=einat CARGO_PKG_README=README.md CARGO_PKG_REPOSITORY='https://github.com/EHfive/einat-ebpf' CARGO_PKG_RUST_VERSION=1.74 CARGO_PKG_VERSION=0.1.2 CARGO_PKG_VERSION_MAJOR=0 CARGO_PKG_VERSION_MINOR=1 CARGO_PKG_VERSION_PATCH=2 CARGO_PKG_VERSION_PRE='' CARGO_PRIMARY_PACKAGE=1 LD_LIBRARY_PATH=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps rustc --crate-name build_script_build --edition=2021 build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off -C overflow-checks=on --cfg 'feature="default"' --cfg 'feature="ipv6"' -C metadata=13037c25982734ec -C extra-filename=-13037c25982734ec --out-dir /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec -C linker=x86_64-openwrt-linux-gnu-gcc -C strip=symbols -L dependency=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps --extern libbpf_cargo=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-fc7173ecf7072a22.rlib -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release -L native=/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out`
warning: `toml` (lib) generated 1 warning
error: linking with `x86_64-openwrt-linux-gnu-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/bin:/home/user/works/openwrt/staging_dir/hostpkg/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/host/bin:/home/user/works/openwrt/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" VSLANG="1033" "x86_64-openwrt-linux-gnu-gcc" "-m64" "/home/user/works/openwrt/tmp/rustck84dLP/symbols.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec/build_script_build-13037c25982734ec.build_script_build.900026c213c08482-cgu.0.rcgu.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec/build_script_build-13037c25982734ec.2kuwii56gzittnew.rcgu.o" "-Wl,--as-needed" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-fc7173ecf7072a22.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_metadata-77b698b469ccad23.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libserde_json-d18e1952940c9d6c.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libitoa-c99bf860a1243733.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libryu-42733ab1749368f9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libthiserror-aa231274481bae73.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_platform-c20f7cad56b30019.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcamino-a1fd712e042bd596.rlib" " [... cut ...]
make[2]: *** [Makefile:92: /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/.built] Error 101
make[2]: Leaving directory '/home/user/works/openwrt/feeds/packages/net/einat'
time: package/feeds/packages/einat/compile#137.74#13.65#30.42
    ERROR: package/feeds/packages/einat failed to build.
make[1]: *** [package/Makefile:173: package/feeds/packages/einat/compile] Error 1
make[1]: Leaving directory '/home/user/works/openwrt'
make: *** [/home/user/works/openwrt/include/toplevel.mk:233: package/einat/compile] Error 2
```

lu-zero, 2024-06-14:
that I can see they are not present.

EHfive, 2024-06-14:
Hi, I am the author of https://github.com/EHfive/einat-ebpf.

I think there is some misleading going on here.

So both einat and its build script link against `libelf`, the dependency path for the former is `einat -> libbpf(-sys) -> libelf`, and it's `"build script of einat" -> libbpf-cargo -> libbpf(-sys) -> libelf` for the latter.

As you can see, einat does not use `libintl`, it's just dynamically linked by `libelf`, so the issue is not really on einat or cargo/rustc.

The main problem here is both `ld` or `ld-linux.so` can not find `libintl.so` used by `libelf.so`. In the case of `ld`, it has caused `ld` to report the unresolved symbol error due to `ld` 's default `--unresolved-symbols=report` policy. And in the case of `ld-linux.so`, it would cause binary linking with `libelf` thus indirectly using `libintl` fail to execute.

What need to addressed is why `libintl` has not been setup properly so both `ld` or `ld-linux.so` can locate it? Could it be rpath to `libintl` was not set properly in `libelf`? Or is there some misconfiguration in Rust build process? Note I am just guessing here as I know little about OpenWrt build system.

Another minor issue here is einat's build script is linking with target build of `libelf` instead of host build of `libelf`(in `tools` or build machine?).

EHfive, 2024-06-14:
> It seems like depend $(INTL_DEPENDS) is missing in Makefile.

You mean in openwrt-einat-ebpf's Makefile? But einat does not link to `libintl` or use its symbols.

cc @muink

muink, 2024-06-15:
> > It seems like depend $(INTL_DEPENDS) is missing in Makefile.Makefile 中似乎缺少 dependent $(INTL_DEPENDS) 。
> 
> You mean in openwrt-einat-ebpf's Makefile? But einat does not link to `libintl` or use its symbols.

I tryied, still failed.
https://github.com/muink/openwrt-einat-ebpf/compare/master...dev
And in fact `$(INTL_DEPENDS)` has been depended on by `libelf`, and there is no need to depend on it again in `einat`.
https://github.com/openwrt/openwrt/blob/22f92cce22401ef7df44357a7e6921d2cf45c840/package/libs/elfutils/Makefile#L51

EHfive, 2024-06-21:
Noticed that there is a commented

```make
# RUSTC_LDFLAGS+= -C link-args=$(INTL_LDFLAGS)
```

line in @muink 's openwrt-einat-ebpf dev branch. `RUSTC_LDFLAGS` would eventually be included into `RUSTFLAGS`, so it seems right at first glance.

However `INTL_LDFLAGS` has more than one link flags separated with space, so only the first flag in `INTL_LDFLAGS` would be passed as `link-args`.

The following change should make openwrt-einat-ebpf built again with `CONFIG_BUILD_NLS` for musl target.

```make
RUSTC_LDFLAGS+= $(foreach arg,$(INTL_LDFLAGS),-C link-arg=$(arg))
# or
RUSTC_LDFLAGS+= -C link-arg=-L$(INTL_PREFIX)/lib -C link-arg=-Wl,-rpath-link=$(INTL_PREFIX)/lib
```

The following also works as rustc would pass proper link flags for libintl it found to linker(via `cc -Wl`).

```
RUSTC_LDFLAGS+= -L$(INTL_PREFIX)/lib -lintl
```
---

einat has build script depends on host libelf, and the script build target binary that depends on target libelf.

@vortexilation build openwrt-einat-ebpf for `x86_64-unknown-linux-gnu` target on `x86_64-unknown-linux-gnu` host. And due to Cargo's lack of host and target separation for build script, https://github.com/rust-lang/cargo/issues/4423 , Cargo would link to target libelf due to share of `x86_64-unknown-linux-gnu` triple on host and target.

This means the libintl searing error would also happens on build script if host triple and target triple are same.

I guess passing the flags above can also make the build script built, as `RUSTFLAGS` only apply to compilation for target object if `--target` or `CARGO_BUILD_TARGET` is specified, this requires patching [rust-values.mk](https://github.com/openwrt/packages/blob/1d8d4b79adfffa38abb8c99ae64699d8066b9b57/lang/rust/rust-values.mk#L100) to use `CARGO_TARGET_<TARGET>_RUSTFLAGS` instead so these flags can also apply to  build script compilation. See also https://doc.rust-lang.org/cargo/reference/config.html#buildrustflags .

```diff
diff --git a/lang/rust/rust-values.mk b/lang/rust/rust-values.mk
index 534aaa0..aca2c24 100644
--- a/lang/rust/rust-values.mk
+++ b/lang/rust/rust-values.mk
@@ -97,7 +97,7 @@ CARGO_PKG_CONFIG_VARS= \
        CARGO_PROFILE_RELEASE_PANIC=unwind \
        CARGO_PROFILE_RELEASE_RPATH=false \
        CARGO_TARGET_$(subst -,_,$(call toupper,$(RUSTC_TARGET_ARCH)))_LINKER=$(TARGET_CC_NOCACHE) \
-       RUSTFLAGS="$(CARGO_RUSTFLAGS)" \
+       CARGO_TARGET_$(subst -,_,$(call toupper,$(RUSTC_TARGET_ARCH)))_RUSTFLAGS="$(CARGO_RUSTFLAGS)" \
        TARGET_CC=$(TARGET_CC_NOCACHE) \
        TARGET_CFLAGS="$(TARGET_CFLAGS) $(RUSTC_CFLAGS)"
```

Although on execution the build script of einat would probably instead link to host libelf by `ld-linux.so`. So this is just an ugly hack. I did not test this part as I am using musl SDK and I don't want to spend times on compilation of GNU toolchain builds.

muink, 2024-06-21:
https://github.com/muink/openwrt-einat-ebpf/compare/master...dev
Build passed.

vortexilation, 2024-06-21:
> ```makefile
> RUSTC_LDFLAGS+= $(foreach arg,$(INTL_LDFLAGS),-C link-arg=$(arg))
> # or
> RUSTC_LDFLAGS+= -C link-arg=-L$(INTL_PREFIX)/lib -C link-arg=-Wl,-rpath-link=$(INTL_PREFIX)/lib
> ```

> RUSTC_LDFLAGS+= -L$(INTL_PREFIX)/lib -lintl

@EHfive 
Thanks a lot for spending time debugging these, weird It seems still doesn't get passed to ld in glibc, haven't tried musl : 

```
   Compiling einat v0.1.2 (/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3)
     Running `rustc --crate-name build_script_build --edition=2021 build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off -C overflow-checks=on --cfg 'feature="default"' --cfg 'feature="ipv6"' -C metadata=c676d821f7a18d82 -C extra-filename=-c676d821f7a18d82 --out-dir /home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-c676d821f7a18d82 -C linker=x86_64-openwrt-linux-gnu-gcc -C strip=symbols -L dependency=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps --extern libbpf_cargo=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-0e08b88b74bdd281.rlib -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release -L native=/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib -L native=/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out`
error: linking with `x86_64-openwrt-linux-gnu-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/bin:/home/user/works/openwrt/staging_dir/hostpkg/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/host/bin:/home/user/works/openwrt/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" VSLANG="1033" "x86_64-openwrt-linux-gnu-gcc" "-m64" "/home/user/works/openwrt/tmp/rustcF0JBOI/symbols.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-c676d821f7a18d82/build_script_build-c676d821f7a18d82.build_script_build.47baaadd226efc23-cgu.0.rcgu.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-c676d821f7a18d82/build_script_build-c676d821f7a18d82.xlv39qanpodgvnm.rcgu.o" "-Wl,--as-needed" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-0e08b88b74bdd281.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_metadata-e8a4001291ac79c8.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libserde_json-a5b38a65cb4ef017.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libitoa-c99bf860a1243733.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libryu-42733ab1749368f9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libthiserror-86094f3a3cb5876d.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_platform-f5c5cc58210f0695.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcamino-2bffcd7edd9fbed6.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libmemmap2-73fca818d05ca4c9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_rs-63fc6c477f434b84.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libvsprintf-52f5d1afc74d566c.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf- [... cut ...]
warning: build failed, waiting for other jobs to finish...
error: failed to compile `einat v0.1.2 (/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3)`, intermediate artifacts can be found at `/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

@muink 

Are you sure haven't forgot to re-enable "CONFIG_BUILD_NLS" ?. Haven't tried with musl but glibc still error on me.

EHfive, 2024-06-21:
@vortexilation You didn't read my previous comment carefully, right? Especially the second section.

And you are not even trying to read logs from yourself or changes in @muink 's commit. As if you read, you would know that error is for build script instead if einat target binary, and @muink 's commit only covers the first change mentioned in my comment that only fixes building for musl target.

> ... built again with CONFIG_BUILD_NLS for musl target.

> ... share of x86_64-unknown-linux-gnu triple on host and target.
> This means the libintl searing error would also happens on build script if host triple and target triple are same.

Question yourself before question someone else next time.

vortexilation, 2024-06-21:
@EHfive 
I have already had read my logs or @muink changes, but maybe because typos and gotten the hypes up of @muink's succeed.

> ```diff
> diff --git a/lang/rust/rust-values.mk b/lang/rust/rust-values.mk
> index 534aaa0..aca2c24 100644
> --- a/lang/rust/rust-values.mk
> +++ b/lang/rust/rust-values.mk
> @@ -97,7 +97,7 @@ [...]

I have already previously tried adding CARGO_TARGET_x86_64_UNKNOWN_LINUX_GNU_RUSTFLAGS="-C link-args=-Wl,-rpath,$(STAGING_DIR)/usr/lib/libintl-full/lib" into rust-values.mk but it was seems getting no differences.

Let me retry it now with your exact changes as snippet above.

In Makefile, i just need to do the following ? : 
```
CARGO_RUSTFLAGS += -C link-args=-Wl,-rpath,$(STAGING_DIR)/usr/lib/libintl-full/lib
```