openwrt/packages #24386: curl: getting "curl: (55) error"

AcidSlide, 2024-06-14:
Maintainer: @krant @dibdot @stangri 
Environment: OpenWrt SNAPSHOT, r26592-83311b7470

Description:

In latest curl 8.8.0, some url's are not working (github raw files)

```
# curl --version
curl 8.8.0 (x86_64-openwrt-linux-gnu) libcurl/8.8.0 mbedTLS/3.6.0 nghttp2/1.62.1
Release-Date: 2024-05-22
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe
```

Trying: `curl https://raw.githubusercontent.com/duggytuxy/malicious_ip_addresses/main/botnets_zombies_scanner_spam_ips.txt`
Ends with: `curl: (55) Error`

I can confirm that it is still working with curl `curl 8.7.1 (aarch64-openwrt-linux-gnu) libcurl/8.7.1 mbedTLS/3.6.0 nghttp2/1.61.0`

Ra2-IFV, 2024-06-14:
duplicate of #24365

Ra2-IFV, 2024-06-14:
in short: In `Libraries>libcurl`, select openssl. mbedtls is broken

AcidSlide, 2024-06-16:
> in short: In `Libraries>libcurl`, select openssl. mbedtls is broken

Thanks.. I had to change all my router builds (got 4 different router brand/models).. didn't notice it defaulted to mbedtls

closed 2024-06-20 by 1715173329