openwrt/packages #24410: nginx 1.26.1 : failed to load the 'resty.core' module

catsimple, 2024-06-18:
# Issue template (remove lines from top till here)

Maintainer:  @Ansuel 
Environment: (put here arch, model, OpenWrt version)
x86/64, OpenWrt SNAPSHOT r26684-7ccd7dac7e / LuCI Master 24.158.03388~a6f8361

```
root@OpenWrt:~# nginx -c /etc/nginx/nginx.conf
2024/06/19 02:22:17 [alert] 9372#0: failed to load the 'resty.core' module (https://github.com/openresty/lua-resty-core); ensure you are using an OpenResty release from https://openresty.org/en/download.html (reason: module 'resty.core' not found:
        no field package.preload['resty.core']
        no file './resty/core.lua'
        no file '/usr/share/luajit-2.1/resty/core.lua'
        no file '/usr/share/lua/resty/core.lua'
        no file '/usr/share/lua/resty/core/init.lua'
        no file '/usr/share/lua/resty/core.lua'
        no file '/usr/share/lua/resty/core/init.lua'
        no file './resty/core.so'
        no file '/usr/lib/lua/resty/core.so'
        no file '/usr/lib/lua/resty/core.so'
        no file '/usr/lib/lua/loadall.so'
        no file './resty.so'
        no file '/usr/lib/lua/resty.so'
        no file '/usr/lib/lua/resty.so'
        no file '/usr/lib/lua/loadall.so') in /etc/nginx/nginx.conf:73
```

catsimple, 2024-06-18:
```
root@OpenWrt:~# opkg files nginx-mod-lua-resty-core
Package nginx-mod-lua-resty-core (1.26.1-r1) is installed on root and has the following files:
/usr/lib/lua/resty/core/var.lua
/usr/lib/lua/resty/core/misc.lua
/usr/lib/lua/ngx/semaphore.lua
/usr/lib/lua/resty/core/hash.lua
/usr/lib/lua/ngx/re.lua
/usr/lib/lua/resty/core/ndk.lua
/usr/lib/lua/ngx/pipe.lua
/usr/lib/lua/resty/core/ctx.lua
/usr/lib/lua/resty/core/param.lua
/usr/lib/lua/resty/core/base64.lua
/usr/lib/lua/ngx/ssl.lua
/usr/lib/lua/resty/core/phase.lua
/usr/lib/lua/resty/core/base.lua
/usr/lib/lua/resty/core/shdict.lua
/usr/lib/lua/ngx/errlog.lua
/usr/lib/lua/resty/core/uri.lua
/usr/lib/lua/resty/core/utils.lua
/usr/lib/lua/ngx/process.lua
/usr/lib/lua/resty/core/response.lua
/usr/lib/lua/resty/core/request.lua
/usr/lib/lua/resty/core.lua
/usr/lib/lua/ngx/balancer.lua
/usr/lib/lua/resty/core/regex.lua
/usr/lib/lua/ngx/req.lua
/usr/lib/lua/resty/core/socket.lua
/usr/lib/lua/ngx/ssl/clienthello.lua
/usr/lib/lua/ngx/ocsp.lua
/usr/lib/lua/resty/core/exit.lua
/usr/lib/lua/ngx/resp.lua
/usr/lib/lua/resty/core/coroutine.lua
/usr/lib/lua/resty/core/worker.lua
/usr/lib/lua/ngx/base64.lua
/usr/lib/lua/ngx/ssl/session.lua
/usr/lib/lua/resty/core/time.lua
```

Ansuel, 2024-06-23:
@catsimple nginx.conf ?

Also this happen now? Are you correctly including the module?

catsimple, 2024-06-23:
@Ansuel 
Sorry, I built luci (uhttpd) and luci-nginx at the same time, but I only use uhttpd for luci. I found some modules in `/etc/nginx/module.d/` and I deleted them. After that, nginx started normally. However, I'm not sure if those modules were actually there before, as I didn't make any changes to nginx.conf. Nginx was working fine for a long time, but it stopped working after some updates.
I'm not currently using any of the modules, and I'm only using nginx as a reverse proxy. Could it be a conflict between luci-nginx and luci? I'm not sure.

closed 2024-06-25 by catsimple