openwrt/packages #24415: Tailscale not available in arrch64_cortex-a53

upsangel, 2024-06-19:
# Issue template (remove lines from top till here)

Maintainer: @Jan Pavlinec <jan.pavlinec1@gmail.com>
Environment: arrch64_cortex-a53 , MTK filogic, 23.05.3

Description:
https://forum.openwrt.org/t/unknown-package-tailscale-no-longer-supported/201496
```
Unknown package 'tailscale'.
Collected errors:
 * opkg_install_cmd: Cannot install package tailscale.
```

I check Index of /releases/23.05.3/packages/aarch64_cortex-a53/packages/ (openwrt.org)

there is no tailscale package,

but in Index of /releases/23.05.3/packages/aarch64_generic/packages/ (openwrt.org)

there is tailscale_1.58.2-1_aarch64_generic.ipk

jan, 2024-06-19:
Yeah, I'm really not the maintainer of this thing :D

1715173329, 2024-06-19:
Available now.

closed 2024-06-19 by 1715173329

dc-me, 2024-06-25:
It's not Available as of now, I just updated the package index, but it didn't show up.

1715173329, 2024-06-25:
So the network in buildbot was broken again.