openwrt/packages #24420: stty don't change baudrate

SapeNeCo, 2024-06-19:
Hello.
I'm using OpenWRT Linux 6.1.53 with rk3568 microchip.

I'm sure dts and dtsi are spelled correctly. U-boot is also spelled correctly. All clks are indicated correctly, pins too.

I have 1 console port and 3 UARTs on the board.

For some reason, initially all UARTs are given a speed of 9600, and if you connect via tio with a speed of 9600, then everything works correctly, but when you change the speed via stty or tio, the interface completely breaks down, the speed is not correct, and if you return the setting back to 9600, now and at this speed there is only garbage in the terminal.

brada4, 2024-06-20:
Can you help with
`ubus call system board`

SapeNeCo, 2024-06-27:
> Can you help with `ubus call system board`

What's wrong with him?
I have solved my problem and I can throw a solution here.

brada4, 2024-06-27:
You need to figure out if it is wrong in openwrt or in upstream kernel and submit patches accordingly.

SapeNeCo, 2024-06-28:
The problem turned out to be in the 8250 driver, namely in two files 8250_dw.c and 8250_port.c. I found a Github for an OpenWRT project whose 8250 driver had rockchip support. I changed some functions and created a patch, with which everything began to work perfectly and properly.

brada4, 2024-06-28:
you can find them in ubuntu and redhat, go for kernel.org and leave  your patch in ../openwrt if you need them  in openwrts lts kernels.
https://github.com/torvalds/linux/tree/master/drivers/tty/serial/8250