openwrt/packages #24443: aria2 running error

OldCoding, 2024-06-22:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: openwrt master

Description:

```
Format code blocks by wrapping them with pairs of ```
```
I run aria2 using the default luci interface, but it does not work properly. Both the root user and the aria2 user report the same error. Check the log as follows

Sat Jun 22 14:12:03 2024 daemon.info aria2: Instance "main" disabled.
Sat Jun 22 14:11:32 2024 daemon.info procd: Instance aria2::aria2.main s in a crash loop 6 crashes, 0 seconds since last crash
Sat Jun 22 14:11:32 2024 daemon.info aria2c[23104]: jail: jail (23117) exited with exit: 1
Sat Jun 22 14:11:32 2024 daemon.err aria2c[23104]:
Sat Jun 22 14:11:32 2024 daemon.err aria2c[23104]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Jun 22 14:11:32 2024 daemon.err aria2c[23104]: Exception caught
Sat Jun 22 14:11:32 2024 daemon.info aria2c[23104]: jail: exec-ing /usr/bin/aria2c
Sat Jun 22 14:11:32 2024 daemon.info aria2: Aria2 will run with user 'root'.
Sat Jun 22 14:11:29 2024 daemon.info aria2c[22957]: jail: jail (22958) exited with exit: 1
Sat Jun 22 14:11:29 2024 daemon.err aria2c[22957]:
Sat Jun 22 14:11:29 2024 daemon.err aria2c[22957]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Jun 22 14:11:29 2024 daemon.err aria2c[22957]: Exception caught
Sat Jun 22 14:11:29 2024 daemon.info aria2c[22957]: jail: exec-ing /usr/bin/aria2c
Sat Jun 22 14:11:24 2024 daemon.info aria2c[22898]: jail: jail (22899) exited with exit: 1
Sat Jun 22 14:11:24 2024 daemon.err aria2c[22898]:
Sat Jun 22 14:11:24 2024 daemon.err aria2c[22898]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Jun 22 14:11:24 2024 daemon.err aria2c[22898]: Exception caught
Sat Jun 22 14:11:23 2024 daemon.info aria2c[22898]: jail: exec-ing /usr/bin/aria2c
Sat Jun 22 14:11:18 2024 daemon.info aria2c[22829]: jail: jail (22830) exited with exit: 1
Sat Jun 22 14:11:18 2024 daemon.err aria2c[22829]:
Sat Jun 22 14:11:18 2024 daemon.err aria2c[22829]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Jun 22 14:11:18 2024 daemon.err aria2c[22829]: Exception caught
Sat Jun 22 14:11:18 2024 daemon.info aria2c[22829]: jail: exec-ing /usr/bin/aria2c
Sat Jun 22 14:11:13 2024 daemon.info aria2c[22764]: jail: jail (22767) exited with exit: 1
Sat Jun 22 14:11:13 2024 daemon.err aria2c[22764]:
Sat Jun 22 14:11:13 2024 daemon.err aria2c[22764]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Jun 22 14:11:13 2024 daemon.err aria2c[22764]: Exception caught
Sat Jun 22 14:11:13 2024 daemon.info aria2c[22764]: jail: exec-ing /usr/bin/aria2c
Sat Jun 22 14:11:08 2024 daemon.info aria2c[22679]: jail: jail (22692) exited with exit: 1
Sat Jun 22 14:11:08 2024 daemon.err aria2c[22679]:
Sat Jun 22 14:11:08 2024 daemon.err aria2c[22679]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Jun 22 14:11:08 2024 daemon.err aria2c[22679]: Exception caught
Sat Jun 22 14:11:08 2024 daemon.info aria2c[22679]: jail: exec-ing /usr/bin/aria2c
Sat Jun 22 14:11:08 2024 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda1/aria2/downloads
Sat Jun 22 14:11:08 2024 daemon.info aria2: Aria2 will run with user 'aria2'.

brada4, 2024-06-22:
Version? Hardware?

OldCoding, 2024-06-22:
> Version? Hardware?

Version：Master 24.171.69191~c06c6d1
Hardware：ARMv8，Phicomm N1
I found some methods online to get it running
1: Install libopenssl-legacy package
2: Modify /etc/init.d/aria2, add procd_add_jail_mount "/usr/lib"

        procd_add_jail "$NAME.$section" log
+       procd_add_jail_mount "/usr/lib" 
        procd_add_jail_mount "$ca_certificate" "$certificate" "$rpc_certificate" "$rpc_private_key"
        procd_add_jail_mount_rw "$dir" "$config_dir" "$log"
        procd_close_instance

brada4, 2024-06-22:
Want to make a pr with new dependency and extra mount?

OldCoding, 2024-06-23:
> Want to make a pr with new dependency and extra mount?

Yes, I need this pr to fix it, otherwise it won't boot again after the next firmware update

neheb, 2024-07-03:
```
Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
```

sounds like engines can't be loaded because of jails.

sinancetinkaya, 2025-01-14:
Issue still exists
```
Wed Jan 15 00:59:11 2025 daemon.info aria2: Aria2 will run with user 'aria2'.
Wed Jan 15 00:59:11 2025 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/disk/downloads
Wed Jan 15 00:59:11 2025 daemon.info aria2c[8767]: jail: exec-ing /usr/bin/aria2c
Wed Jan 15 00:59:11 2025 daemon.err aria2c[8767]: Exception caught
Wed Jan 15 00:59:11 2025 daemon.err aria2c[8767]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
```
```
root@router-sinan:~# aria2c -v
aria2 version 1.37.0
Copyright (C) 2006, 2019 Tatsuhiro Tsujikawa

This program is free software; you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation; either version 2 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

** Configuration **
Enabled Features: BitTorrent, GZip, HTTPS, Message Digest
Hash Algorithms: sha-1, sha-224, sha-256, sha-384, sha-512, md5, adler32
Libraries: zlib/1.3.1 OpenSSL/3.0.0o
Compiler: gcc 13.3.0
  built by  x86_64-pc-linux-gnu
  targeting aarch64-openwrt-linux-gnu
  on        Jan 14 2025 14:02:58
System: Linux 6.6.71 #0 SMP Tue Jan 14 14:02:58 2025 aarch64

Report bugs to https://github.com/aria2/aria2/issues
Visit https://aria2.github.io/
```

hoang-rio, 2025-02-10:
Here is the temporaty fix https://github.com/aria2/aria2/issues/2152#issuecomment-1848164124