openwrt/packages #24450: BPI R4 in  kernel 6.6.35 build  ruby-3.3.3 error

openips, 2024-06-23:
### Describe the bug

build BPI R4 with master git tree https://github.com/openwrt/openwrt/commit/0ed72c271b13078d2daff45349368ed80d4a1fd5
build failed with ruby-3.3.3 
log below

```
compiling prism/util/pm_newline_list.c
compiling prism/util/pm_state_stack.c
compiling prism/util/pm_string.c
compiling prism/util/pm_string_list.c
compiling prism/util/pm_strncasecmp.c
compiling prism/util/pm_strpbrk.c
compiling prism/prism.c
compiling prism_init.c
compiling yjit.c
building Rust YJIT (release mode)
touch yjit/target/release/libyjit.a
partial linking yjit/target/release/libyjit.a into yjit/target/release/libyjit.o
assembling coroutine/amd64/Context.S
compiling enc/ascii.c
compiling enc/us_ascii.c
compiling enc/unicode.c
compiling enc/utf_8.c
compiling enc/trans/newline.c
compiling ./missing/setproctitle.c
compiling ./missing/strlcat.c
compiling ./missing/strlcpy.c
compiling addr2line.c
compiling dmyenc.c
linking miniruby
/bin/sh ./tool/ifchange "--timestamp=.rbconfig.time" rbconfig.rb rbconfig.tmp
rbconfig.rb updated
generating encdb.h
encdb.h updated
generating enc.mk
making srcs under enc
make[5]: Entering directory '/LEDE/openwrt-r4/build_dir/hostpkg/ruby-3.3.3'
make[5]: Nothing to be done for 'srcs'.
make[5]: Leaving directory '/LEDE/openwrt-r4/build_dir/hostpkg/ruby-3.3.3'
generating transdb.h
transdb.h updated
compiling dln.c
compiling localeinit.c
creating verconf.h
verconf.h updated
compiling loadpath.c
builtin_binary.inc updated
e73bcc3987c9f586e5e54588cee413af188ca7cac8249fddaec25bfb123f03b6  builtin_binary.inc
compiling builtin.c
linking static-library libruby-static.a
LLVM ERROR: Invalid encoding
make[4]: *** [Makefile:318: libruby-static.a] Aborted (core dumped)
make[4]: *** Deleting file 'libruby-static.a'
make[4]: Leaving directory '/LEDE/openwrt-r4/build_dir/hostpkg/ruby-3.3.3'
make[3]: *** [Makefile:1189: /LEDE/openwrt-r4/build_dir/hostpkg/ruby-3.3.3/.built] Error 2
make[3]: Leaving directory '/LEDE/openwrt-r4/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#148.55#12.37#138.15
    ERROR: package/feeds/packages/ruby [host] failed to build.
make[2]: *** [package/Makefile:171: package/feeds/packages/ruby/host/compile] Error 1
make[2]: Leaving directory '/LEDE/openwrt-r4'
make[1]: *** [package/Makefile:167: /LEDE/openwrt-r4/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/LEDE/openwrt-r4'
make: *** [/LEDE/openwrt-r4/include/toplevel.mk:248: world] Error 2
```

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

master

### Device

BPI R4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yhl452493373, 2024-06-26:
我6.6内核用3.3.3ruby没问题。但是6.1内核就不行，最后把它恢复成3.2.2的就可以了。

openips, 2024-06-26:
> 我6.6内核用3.3.3ruby没问题。但是6.1内核就不行，最后把它恢复成3.2.2的就可以了。

X64 with kernel 6.6 build finished but Arm 64 build failed

openips, 2024-06-26:
> 我6.6内核用3.3.3ruby没问题。但是6.1内核就不行，最后把它恢复成3.2.2的就可以了。目前我的解决方案也是降级到3.2.2