openwrt/packages #24467: wpad-mesh-mbedtls broken for Comfast EW-72v2 (MT7621)

ayrstone, 2024-06-26:
Trying to set up encrypted mesh in Snapshot downloaded today. 2.4 GHz. radio on with AP and mesh interfaces, 5 GHz. radio off. Wireless config:

```
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio0.band='2g'
wireless.radio0.channel='6'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='0'
wireless.AP=wifi-iface
wireless.AP.device='radio0'
wireless.AP.network='lan'
wireless.AP.mode='ap'
wireless.AP.ssid='Booting now - do not connect'
wireless.AP.encryption='psk2'
wireless.AP.key='DoNotEvenTryToUse'
wireless.mesh=wifi-iface
wireless.mesh.device='radio0'
wireless.mesh.network='lan'
wireless.mesh.mode='mesh'
wireless.mesh.mesh_id='d7e8b340b41d54a7'
wireless.mesh.encryption='psk2'
wireless.mesh.key='xxxxxxxxxx'
wireless.mesh.hidden='1'
wireless.mesh.disabled='0'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.path='1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio1.band='5g'
wireless.radio1.channel='36'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

```
Config does correctly generate /var/run/wpa_supplicant-phy0-mesh0.conf:
```
network={
	
	ssid="d7e8b340b41d54a7"
	key_mgmt=SAE
	mode=5
	fixed_freq=1
	frequency=2437
	disable_ht40=1
	disable_vht=1
	noscan=1
	sae_password="xxxxxxxxxx"
	beacon_int=100
}
```
but generates errors in the logfile:

```
Wed Jun 26 20:43:42 2024 daemon.err wpa_supplicant[1349]: Line 8: too large mode (value=5 max_value=4)
Wed Jun 26 20:43:42 2024 daemon.err wpa_supplicant[1349]: Line 8: failed to parse mode '5'.
Wed Jun 26 20:43:42 2024 daemon.err wpa_supplicant[1349]: Line 16: failed to parse network block.
Wed Jun 26 20:43:42 2024 daemon.err wpa_supplicant[1349]: Failed to read or parse configuration '/var/run/wpa_supplicant-phy0-mesh0.conf'.

```
I tried the same config on an Alfa N2Q (ath79) and verified that it works in the same snapshot build, so it appears to be unique to this platform.

ayrstone, 2024-06-26:
Same result using wpad-mesh-wolfssl and wpad-mesh-openssl.

brada4, 2024-06-26:
> wireless.mesh.encryption='psk2'

Should be sae or none , use luci if in doubt.

ayrstone, 2024-06-27:
I had wondered about that, so I used wireless.mesh.encryption='psk2', 'psk', and 'sae' - they all yielded the same result:

root@OpenWrt:/tmp/run# cat wpa_supplicant-phy0-mesh0.conf

```
network={
	
	ssid="d7e8b340b41d54a7"
	key_mgmt=SAE
	mode=5
	fixed_freq=1
	frequency=2437
	disable_ht40=1
	disable_vht=1
	noscan=1
	sae_password="xxxxxxxxxx"
	beacon_int=100
}
```
Since it seems to be so flexible, for laughs I tried encryption='woof' and it turns out that's a synonym for 'none'

Thanks anyway!

ayrstone, 2024-06-27:
OK, I think this turned out to be a n00b problem. Searching around, I found this thread:

[https://forum.openwrt.org/t/mesh-encryption-doesnt-work-on-mi-router-3-pro/106610](https://forum.openwrt.org/t/mesh-encryption-doesnt-work-on-mi-router-3-pro/106610)

I had uninstalled wpad-basic, but not rebooted, so it remained in memory. Stupidly, I thought it would flush and reload.

After rebooting, it appears wpad-mesh-mbedtls does work... y'know, if you let it. :-)

Thanks again!

closed 2024-06-27 by ayrstone