openwrt/packages #24484: mbedtls 3.6.0 breaks wget

Ra2-IFV, 2024-06-29:
wget (`/bin/uclient-fetch`) is broken too:
```
# wget https://raw.githubusercontent.com
Downloading 'https://raw.githubusercontent.com'
Connecting to 185.199.111.133:443
Redirected to / on github.com
SSL error: SSL - Bad input parameters to function
Connection error: Connection failed
```
While this works:
```
# wget https://github.com
Downloading 'https://github.com'
Connecting to 140.82.121.3:443
Writing to 'index.html'

Download completed (237596 bytes)
```

_Originally posted by @krant in https://github.com/openwrt/packages/issues/24365#issuecomment-2171754337_

krant, 2024-07-28:
Looks like the same issues:

- https://github.com/openwrt/uclient/issues/4
- https://github.com/openwrt/openwrt/issues/15427

Neustradamus, 2024-10-25:
@Ra2-IFV, @krant: It has been fixed?
@McGiverGim has commented specified tickets.

Ra2-IFV, 2024-10-25:
> All the issues causing the failures mentioned here have been fixed in [Mbed TLS 3.6.1](https://github.com/Mbed-TLS/mbedtls/releases/tag/mbedtls-3.6.1), so I'm closing this issue. Please give the new version a try and let us know if it works for you!

It's said it was fixed in mbedtls 3.6.1, shall we try it?

Neustradamus, 2024-10-25:
@Ra2-IFV: Yes, please, can you try?

Ra2-IFV, 2024-10-27:
```
root@OpenWrt:~# curl --tlsv1.3 https://cloudflare.com/cdn-cgi/trace
fl=REDACTED
h=cloudflare.com
ip=REDACTED
ts=REDACTED
visit_scheme=https
uag=curl/8.10.1
colo=REDACTED
sliver=none
http=http/2
loc=REDACTED
tls=TLSv1.3
sni=plaintext
warp=off
gateway=off
rbi=off
kex=X25519
root@OpenWrt:~# curl -V
curl 8.10.1 (aarch64-openwrt-linux-gnu) libcurl/8.10.1 mbedTLS/3.6.2 nghttp2/1.62.1
Release-Date: 2024-09-18
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
root@OpenWrt:~#
```
So I guess that's it? should work for wget too

Ra2-IFV, 2024-10-27:
To be precise, the `wget-ssl` package does not use `mbedtls`, it relies on `openssl`. As @krant said it's actually a symlink to `uclient-fetch`, my bad
```text
root@OpenWrt:~# which wget
/usr/bin/wget
root@OpenWrt:~# ls -lh /usr/bin/wget
lrwxrwxrwx    1 root     root          REDACTED /usr/bin/wget -> /bin/uclient-fetch
root@OpenWrt:~# wget -O - https://cloudflare.com/cdn-cgi/trace | cat
Downloading 'https://cloudflare.com/cdn-cgi/trace'
Connecting to REDACTED
Writing to stdout
fl=467f126
h=cloudflare.com
ip=REDACTED
ts=REDACTED
visit_scheme=https
uag=uclient-fetch
colo=REDACTED
sliver=none
http=http/1.1
loc=REDACTED
tls=TLSv1.3
sni=plaintext
warp=off
gateway=off
rbi=off
kex=X25519

Download completed (232 bytes)
root@OpenWrt:~#
```

Ra2-IFV, 2024-10-27:
closing as fixed in mbedtls 3.6.2

closed 2024-10-27 by Ra2-IFV