openwrt/packages #24492: dockerd: Configuration in nftables breaks IPv6 routing

Fail-Safe, 2024-06-30:
# Issue template (remove lines from top till here)

Maintainer: @G-M0N3Y-2503 
Environment:
```
root@OpenWrt:~# cat /etc/os-release | grep OPENWRT
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r26816-55bda9863d"

root@OpenWrt:~# docker --version
Docker version 27.0.2, build 912c1dd
```

Description:

When the dockerd service starts (or restarts) it adds the following configuration to the nft ruleset:

```
table ip6 nat {
	chain DOCKER {
	}
}
table ip6 filter {
	chain FORWARD {
		type filter hook forward priority filter; policy drop;
		counter packets 308 bytes 34270 jump DOCKER-USER
	}

	chain DOCKER-USER {
		counter packets 695 bytes 87813 return
	}

	chain DOCKER {
	}

	chain DOCKER-ISOLATION-STAGE-1 {
		iifname "docker0" oifname != "docker0" counter packets 0 bytes 0 jump DOCKER-ISOLATION-STAGE-2
		counter packets 0 bytes 0 return
	}

	chain DOCKER-ISOLATION-STAGE-2 {
		oifname "docker0" counter packets 0 bytes 0 drop
		counter packets 0 bytes 0 return
	}
}
```

This breaks IPv6 forwarding/routing from all of my client subnets on my OpenWrt box. Also, the OpenWrt web UI indicates "Legacy rules detected" in the Firewall section while the above rules are in the ruleset.

To restore IPv6 functionality for my client subnets, I have to run `/etc/init.d/firewall stop && /etc/init.d/firewall start`. After stop & start of the firewall service, the above rules are no longer in the nftables ruleset and IPv6 again works properly for my clients.

Is there a configuration option in my dockerd config that is incorrect or missing? Or is this a bug?

```
root@OpenWrt:~# cat /etc/config/dockerd

config globals 'globals'
	option data_root '/opt/docker/'
	option log_level 'warn'
	option iptables '0'
	option remote_endpoint '0'

config firewall 'firewall'
	option device 'docker0'
	list blocked_interfaces 'wan'
```

brada4, 2024-07-01:
```
ubus call system board
```
What made you think routing is affected?

Fail-Safe, 2024-07-02:
```shell
root@OpenWrt:~# ubus call system board
{
	"kernel": "6.6.35",
	"hostname": "OpenWrt",
	"system": "Intel(R) N100",
	"model": "AZW EQ",
	"board_name": "azw-eq",
	"rootfs_type": "ext4",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r26816-55bda9863d",
		"target": "x86/64",
		"description": "OpenWrt SNAPSHOT r26816-55bda9863d"
	}
}
```

I'm not definitively saying **routing** was affected. But I could successfully ping6 the IPv6 LL address of my OpenWrt gateway (the box I provided details for above) from my client hosts. But I could not ping6 any IPv6 hops beyond this gateway while the ruleset statements in my OP were in place. As soon as I stopped and then restarted the firewall service, thus removing that ruleset, I could immediately regain IPv6 connectivity through the gateway to internet hosts.

G-M0N3Y-2503, 2024-07-02:
Looks like this issue is the root cause
https://github.com/openwrt/packages/issues/17766

brada4, 2024-07-02:
iptables-nft-save and ip6tables-nft-save are needed along with full `nft list ruleset`
Current exerpts do not list source of compat warning or forward chain defaulting to drop packets.

vincejv, 2024-07-05:
@G-M0N3Y-2503 that might be unrelated, after upgrading to Docker 27 from 26, IPv6 clients doesn't have connectivity anymore and MTR shows

```
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                                                                                               Packets               Pings
 Host                                                                                                                        Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. 2001:xxx:xxx:107c::1                                                                                                     0.0%    47    0.2   0.3   0.2   0.4   0.0
 2. (waiting for reply)
```

[image]
1 being the Lan ipv6 gateway, the Ipv6 address here in this screenshot

downgrading to Docker 26 fixes ipv6 connectivity on my clients... when checking Docker 27 changes it had major overhaul in the Ipv6 stack so there maybe some breaking changes here that needs to be investigated... [IPv6](https://docs.docker.com/engine/release-notes/27.0/#ipv6) or perhaps an update to my config, not too sure at the moment, i'll give it more reading...

vincejv, 2024-07-05:
Okay can confirm this is the cause...

I edited `/etc/init.d/dockerd` inserted on line 193
```
json_add_boolean "ip6tables" "false"
```

so it now looks like
```
  json_add_string "data-root" "${data_root}"
  json_add_string "log-level" "${log_level}"
  json_add_boolean "iptables" "${iptables}"
  json_add_boolean "ip6tables" "false"
  [ -z "${log_driver}" ] || json_add_string "log-driver" "${log_driver}"
  [ -z "${bip}" ] || json_add_string "bip" "${bip}"
```

as explained by docker's documentation, cc @Fail-Safe

Fail-Safe, 2024-07-05:
> after upgrading to Docker 27 from 26, IPv6 LAN clients doesn't have connectivity anymore and MTR shows

Yes! I knew this was a non-issue until just very recently. I didn't connect the dots that it was related to the version bump to Docker 27, though. Awesome discovery, and thank you!

> I edited `/etc/init.d/dockerd` inserted on line 193
> 
> ```
> json_add_boolean "ip6tables" "false"
> ```

Again--thank you! This worked perfectly for me as well. I realize there may be a more preferred solution in the future, but for now, this is golden. I am now able to restart my `dockerd` service without breaking IPv6 for all of my clients.

@vincejv I owe you one for finding this! 👍

Fail-Safe, 2024-07-05:
Still not sure this is the right approach in the end, but here's a quick patch to default `ip6tables` to enabled, but allow an `ip6tables` option within `/etc/config/dockerd`.

```
--- /etc/init.d/dockerd	2024-07-03 21:12:51.000000000 -0400
+++ /etc/init.d/dockerd	2024-07-05 19:37:28.291146383 -0400
@@ -150,7 +150,7 @@
 }

 process_config() {
-	local alt_config_file data_root log_level iptables bip
+	local alt_config_file data_root log_level iptables ip6tables bip

 	[ -f /etc/config/dockerd ] || {
 		# Use the daemon default configuration
@@ -172,6 +172,7 @@
 	config_get data_root globals data_root "/opt/docker/"
 	config_get log_level globals log_level "warn"
 	config_get_bool iptables globals iptables "1"
+	config_get_bool ip6tables globals ip6tables "1"

 	# Don't add these options by default
 	# omission == docker defaults
@@ -190,6 +191,7 @@
 	json_add_string "data-root" "${data_root}"
 	json_add_string "log-level" "${log_level}"
 	json_add_boolean "iptables" "${iptables}"
+	json_add_boolean "ip6tables" "${ip6tables}"
 	[ -z "${log_driver}" ] || json_add_string "log-driver" "${log_driver}"
 	[ -z "${bip}" ] || json_add_string "bip" "${bip}"
 	[ -z "${registry_mirrors}" ] || json_add_array "registry-mirrors"
```

Disabling `ip6tables` is then as easy as adding this to your `dockerd` config:
`option ip6tables '0'`

vincejv, 2024-07-06:
yeah im not sure too if it's the correct fix, not familiar with iptables/nftables, just ROS (router os) fw rules, since only the forwarding aspect is broken, IPv6 still works, SLAAC, DHCPv6, etc and when the ipv6 connection is initiated from the Openwrt router itself and docker containers which is in an ipv6 docker network

Some may want this feature if they want ipv6 in their docker containers... downside is ipv6 won't work for lan clients..

brada4, 2024-07-09:
The problem is default forward policy to drop packets as opposed to accept in ip4.

Fail-Safe, 2024-07-09:
@brada4 Noted, however I am not familiar enough with how that policy should be modified to introduce a fix. Hopefully someone more skilled in this can bring that forward.

brada4, 2024-07-09:
It is definition when ip6tables adding "forward" chain.

uosbox, 2024-07-14:
> Still not sure this is the right approach in the end, but here's a quick patch to default `ip6tables` to enabled, but allow an `ip6tables` option within `/etc/config/dockerd`.
>
> ```
> --- [...]

这个方法是有效的。
不知道是否能直接去掉iptables的依赖！用nftables
这样openwrt就很简洁了！

Fail-Safe, 2024-07-26:
Fixed by https://github.com/openwrt/packages/pull/24578. Thanks @p-w-p!

closed 2024-07-26 by Fail-Safe

jpiszcz, 2024-08-27:
I am having this exact same issue using the official docker repo with Debian stable.  The docker repo does not have /etc/init.d/dockerd, is there a known workaround for this issue within the official version?

# docker --version
Docker version 27.1.2, build d01f264

jpiszcz, 2024-08-27:
I am having the exact same issue reported here, using the official docker repository with Debian stable: Docker version 27.1.2, build d01f264.  Is there a workaround for this issue on Debian stable?

```
        chain FORWARD {
                type filter hook forward priority filter; policy drop;
                counter packets 116 bytes 8700 jump DOCKER-USER
        }
```