openwrt/packages #24498: a luci-app-acme warnning about length

dalaohuuu, 2024-07-02:
<package-name>: luci-app-acme 
<package-version>: git-25.180.44951-b47dc41
<system-version>:OpenWrt 06.29.2024 by Kiddin' / LuCI Master git-25.171.62738-37db35b
ARMv8 Processor
Here is the warnning :
**daemon.warn acme: Option "keylength" is deprecated, please use key_type (e.g., ec256, rsa2048) instead**
the package doesn't work ,

brada4, 2024-07-02:
```
ubus call  system board
opkg list-installed luci-app-acme
```
This is to report problems with openwrt.org packages and fiemwares

haa4, 2024-12-09:
```
# ubus call  system board
{
        "kernel": "5.15.162",
        "hostname": "OpenWrt",
        "system": "MediaTek MT7621 ver:1 eco:3",
        "model": "TP-Link Archer C6 v3",
        "board_name": "tplink,archer-c6-v3",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.4",
                "revision": "r24012-d8dd03c46f",
                "target": "ramips/mt7621",
                "description": "OpenWrt 23.05.4 r24012-d8dd03c46f"
        }
}
# opkg list-installed luci-app-acme
luci-app-acme - git-24.338.58779-24d8916
```

error while 
```
# Tue Dec 10 04:33:30 2024 daemon.warn acme: Option "keylength" is deprecated, please use key_type (e.g., ec256, rsa2048) instead.
```

brada4, 2024-12-10:
what are versions of acmrsh and acme common?

PerkelSimon, 2024-12-31:
> what are versions of acmrsh and acme common?

same errro on snapshot

zavaruev, 2025-01-03:
Same problem here:
{
        "kernel": "5.15.167",
        "hostname": "router86",
        "system": "Intel(R) Core(TM) i3-2100 CPU @ 3.10GHz",
        "model": "Gigabyte Technology Co., Ltd. Z68XP-UD3",
        "board_name": "gigabyte-technology-co-ltd-z68xp-ud3",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.5",
                "revision": "r24106-10cc5fcd00",
                "target": "x86/64",
                "description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
        }
}
luci-app-acme - git-24.364.71483-75d2b84

moetayuko, 2025-01-19:
@stokito I'm experiencing the same issue and found your migration patches for luci. Unfortunately, they were not backported to the 23.05 branch, so luci cannot perform the migration.

`/etc/init.d/acme` has another migration path backported to 23.05 https://github.com/openwrt/packages/commit/66894032d482625a1a7e22ba4336c6fa5dd35d26, but the maintainer forgot to bump the package version. Hence, existing users need to run `opkg install acme-common --force-reinstall` to receive the update.

tohojo, 2025-01-19:
Pushed a PKG_RELEASE bump. However, the log warning cited above comes from that commit (66894032d482625a1a7e22ba4336c6fa5dd35d26), so I don't think the commit is actually missing.