openwrt/packages #24512: perl: SIGSEGV on start in musl's strlen() called by Perl_sv_vcatpvfn_flags()

egorenar, 2024-07-04:
Maintainer: Esaaprilia Salsabila
Environment: ARMv7 Processor rev 4 (v7l), custom OpenWRT fw

Description:
* `perl` always crashes on startup after upgrade

GDB output:
```
Reading symbols from ./build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/perl/perl-5.40.0/perl...
0xb6fe69c8 in _dlstart () from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/lib/ld-musl-armhf.so.1
=> 0xb6fe69c8 <_dlstart+0>:     e3a0b000        mov     r11, #0
(gdb) r
Starting program: /home/egorenar/Repositories/openwrt-rel/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/perl/perl-5.40.0/perl

Program received signal SIGSEGV, Segmentation fault.
0xb6fd8d18 in strlen (s=<optimized out>) at src/string/strlen.c:17
17              for (w = (const void *)s; !HASZERO(*w); w++);
   0xb6fd8d0c <strlen+52>:      e308e080        movw    lr, #32896      @ 0x8080
   0xb6fd8d10 <strlen+56>:      e34f4efe        movt    r4, #65278      @ 0xfefe
   0xb6fd8d14 <strlen+60>:      e348e080        movt    lr, #32896      @ 0x8080
=> 0xb6fd8d18 <strlen+64>:      e593c000        ldr     r12, [r3]
   0xb6fd8d1c <strlen+68>:      e1a02003        mov     r2, r3
   0xb6fd8d20 <strlen+72>:      e2833004        add     r3, r3, #4
   0xb6fd8d24 <strlen+76>:      e08c1004        add     r1, r12, r4
   0xb6fd8d28 <strlen+80>:      e1c1100c        bic     r1, r1, r12
   0xb6fd8d2c <strlen+84>:      e111000e        tst     r1, lr
   0xb6fd8d30 <strlen+88>:      0afffff8        beq     0xb6fd8d18 <strlen+64>
   0xb6fd8d34 <strlen+92>:      e1a03002        mov     r3, r2
(gdb) bt
#0  0xb6fd8d18 in strlen (s=<optimized out>) at src/string/strlen.c:17
#1  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#2  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#3  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#4  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#5  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#6  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#7  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#8  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#9  0xb6e3265c in Perl_sv_vcatpvfn_flags ()
   from /home/egorenar/Repositories/openwrt-rel/scripts/../staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-alpine/usr/lib/perl5/5.40/CORE/libperl.so
#10 0xb6e3265c in Perl_sv_vcatpvfn_flags ()
```

brada4, 2024-07-04:
```
gdb> info reg # r3 is from....
gdb> info proc mappings # same offsets look like excess recursion -> stack depletion -> ??? overflow to trap page ???

egorenar, 2024-07-04:
> info proc mappings

thanks for the hints!

```
Program received signal SIGSEGV, Segmentation fault.
0xb6fd8d18 in strlen (s=<optimized out>) at src/string/strlen.c:17
17              for (w = (const void *)s; !HASZERO(*w); w++);
   0xb6fd8d0c <strlen+52>:      e308e080        movw    lr, #32896      @ 0x8080
   0xb6fd8d10 <strlen+56>:      e34f4efe        movt    r4, #65278      @ 0xfefe
   0xb6fd8d14 <strlen+60>:      e348e080        movt    lr, #32896      @ 0x8080
=> 0xb6fd8d18 <strlen+64>:      e593c000        ldr     r12, [r3]
   0xb6fd8d1c <strlen+68>:      e1a02003        mov     r2, r3
   0xb6fd8d20 <strlen+72>:      e2833004        add     r3, r3, #4
   0xb6fd8d24 <strlen+76>:      e08c1004        add     r1, r12, r4
   0xb6fd8d28 <strlen+80>:      e1c1100c        bic     r1, r1, r12
   0xb6fd8d2c <strlen+84>:      e111000e        tst     r1, lr
   0xb6fd8d30 <strlen+88>:      0afffff8        beq     0xb6fd8d18 <strlen+64>
   0xb6fd8d34 <strlen+92>:      e1a03002        mov     r3, r2
(gdb) info reg
r0             0x5c8               0x5c8
r1             0xbefffbe0          0xbefffbe0
r2             0x5c8               0x5c8
r3             0x5c8               0x5c8
r4             0xfefefeff          0xfefefeff
r5             0x0                 0x0
r6             0xb6e87e56          0xb6e87e56
r7             0xb6e87e5d          0xb6e87e5d
r8             0xffffffff          0xffffffff
r9             0x73                0x73
r10            0x0                 0x0
r11            0x6666666           0x6666666
r12            0xb6f71c98          0xb6f71c98
sp             0xbefffa28          0xbefffa28
lr             0x80808080          0x80808080
pc             0xb6fd8d18          0xb6fd8d18 <strlen+64>
cpsr           0x60000010          0x60000010
fpscr          0x10                0x10
tpidruro       <unavailable>
(gdb) info proc mappings
process 17207
Mapped address spaces:

        Start Addr   End Addr       Size     Offset  Perms   objfile
           0x10000    0x11000     0x1000        0x0  r-xp   /usr/bin/perl5.40.0
           0x11000    0x12000     0x1000        0x0  r--p   /usr/bin/perl5.40.0
           0x12000    0x13000     0x1000        0x0  rw-p   [heap]
           0x13000    0x14000     0x1000        0x0  ---p   [heap]
           0x14000    0x15000     0x1000        0x0  rw-p   [heap]
        0xb6c87000 0xb6ca3000    0x1c000        0x0  rw-p
        0xb6ca3000 0xb6cae000     0xb000        0x0  r-xp   /lib/libgcc_s.so.1
        0xb6cae000 0xb6caf000     0x1000     0xa000  r--p   /lib/libgcc_s.so.1
        0xb6caf000 0xb6cb0000     0x1000     0xb000  rw-p   /lib/libgcc_s.so.1
        0xb6cb0000 0xb6f69000   0x2b9000        0x0  r-xp   /usr/lib/perl5/5.40/CORE/libperl.so
        0xb6f69000 0xb6f72000     0x9000   0x2b9000  r--p   /usr/lib/perl5/5.40/CORE/libperl.so
        0xb6f72000 0xb6f73000     0x1000   0x2c2000  rw-p   /usr/lib/perl5/5.40/CORE/libperl.so
        0xb6f73000 0xb6f7a000     0x7000        0x0  rw-p
        0xb6f7a000 0xb6f7b000     0x1000        0x0  r-xp   [sigpage]
        0xb6f7b000 0xb6f7c000     0x1000        0x0  r--p   [vvar]
        0xb6f7c000 0xb6f7d000     0x1000        0x0  r-xp   [vdso]
        0xb6f7d000 0xb6ffc000    0x7f000        0x0  r-xp   /lib/libc.so
        0xb6ffc000 0xb6ffe000     0x2000    0x7f000  rw-p   /lib/libc.so
        0xb6ffe000 0xb7000000     0x2000        0x0  rw-p
        0xbefdf000 0xbf000000    0x21000        0x0  rw-p   [stack]
        0xffff0000 0xffff1000     0x1000        0x0  r-xp   [vectors]
(gdb)
```

I will try to obtain more information like suggested.

brada4, 2024-07-04:
The r3 does not look like a pointer totally....
`        0xb6ffc000 0xb6ffe000     0x2000    0x7f000  rw-p   /lib/libc.so `
Is in libc.

More excersises:
Confirm that perl function in backtrace  is in libperl mapping
Set breakpoint `gdb> b Perl_sv_vcatpvfn_flags` then once it is hit `c`ontinue and check registers -> what is in R3, but should be only SP changing until runs out of stack.

egorenar, 2024-07-04:
More useful backtrace with `-g -O0`:

```
Program received signal SIGSEGV, Segmentation fault.
0xb6fd8d18 in strlen (s=<optimized out>, s@entry=0x5c8 <error: Cannot access memory at address 0x5c8>) at src/string/strlen.c:17
17              for (w = (const void *)s; !HASZERO(*w); w++);
   0xb6fd8d0c <strlen+52>:      e308e080        movw    lr, #32896      @ 0x8080
   0xb6fd8d10 <strlen+56>:      e34f4efe        movt    r4, #65278      @ 0xfefe
   0xb6fd8d14 <strlen+60>:      e348e080        movt    lr, #32896      @ 0x8080
=> 0xb6fd8d18 <strlen+64>:      e593c000        ldr     r12, [r3]
   0xb6fd8d1c <strlen+68>:      e1a02003        mov     r2, r3
   0xb6fd8d20 <strlen+72>:      e2833004        add     r3, r3, #4
   0xb6fd8d24 <strlen+76>:      e08c1004        add     r1, r12, r4
   0xb6fd8d28 <strlen+80>:      e1c1100c        bic     r1, r1, r12
   0xb6fd8d2c <strlen+84>:      e111000e        tst     r1, lr
   0xb6fd8d30 <strlen+88>:      0afffff8        beq     0xb6fd8d18 <strlen+64>
   0xb6fd8d34 <strlen+92>:      e1a03002        mov     r3, r2
(gdb) bt
#0  0xb6fd8d18 in strlen (s=<optimized out>, s@entry=0x5c8 <error: Cannot access memory at address 0x5c8>) at src/string/strlen.c:17
#1  0xb6e3265c in Perl_sv_vcatpvfn_flags (my_perl=my_perl@entry=0xb6ca1020, sv=sv@entry=0xb6c9ea40, pat=pat@entry=0xb6e87e48 "%s: %: panic: %s%s%s\n", patlen=patlen@entry=0x15, args=<optimized out>,
    args@entry=0xbefffbe0, svargs=<optimized out>, svargs@entry=0xb6e76764 <Perl_vmess+84>, sv_count=<optimized out>, sv_count@entry=0xbefffbe0, maybe_tainted=<optimized out>,
    maybe_tainted@entry=0x0, flags=<optimized out>, flags@entry=0x0) at sv.c:12669
#2  0xb6e4984c in Perl_sv_vsetpvfn (my_perl=my_perl@entry=0xb6ca1020, sv=sv@entry=0xb6c9ea40, pat=pat@entry=0xb6e87e48 "%s: %: panic: %s%s%s\n", patlen=0x15, args=args@entry=0xbefffbe0,
    svargs=svargs@entry=0x0, sv_count=sv_count@entry=0x0, maybe_tainted=maybe_tainted@entry=0x0) at sv.c:11295
#3  0xb6e76764 in Perl_vmess (my_perl=my_perl@entry=0xb6ca1020, pat=0xb6e87e48 "%s: %: panic: %s%s%s\n", pat@entry=0xb6ca1020 "`\300\377\266", args=0xbefffbe0, args@entry=0xbefffbc0) at util.c:1684
#4  0xb6e75ee0 in Perl_vcroak (my_perl=my_perl@entry=0xb6ca1020, pat=pat@entry=0xb6ca1020 "`\300\377\266", args=args@entry=0xbefffbc0) at util.c:1899
#5  0xb6e76af8 in Perl_croak (my_perl=my_perl@entry=0xb6ca1020, pat=0xb6e87e48 "%s: %: panic: %s%s%s\n") at util.c:1952
#6  0xb6d3e38c in Perl_locale_panic (msg=0xb6ffffb0 "'C.UTF-8;C;C;C;C;C' needs an '=' to split name=value\n", immediate_caller_line=immediate_caller_line@entry=0x5c8,
    higher_caller_file=<optimized out>, higher_caller_line=0x1001, higher_caller_line@entry=0xb6d40b98) at locale.c:1113
#7  0xb6d40a34 in S_parse_LC_ALL_string (my_perl=0xb6e880c8, my_perl@entry=0xb6ca1020, string=string@entry=0xb6fffcf0 "C.UTF-8;C;C;C;C;C", output=0xb6ffff98, output@entry=0xbefffc84,
    always_use_full_array=0x20, always_use_full_array@entry=0x1, panic_on_error=0x1, caller_line=0x1001, override=override_if_ignored) at locale.c:1480
#8  0xb6d40b98 in S_new_LC_ALL (my_perl=my_perl@entry=0xb6ca1020, lc_all=lc_all@entry=0xb6fffcf0 "C.UTF-8;C;C;C;C;C", force=force@entry=0x1) at locale.c:4090
#9  0xb6d439b8 in S_give_perl_locale_control (caller_line=0x23fd, lc_all_string=0xb6fffcf0 "C.UTF-8;C;C;C;C;C", my_perl=0xb6ca1020) at locale.c:8575
#10 Perl_init_i18nl10n (my_perl=0xb6ca1020, printwarn=printwarn@entry=0x1) at locale.c:9213
#11 0xb6cfc95c in perl_construct (my_perl=<optimized out>) at perl.c:464
#12 0x0001091c in main (argc=<optimized out>, argv=<optimized out>, env=<optimized out>) at perlmain.c:102
(gdb)
```

brada4, 2024-07-04:
Try to set `LC_ALL=C` and see if backtrace is same or it goes through. I am no expert in perl internals, that comes just from superfluus look at the text messages.

egorenar, 2024-07-04:
> LC_ALL

Yes, i noticed that too, Perl seems to crash while parsing LC_ALL :rofl: 

The problem disappears with `export LC_ALL=C` or `export LC_ALL=C`.

egorenar, 2024-07-04:
I think something goes wrong in `locale.c` (generated file ???) of Perl

```
    /* diag_listed_as: panic: %s */
    Perl_croak(aTHX_ "%s: %" LINE_Tf ": panic: %s%s%s\n",
                     __FILE__, immediate_caller_line,
                     msg, errno_text, called_by);
```

and from `handy.h`

```
#define LINE_Tf  U32uf
```

It's a typical format buffer error. `immediate_caller_line` (0x5c8 in backtrace) is interpreted as `%s`.

egorenar, 2024-07-04:
Look what i found while re-compiling Perl:

```
locale.c: In function 'Perl_locale_panic':
locale.c:1089:37: warning: unknown conversion type character '\x0a' in format [-Wformat=]
 1089 |         called_by = Perl_form(aTHX_ "\nCalled by %s: %" LINE_Tf "\n",
      |                                     ^~~~~~~~~~~~~~~~~~~
locale.c:1089:66: note: format string is defined here
 1089 |         called_by = Perl_form(aTHX_ "\nCalled by %s: %" LINE_Tf "\n",
      |                                                                  ^~
locale.c:1089:37: warning: too many arguments for format [-Wformat-extra-args]
 1089 |         called_by = Perl_form(aTHX_ "\nCalled by %s: %" LINE_Tf "\n",
      |                                     ^~~~~~~~~~~~~~~~~~~
locale.c:1113:22: warning: unknown conversion type character ':' in format [-Wformat=]
 1113 |     Perl_croak(aTHX_ "%s: %" LINE_Tf ": panic: %s%s%s\n",
      |                      ^~~~~~~
locale.c:1113:39: note: format string is defined here
 1113 |     Perl_croak(aTHX_ "%s: %" LINE_Tf ": panic: %s%s%s\n",
      |                                       ^
locale.c:1113:22: warning: format '%s' expects argument of type 'char *', but argument 4 has type 'line_t' {aka 'long unsigned int'} [-Wformat=]
 1113 |     Perl_croak(aTHX_ "%s: %" LINE_Tf ": panic: %s%s%s\n",
      |                      ^~~~~~~
 1114 |                      __FILE__, immediate_caller_line,
      |                                ~~~~~~~~~~~~~~~~~~~~~
      |                                |
      |                                line_t {aka long unsigned int}
locale.c:1113:49: note: format string is defined here
 1113 |     Perl_croak(aTHX_ "%s: %" LINE_Tf ": panic: %s%s%s\n",
      |                                                ~^
      |                                                 |
      |                                                 char *
      |                                                %ld
locale.c:1113:22: warning: too many arguments for format [-Wformat-extra-args]
 1113 |     Perl_croak(aTHX_ "%s: %" LINE_Tf ": panic: %s%s%s\n",
      |                      ^~~~~~~
In file included from locale.c:383:
locale.c: In function 'S_bool_setlocale_2008_i':
locale.c:2571:39: warning: unknown conversion type character ')' in format [-Wformat=]
 2571 |         locale_panic_(Perl_form(aTHX_ "(called from %" LINE_Tf "):"
      |                                       ^~~~~~~~~~~~~~~~
perl.h:7525:56: note: in definition of macro 'locale_panic_via_'
 7525 | #define locale_panic_via_(m, f, l)  Perl_locale_panic((m), __LINE__, f, l)
      |                                                        ^
locale.c:2571:9: note: in expansion of macro 'locale_panic_'
 2571 |         locale_panic_(Perl_form(aTHX_ "(called from %" LINE_Tf "):"
      |         ^~~~~~~~~~~~~
locale.c:2571:54: note: format string is defined here
 2571 |         locale_panic_(Perl_form(aTHX_ "(called from %" LINE_Tf "):"
      |                                                      ^
locale.c:2571:39: warning: too many arguments for format [-Wformat-extra-args]
 2571 |         locale_panic_(Perl_form(aTHX_ "(called from %" LINE_Tf "):"
      |                                       ^~~~~~~~~~~~~~~~
perl.h:7525:56: note: in definition of macro 'locale_panic_via_'
 7525 | #define locale_panic_via_(m, f, l)  Perl_locale_panic((m), __LINE__, f, l)
      |                                                        ^
locale.c:2571:9: note: in expansion of macro 'locale_panic_'
 2571 |         locale_panic_(Perl_form(aTHX_ "(called from %" LINE_Tf "):"
      |         ^~~~~~~~~~~~~
locale.c: In function 'S_setlocale_failure_panic_via_i':
locale.c:3379:56: warning: spurious trailing '%' in format [-Wformat=]
 3379 |         proxy_text = Perl_form(aTHX_ "\nCalled via %s: %" LINE_Tf,
      |                                                        ^
locale.c:3379:38: warning: too many arguments for format [-Wformat-extra-args]
 3379 |         proxy_text = Perl_form(aTHX_ "\nCalled via %s: %" LINE_Tf,
      |                                      ^~~~~~~~~~~~~~~~~~~~
locale.c:3386:58: warning: spurious trailing '%' in format [-Wformat=]
 3386 |         proxy_text = Perl_form(aTHX_ "%s\nCalled via %s: %" LINE_Tf,
      |                                                          ^
locale.c:3386:38: warning: too many arguments for format [-Wformat-extra-args]
 3386 |         proxy_text = Perl_form(aTHX_ "%s\nCalled via %s: %" LINE_Tf,
      |                                      ^~~~~~~~~~~~~~~~~~~~~~
```

This is also interesting: https://github.com/Perl/perl5/pull/21956

egorenar, 2024-07-04:
Okay, `U32uf` seems to be a Perl-specific format string specifier: https://perldoc.perl.org/5.37.3/perl5372delta#Internal-Changes

egorenar, 2024-07-04:
To me it seems that `Perl_sv_vcatpvfn_flags` doesn't support the format string specifier `U32uf`, and therefore, just skips it and goes straight for the next specifier which is `%s`, and attempts to interpret a line number as a string pointer :stuck_out_tongue_winking_eye:

egorenar, 2024-07-04:
According to Perl's devs, OpenWrt's Perl package configuration is broken :disappointed: 
`config.sh` generation is outdated and needs to be fixed for 5.40.

mauke, 2024-07-04:
The missing format string symbols have been a problem since perl 5.38, but apparently nobody noticed then.

brada4, 2024-07-04:
Helps to tag maintainer @pprindeville and recent commiter @esaaprillia

esaaprillia, 2024-07-04:
maybe there can see a change here

perl 5.38.0
https://github.com/openwrt/packages/commit/0142d0659822413a3ac5ed4c76d74cbcf5a1ca7e

perl 5.38.2
https://github.com/openwrt/packages/commit/24847159857ccd64d5d823d109a03b1908b25bd0

5.40.0
https://github.com/openwrt/packages/commit/c5b97d4d967fde1e090b90b547500abaa40cc925

esaaprillia, 2024-07-04:
maybe the reason @predators46 removed this is because powerpc_464fp powerpc_8548 riscv64_riscv64 failed while building Perl 5.38.0

```
--- a/lang/perl/files/libc.config
+++ b/lang/perl/files/libc.config
@@ -10,10 +10,6 @@
 }
 
 ($owrt:libc eq 'musl') {
-	# musl does not provide a working setlocale(). It accepts arbitrary locales
-	# and makes them act as if they were C.UTF-8.
-	d_setlocale='undef'
-	
 	d_stdio_ptr_lval='undef'
 	d_stdio_ptr_lval_sets_cnt='undef'
 	d_stdiobase='undef'
```

predators46, 2024-07-06:
@esaaprillia 

the perl package should be updated. I see many are outdated. but at that time I was too lazy to update everything. let the perl package maintainer update it

egorenar, 2024-07-06:
Okay, i dropped outdated symbols from `base.config` and added new formats to `architecture.config`
and got this error:

```
$ perl
locale.c: 1480: panic: 'C.UTF-8;C;C;C;C;C' needs an '=' to split name=value
; errno=0
Called by locale.c: 4097
```

egorenar, 2024-07-06:
Okay, found out that `d_perl_lc_all_uses_name_value_pairs='define'` which was added in commit https://github.com/openwrt/packages/commit/c5b97d4d967fde1e090b90b547500abaa40cc925.

esaaprillia, 2024-07-06:
> Okay, found out that `d_perl_lc_all_uses_name_value_pairs='define'` is responsible for the above panic which was added in commit [c5b97d4](https://github.com/openwrt/packages/commit/c5b97d4d967fde1e090b90b547500abaa40cc925).

do you succeed in building perl if you change define to undef?

egorenar, 2024-07-06:
This might give the hint why:
```
d_perl_lc_all_category_positions_init (disparate_lc_all.U):
	This symbol, if defined, indicates that $lc_all_category_positions_init
	is valid

d_perl_lc_all_separator (disparate_lc_all.U):
	This symbol, if defined, indicates that $perl_lc_all_separator is valid

d_perl_lc_all_uses_name_value_pairs (disparate_lc_all.U):
	This symbol, if defined, indicates that the string returned by
	setlocale(LC_ALL, NULL) uses 'name=value;' pairs to indicate what each
	category's locale is when they aren't all set to the same locale.
	When not defined, the platform uses positional notation.

```

egorenar, 2024-07-06:
Okay, this seems to help:

```
modified   lang/perl/files/base.config
@@ -365,9 +365,9 @@ d_open3='define'
 d_openat='define'
 d_pathconf='define'
 d_pause='define'
-d_perl_lc_all_category_positions_init='undef'
-d_perl_lc_all_separator='undef'
-d_perl_lc_all_uses_name_value_pairs='define'
+d_perl_lc_all_category_positions_init='define'
+d_perl_lc_all_separator='define'
+d_perl_lc_all_uses_name_value_pairs='undef'
 d_perl_otherlibdirs='undef'
 d_phostname='undef'
 d_pipe='define'
@@ -842,6 +842,8 @@ passcat='cat /etc/passwd'
 path_sep=':'
 perl=''
 perl5=''
+perl_lc_all_category_positions_init='{ 0, 1, 5, 2, 3, 4 }'
+perl_lc_all_separator='";"'
 perl_patchlevel=''
 perl_static_inline='static __inline__'
 perlpath='/usr/bin/perl'

```

egorenar, 2024-07-06:
I made a branch for my Perl fixes: https://github.com/egorenar/openwrt-packages/tree/perl

esaaprillia, 2024-07-06:
@egorenar 

if the problem is solved please update perl. thanks

vortexilation, 2024-07-07:
@egorenar 
Does your branch is already compile-able?, getting the following when compiling with gcc 14 : 
```
In file included from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include/features.h:503,
                 from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include/sys/types.h:25,
                 from perl.h:1117,
                 from locale.c:383:
locale.c: In function 'Perl_init_i18nl10n':
perl.h:4280:36: error: negative width in bit-field '__error_if_negative'
 4280 | #  define STATIC_ASSERT_DECL(COND) static_assert(COND, #COND)
      |                                    ^~~~~~~~~~~~~
perl.h:4296:52: note: in expansion of macro 'STATIC_ASSERT_DECL'
 4296 | #define STATIC_ASSERT_STMT(COND)      STMT_START { STATIC_ASSERT_DECL(COND); } STMT_END
      |                                                    ^~~~~~~~~~~~~~~~~~
locale.c:8813:9: note: in expansion of macro 'STATIC_ASSERT_STMT'
 8813 |         STATIC_ASSERT_STMT(   C_ARRAY_LENGTH(lc_all_category_positions)
      |         ^~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:266: locale.o] Error 1
```

egorenar, 2024-07-07:
> @egorenar Does your branch is already compile-able?, getting the following when compiling with gcc 14 targeting x86_64 glibc :
> 
> ```
> In file included from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include/features.h:503,
>                  from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include/sys/types.h:25,
>                  from perl.h:1117,
>                  from locale.c:383:
> locale.c: In function 'Perl_init_i18nl10n':
> perl.h:4280:36: error: negative width in bit-field '__error_if_negative'
>  4280 | #  define STATIC_ASSERT_DECL(COND) static_assert(COND, #COND)
>       |                                    ^~~~~~~~~~~~~
> perl.h:4296:52: note: in expansion of macro 'STATIC_ASSERT_DECL'
>  4296 | #define STATIC_ASSERT_STMT(COND)      STMT_START { STATIC_ASSERT_DECL(COND); } STMT_END
>       |                                                    ^~~~~~~~~~~~~~~~~~
> locale.c:8813:9: note: in expansion of macro 'STATIC_ASSERT_STMT'
>  8813 |         STATIC_ASSERT_STMT(   C_ARRAY_LENGTH(lc_all_category_positions)
>       |         ^~~~~~~~~~~~~~~~~~
> make[3]: *** [Makefile:266: locale.o] Error 1
> ```

Yes, but i tested only with musl.
I think `perl_lc_all_category_positions_init='{ 0, 1, 5, 2, 3, 4 }'` should be different for glibc.

egorenar, 2024-07-07:
@vortexilation Try this for `perl_lc_all_category_positions_init`
https://github.com/Perl/perl5/blob/b786e5e6e5e961082d2f48dac2a1f3ab69339b14/perl.h#L1239

vortexilation, 2024-07-07:
@egorenar 
Thanks, it compile-able now, but haven't tested functionalities wise.

Then you will also use perl_lc_all_category_positions_init for glibc as for musl?, if not how you overcome glibc and musl differences for base.config ?.

egorenar, 2024-07-07:
> Then you will also use perl_lc_all_category_positions_init for glibc as for musl?, if not how you overcome glibc and musl differences for base.config ?.

We have to put them into libc.config i think.
There are conditionals for musl and glibc.

egorenar, 2024-07-07:
**Perl**
********

LC_ positions (locale_table.h):
```
#define LC_CTYPE    0
#define LC_NUMERIC  1
#define LC_COLLATE  2
#define LC_TIME     3
#define LC_MESSAGES   4
#define LC_MONETARY  5
#define LC_ADDRESS  6
#define LC_IDENTIFICATION  7
#define LC_MEASUREMENT  8
#define LC_PAPER  9
#define LC_TELEPHONE 10
#define LC_NAME  11
#define LC_SYNTAX  12
#define LC_TOD  13
```

**musl**
********

LC_ALL categories:
```
#define LC_CTYPE    0
#define LC_NUMERIC  1
#define LC_TIME     2
#define LC_COLLATE  3
#define LC_MONETARY 4
#define LC_MESSAGES 5
```

`perl_lc_all_category_positions_init` maps libc category positions to Perl's category positions.

`perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4 }'`

egorenar, 2024-07-07:
A simple program to obtain all LC_* values:
```
#include <stdio.h>
#include <stdlib.h>
#include <locale.h>

int main()
{
#ifdef LC_CTYPE
        fprintf(stdout, "LC_CTYPE: %d\n", LC_CTYPE);
#endif
#ifdef LC_NUMERIC
        fprintf(stdout, "LC_NUMERIC: %d\n", LC_NUMERIC);
#endif
#ifdef LC_TIME
        fprintf(stdout, "LC_TIME: %d\n", LC_TIME);
#endif
#ifdef LC_COLLATE
        fprintf(stdout, "LC_COLLATE: %d\n", LC_COLLATE);
#endif
#ifdef LC_MONETARY
        fprintf(stdout, "LC_MONETARY: %d\n", LC_MONETARY);
#endif
#ifdef LC_MESSAGES
        fprintf(stdout, "LC_MESSAGES: %d\n", LC_MESSAGES);
#endif
#ifdef LC_PAPER
        fprintf(stdout, "LC_PAPER: %d\n", LC_PAPER);
#endif
#ifdef LC_NAME
        fprintf(stdout, "LC_NAME: %d\n", LC_NAME);
#endif
#ifdef LC_ADDRESS
        fprintf(stdout, "LC_ADDRESS: %d\n", LC_ADDRESS);
#endif
#ifdef LC_TELEPHONE
        fprintf(stdout, "LC_TELEPHONE: %d\n", LC_TELEPHONE);
#endif
#ifdef LC_MEASUREMENT
        fprintf(stdout, "LC_MEASUREMENT: %d\n", LC_MEASUREMENT);
#endif
#ifdef LC_IDENTIFICATION
        fprintf(stdout, "LC_IDENTIFICATION: %d\n", LC_IDENTIFICATION);
#endif
#ifdef LC_SYNTAX
        fprintf(stdout, "LC_SYNTAX: %d\n", LC_SYNTAX);
#endif
#ifdef LC_TOD
        fprintf(stdout, "LC_TOD: %d\n", LC_TOD);
#endif

        return 0;
}
```

I verified the above LC_* values on musl and glibc.

egorenar, 2024-07-07:
A small Perl program i used to verify config on musl:
```
#!/usr/bin/perl

use v5.10;
use POSIX qw(locale_h);
use locale;

my $old_locale;

$old_locale = setlocale(LC_ALL);

setlocale(LC_ALL, "0;1;2;3;4;5");

{
        my $locale = setlocale(LC_CTYPE);
        say "LC_CTYPE: $locale";
}

{
        my $locale = setlocale(LC_NUMERIC);
        say "LC_NUMERIC: $locale";
}

{
        my $locale = setlocale(LC_TIME);
        say "LC_TIME: $locale";
}

{
        my $locale = setlocale(LC_COLLATE);
        say "LC_COLLATE: $locale";
}

{
        my $locale = setlocale(LC_MONETARY);
        say "LC_MONETARY: $locale";
}

{
        my $locale = setlocale(LC_MESSAGES);
        say "LC_MESSAGES: $locale";
}

setlocale(LC_ALL, $old_locale);

exit 0;
```

vortexilation, 2024-07-07:
Thanks @egorenar , can you give me quick perl program to verfiy config on glibc?

egorenar, 2024-07-07:
> Thanks @egorenar , can you give me quick perl program to verfiy config on glibc?
> 
> [EDIT]
> 
> Latest commit for glibc seems doesn't work :
> 
> ```
> perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4, -1, 9, 11, 6, 10, 8, 7 }'
> ```
> 
> The same error as above :
> 
> ```
> r=pointer-arith -Werror=vla -Wextra -Wno-long-long -Wno-declaration-after-statement -Wc++-compat -Wwrite-strings -fPIC locale.c
> In file included from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include/features.h:503,
>                  from /home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/include/sys/types.h:25,
>                  from perl.h:1117,
>                  from locale.c:383:
> locale.c: In function 'Perl_init_i18nl10n':
> perl.h:4280:36: error: negative width in bit-field '__error_if_negative'
>  4280 | #  define STATIC_ASSERT_DECL(COND) static_assert(COND, #COND)
>       |                                    ^~~~~~~~~~~~~
> perl.h:4296:52: note: in expansion of macro 'STATIC_ASSERT_DECL'
>  4296 | #define STATIC_ASSERT_STMT(COND)      STMT_START { STATIC_ASSERT_DECL(COND); } STMT_END
>       |                                                    ^~~~~~~~~~~~~~~~~~
> locale.c:8813:9: note: in expansion of macro 'STATIC_ASSERT_STMT'
>  8813 |         STATIC_ASSERT_STMT(   C_ARRAY_LENGTH(lc_all_category_positions)
>       |         ^~~~~~~~~~~~~~~~~~
> make[3]: *** [Makefile:266: locale.o] Error 1
> ```

Hmm, okay, with glibc category 6 shouldn't be used or better to say it is mapped to LC_ALL itself, therefore, i used -1 as an invalid value. Let's go with 6 then.

vortexilation, 2024-07-07:
Hmm from your "A simple program to obtain all LC_* values:" program, I am getting the following in the debian sid glibc : 
```
LC_CTYPE: 0
LC_NUMERIC: 1
LC_TIME: 2
LC_COLLATE: 3
LC_MONETARY: 4
LC_MESSAGES: 5
LC_PAPER: 7
LC_NAME: 8
LC_ADDRESS: 9
LC_TELEPHONE: 10
LC_MEASUREMENT: 11
LC_IDENTIFICATION: 12
```

And from locale_table.h arrangement order : 
```
#define LC_CTYPE    0
#define LC_NUMERIC  1
#define LC_COLLATE  2
#define LC_TIME     3
#define LC_MESSAGES   4
#define LC_MONETARY  5
#define LC_ADDRESS  6
#define LC_IDENTIFICATION  7
#define LC_MEASUREMENT  8
#define LC_PAPER  9
#define LC_TELEPHONE 10
#define LC_NAME  11
```

Then isn't it's should be the following : 
```
perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4, 9, 12, 11, 7, 10, 8 }'
```

egorenar, 2024-07-07:
> Thanks @egorenar , can you give me quick perl program to verfiy config on glibc?

Try to run the same program first, for some reason it doesn't work on my PC and `setlocale()` fails.
I guess glibc makes sanity checks, and m,usl doesn't :/
We need a better test program.

egorenar, 2024-07-07:
> Hmm from your "A simple program to obtain all LC_* values:" program, I am getting the following in the debian sid glibc :
>
> ```
> LC_CTYPE: 0
> LC_NUMERIC: 1
> LC_TIME: 2
> LC_COLLATE: 3
> LC_MONETARY: 4 [...]

On Void Linux x86_64 with glibc, i have this output from my C program:
```
LC_CTYPE: 0
LC_NUMERIC: 1
LC_TIME: 2
LC_COLLATE: 3
LC_MONETARY: 4
LC_MESSAGES: 5
LC_PAPER: 7
LC_NAME: 8
LC_ADDRESS: 9
LC_TELEPHONE: 10
LC_MEASUREMENT: 11
LC_IDENTIFICATION: 12
```

Same as you.

Why do you map category position 6 to Perl's LC_PAPER ?
The category position 6 is reserved for LC_ALL.

vortexilation, 2024-07-07:
> The category position 6 is reserved for LC_ALL.

Can you tell me where do you get that conclusion?, I am not perl or c expert but isn't it our order of arrangement for perl_lc_all_category_positions_init is from locale_table.h definition?.

egorenar, 2024-07-07:
> perl_lc_all_category_positions_init

Assuming i understood Perl's C code correctly, `perl_lc_all_category_positions_init` is used to map C library LC_* values
to Perl's internal LC_* values (see locale.c in perl).

`perl_lc_all_category_positions_init` is a C array (indices start with 0), the index into this array is a **C library LC_ value**.
E.g. take musl `LC_TYPE(2)`, this must be mapped to Perl's `LC_TYPE(3)`. Therefore, at index 2 of the array we need to have 3.

In other words: `perl_lc_all_category_positions_init[2] = 3`

egorenar, 2024-07-07:
@vortexilation

i cannot test with glibc on OpenWrt, but glibc might work differently than musl and the original lc_all Perl config was correct. Does Perl crash on glibc w/o my fixes ?

vortexilation, 2024-07-07:
Running your "A small Perl program i used to verify config on musl:" on glibc having the following output : 
```
root@OpenWrt:~# perl locale.pl 
LC_CTYPE: C
LC_NUMERIC: C
LC_TIME: C
LC_COLLATE: C
LC_MONETARY: C
LC_MESSAGES: C
```

No crashes.

vortexilation, 2024-07-07:
```
perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4, 12, 9, 11, 6, 10, 8, 7 }'
```
Not compile-able for glibc, the same error as previously reported.

egorenar, 2024-07-07:
> ```
> perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4, 12, 9, 11, 6, 10, 8, 7 }'
> ```
> 
> Not compile-able for glibc, the same error as previously reported.

okay, i think i understand now, we need to drop the 6th entry in perl_lc_all_category_positions_init.

`perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4, 9, 11, 6, 10, 8, 7 }'`

egorenar, 2024-07-07:
> Running your "A small Perl program i used to verify config on musl:" on glibc having the following output :
> 
> ```
> root@OpenWrt:~# perl locale.pl 
> LC_CTYPE: C
> LC_NUMERIC: C
> LC_TIME: C
> LC_COLLATE: C
> LC_MONETARY: C
> LC_MESSAGES: C
> ```
> 
> No crashes.

what i meant is does perl binary (not my program) crashes if you don't apply the patch `perl: fix symbols 'd_perl_lc_all_*'` but all others applied ?

vortexilation, 2024-07-07:
> perl_lc_all_category_positions_init='{ 0, 1, 3, 2, 5, 4, 9, 11, 6, 10, 8, 7 }'

Thanks, compile-able now.

> what i meant is does perl binary (not my program) crashes if you don't apply the patch `perl: fix symbols 'd_perl_lc_all_*'` but all others applied ?

I haven't tried since it's quite time consuming for me to compile and manually deploy to my x86_64 router.

egorenar, 2024-07-07:
> Thanks, compile-able now.
>
>
> I haven't tried since it's quite time consuming for me to compile and manually deploy to my x86_64 router.

it should be quick.

```
make V=sc -j$(($(nproc)+1)) package/perl/clean
# apply changes to Perl configs in feeds/packages/lang/perl/files/
make V=sc -j$(($(nproc)+1)) package/perl/compile

# make backup of /usr/bin/perl5.40.0 and /usr/lib/perl5/5.40/CORE/libperl.so on router

scp build_dir/target-*/perl/perl-5.40.0/perl root@openwrt:/usr/bin/perl5.40.0
scp build_dir/target-*/perl/perl-5.40.0/libperl.so root@openwrt:/usr/lib/perl5/5.40/CORE/libperl.so

ssh root@openwrt
perl
```

egorenar, 2024-07-07:
I was wrong :/

`perl_lc_all_category_positions_init` maps a position in LC_ALL env var to C library LC_ value !!!

So, correct values are
for musl: `perl_lc_all_category_positions_init='{ 0, 1, 2, 3, 4, 5 }'`
for glibc: `perl_lc_all_category_positions_init='{ 0, 1, 2, 3, 4, 5, 7, 8, 9, 10, 11, 12 }'`

egorenar, 2024-07-07:
On my PC, glibc use name=value pairs in contrast to musl.
But i cannot test it with OpenWrt.

Example:
```
LC_ALL="LC_CTYPE=A;LC_NUMERIC=C;LC_TIME=C;LC_COLLATE=C;LC_MONETARY=C;LC_MESSAGES=C;LC_PAPER=C;LC_NAME=C;LC_ADDRESS=C;LC_TELEPHONE=C;LC_MEASUREMENT=C;LC_IDENTIFICATION=C"
```

egorenar, 2024-07-07:
I verified `perl_lc_all_category_positions_init='{ 0, 1, 2, 3, 4, 5 }'` for musl with a C program.

Test:
```
gcc locale.c

$ ./a.out 'C;C;C;C;C;A'
LC_CTYPE: 0
LC_CTYPE value: C
LC_NUMERIC: 1
LC_NUMERIC value: C
LC_TIME: 2
LC_TIME value: C
LC_COLLATE: 3
LC_COLLATE value: C
LC_MONETARY: 4
LC_MONETARY value: C
LC_MESSAGES: 5
LC_MESSAGES value: A
```

[locale.c.gz](https://github.com/user-attachments/files/16120332/locale.c.gz)

egorenar, 2024-07-07:
And i think for glibc the original config is correct. But somebody should verify.
```
	d_perl_lc_all_category_positions_init='undef'
	d_perl_lc_all_separator='undef'
	d_perl_lc_all_uses_name_value_pairs='define'
```

vortexilation, 2024-07-09:
@egorenar 

Running
```
./a.out "LC_CTYPE=C;LC_NUMERIC=C;LC_TIME=C;LC_COLLATE=C;LC_MONETARY=C;LC_MESSAGES=C;LC_PAPER=C;LC_NAME=C;LC_ADDRESS=C;LC_TELEPHONE=C;LC_MEASUREMENT=C;LC_IDENTIFICATION=C"

```
In live OpenWrt based glibc is getting the same result as you : 
```
LC_CTYPE: 0
LC_CTYPE value: C
LC_NUMERIC: 1
LC_NUMERIC value: C
LC_TIME: 2
LC_TIME value: C
LC_COLLATE: 3
LC_COLLATE value: C
LC_MONETARY: 4
LC_MONETARY value: C
LC_MESSAGES: 5
LC_MESSAGES value: C
LC_PAPER: 7
LC_PAPER value: C
LC_NAME: 8
LC_NAME value: C
LC_ADDRESS: 9
LC_ADDRESS value: C
LC_TELEPHONE: 10
LC_TELEPHONE value: C
LC_MEASUREMENT: 11
LC_MEASUREMENT value: C
LC_IDENTIFICATION: 12
LC_IDENTIFICATION value: C
```

Great job 👍

httpstorm, 2024-07-16:
@egorenar 

- first issue: `lc_all_string` is invalid when `LC_ALL` is not set

> I've never seen a C format string `%U32uf` 😱

- second issue
  - the `LINE_Tf` macro has an empty value. It is not `U32uf` as you suggested
  - the actual format string is `"%s: %: panic: %s%s%s\n"`. The `%:` is ignored as invalid
  - then the first `%s` takes the `immediate_caller_line` argument, where `sizeof(immediate_caller_line) = 4`, `typedef U32 line_t;`, which is guaranteed to crash when interpreted as a string on x64.
  - likely configuration issue

``` C
lc_all_string = "C.UTF-8;C;C;C;C;C"
sizeof(immediate_caller_line) = 4

// this macro is empty
#define LINE_Tf

file = "locale.c"
line = 1483
msg = "'C.UTF-8;C;C;C;C;C' needs an '=' to split name=value"
errno_text = "; errno=29"
called_by= "Called by locale.c: %"
// that % sign at the end is fishy, could be a third issue
```

esaaprillia, 2024-07-18:
fix https://github.com/openwrt/packages/pull/24614

khwilliamson, 2024-11-30:
I have skimmed this issue.  I'm sorry that you've had to expend all this effort to work around a bug in perl.  But I think the patch file in https://github.com/Perl/perl5/issues/22793 when applied to perl's Configure will fix this.  And I would appreciate someone checking this out and posting the result.  I expect that this patch will be placed in perl v5.40.1 and v5.38.3 which are in the process of being released.

predators46, 2025-02-09:
> The missing format string symbols have been a problem since perl 5.38, but apparently nobody noticed then.

after updating the perl symbols are very beautiful when built

```
2025-02-09T09:02:56.4519050Z (cd /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1 && ./Configure -S)
2025-02-09T09:02:56.5140369Z  
2025-02-09T09:02:56.5140854Z Fetching answers from config.sh...
2025-02-09T09:02:56.5149296Z  
2025-02-09T09:02:56.5149799Z Doing variable substitutions on .SH files...
2025-02-09T09:02:56.5221883Z Extracting config.h (with variable substitutions)
2025-02-09T09:02:56.7099460Z cflags.SH: Adding -std=c99.
2025-02-09T09:02:56.8884652Z cflags.SH: Adding -Werror=pointer-arith.
2025-02-09T09:02:57.0675679Z cflags.SH: Adding -Werror=vla.
2025-02-09T09:02:57.2461090Z cflags.SH: Adding -Wextra.
2025-02-09T09:02:57.6036245Z cflags.SH: Adding -Wno-long-long.
2025-02-09T09:02:57.7828708Z cflags.SH: Adding -Wno-declaration-after-statement.
2025-02-09T09:02:57.9596409Z cflags.SH: Adding -Wc++-compat.
2025-02-09T09:02:58.1421516Z cflags.SH: Adding -Wwrite-strings.
2025-02-09T09:02:58.1429934Z cflags.SH: Adding -Wno-use-after-free because of false positives in gccversion '13.3.0'
2025-02-09T09:02:58.1431229Z cflags.SH: cc       = aarch64-openwrt-linux-musl-gcc
2025-02-09T09:02:58.1433420Z cflags.SH: ccflags  = -D_REENTRANT -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -D_LARGEFILE64_SOURCE -I/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify
2025-02-09T09:02:58.1436591Z cflags.SH: stdflags =  -std=c99
2025-02-09T09:02:58.1437067Z cflags.SH: optimize = -O2
2025-02-09T09:02:58.1437789Z cflags.SH: warn     =  -Wall -Werror=pointer-arith -Werror=vla -Wextra -Wno-long-long -Wno-declaration-after-statement -Wc++-compat -Wwrite-strings -Wno-use-after-free
2025-02-09T09:02:58.1475573Z Extracting cflags (with variable substitutions)
2025-02-09T09:02:58.1525217Z Not re-extracting config.h
2025-02-09T09:02:58.1535265Z Extracting makedepend (with variable substitutions)
2025-02-09T09:02:58.1573240Z Extracting makedepend_file (with variable substitutions)
2025-02-09T09:02:58.5159557Z Extracting Makefile (with variable substitutions)
2025-02-09T09:02:58.6132350Z Extracting myconfig (with variable substitutions)
2025-02-09T09:02:58.6146411Z Extracting Policy.sh (with variable substitutions)
2025-02-09T09:02:58.6262199Z Extracting runtests (with variable substitutions)
2025-02-09T09:02:58.6469675Z Extracting pod/Makefile (with variable substitutions)
2025-02-09T09:02:58.6494256Z Extraction done.
2025-02-09T09:02:58.6500704Z install -m 0644 /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1/config.h /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1/xconfig.h
2025-02-09T09:02:58.6533064Z touch /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1/.configured_68b329da9893e34099c7d8ad5cb9c940
2025-02-09T09:02:58.6560301Z rm -f /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1/.built
2025-02-09T09:02:58.6589676Z touch /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1/.built_check
2025-02-09T09:02:58.6615717Z # make depend is required to avoid race conditions: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=996953
2025-02-09T09:02:58.6634988Z make -C /build_dir/target-aarch64_generic_musl/perl/perl-5.40.1 depend
2025-02-09T09:02:58.6680036Z make[3]: Entering directory '/build_dir/target-aarch64_generic_musl/perl/perl-5.40.1'
2025-02-09T09:02:58.6759755Z aarch64-openwrt-linux-musl-gcc -c -DPERL_CORE -D_REENTRANT -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -D_LARGEFILE64_SOURCE -I/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify -std=c99 -O2 -Wall -Werror=pointer-arith -Werror=vla -Wextra -Wno-long-long -Wno-declaration-after-statement -Wc++-compat -Wwrite-strings -Wno-use-after-free -fPIC generate_uudmap.c
2025-02-09T09:02:58.7487306Z /bin/ln -s /staging_dir/hostpkg/usr/bin/generate_uudmap generate_uudmap
2025-02-09T09:02:58.7494430Z ./generate_uudmap uudmap.h bitcount.h mg_data.h
2025-02-09T09:02:58.7503471Z sh ./makedepend MAKE="make" cflags
2025-02-09T09:02:58.7656406Z make[4]: Entering directory '/build_dir/target-aarch64_generic_musl/perl/perl-5.40.1'
2025-02-09T09:02:58.7669245Z rm -f opmini.c
2025-02-09T09:02:58.7676731Z /bin/ln -s op.c opmini.c
2025-02-09T09:02:58.7683528Z rm -f perlmini.c
2025-02-09T09:02:58.7690734Z /bin/ln -s perl.c perlmini.c
2025-02-09T09:02:58.7699086Z rm -f universalmini.c
2025-02-09T09:02:58.7706148Z /bin/ln -s universal.c universalmini.c
2025-02-09T09:02:58.7714536Z echo av.c builtin.c caretx.c class.c deb.c doio.c doop.c dquote.c dump.c globals.c gv.c hv.c keywords.c locale.c mathoms.c mg.c mro_core.c numeric.c op.c pad.c peep.c perl.c perlio.c perly.c pp.c pp_ctl.c pp_hot.c pp_pack.c pp_sort.c pp_sys.c reentr.c regcomp.c regcomp_debug.c regcomp_invlist.c regcomp_study.c regcomp_trie.c regexec.c run.c scope.c sv.c taint.c time64.c toke.c universal.c utf8.c util.c  miniperlmain.c opmini.c per [... cut ...]
2025-02-09T09:03:07.3788313Z Finding dependencies for opmini.o
2025-02-09T09:03:07.5639400Z sh ./makedepend_file perlmini.c perlmini.c.depends cflags
2025-02-09T09:03:07.5664358Z Finding dependencies for perlmini.o
2025-02-09T09:03:07.7445529Z sh ./makedepend_file universalmini.c universalmini.c.depends cflags
2025-02-09T09:03:07.7469405Z Finding dependencies for universalmini.o
2025-02-09T09:03:07.9243074Z make[4]: Leaving directory '/build_dir/target-aarch64_generic_musl/perl/perl-5.40.1'
2025-02-09T09:03:07.9327282Z Updating makefile...
```