openwrt/packages #24514: Openssh: build fails with GCC 14

glassd00r, 2024-07-05:
Openssh 9.8p1 fails to build with GCC 14. Works fine with GCC 13
Before this update, no issues with gcc 14.

Verified with latest snapshot on mediatek bpi-r4 and qualcommax ax3600

EdiTurn, 2024-07-05:
Same issue here 
````
make[2]: Entering directory '/home/editurn/openwrt/openwrt/feeds/packages/net/openssh'
rm -f /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1/.built
touch /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1/.built_check
make -C /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1 DESTDIR="/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1/ipkg-install" STRIP_OPT="" all install
make[3]: Entering directory '/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1'
Makefile:702: warning: ignoring prerequisites on suffix rule definition
(cd openbsd-compat && make)
make[4]: Entering directory '/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1/openbsd-compat'
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1=openssh-9.8p1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -pipe -Wno-error=format-truncation -Wall -Wextra -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-parameter -Wno-unused-result -Wimplicit-fallthrough -Wmisleading-indentation -fno-strict-aliasing -ftrapv -fzero-call-used-regs=used -ftrivial-auto-var-init=zero -fno-builtin-memset -fstack-protector-strong  -fzero-call-used-regs=skip -fPIC -I. -I.. -I. -I./.. -I/home/editurn/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/usr/include -I/home/editurn/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include -I/home/editurn/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify  -D_XOPEN_SOURCE=600 -D_BSD_SOURCE -D_DEFAULT_SOURCE -D_GNU_SOURCE -I/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include -DHAVE_CONFIG_H -c port-linux.c
port-linux.c: In function 'ssh_systemd_notify':
port-linux.c:369:25: error: passing argument 2 of 'connect' from incompatible pointer type [-Wincompatible-pointer-types]
  369 |         if (connect(fd, &addr, sizeof(addr)) != 0) {
      |                         ^~~~~
      |                         |
      |                         struct sockaddr_un *
In file included from /home/editurn/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/fortify/sys/socket.h:22,
                 from ../includes.h:26,
                 from port-linux.c:22:
/home/editurn/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_musl/include/sys/socket.h:386:19: note: expected 'const struct sockaddr *' but argument is of type 'struct sockaddr_un *'
  386 | int connect (int, const struct sockaddr *, socklen_t);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [Makefile:106: port-linux.o] Error 1
make[4]: Leaving directory '/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1/openbsd-compat'
make[3]: *** [Makefile:207: openbsd-compat/libopenbsd-compat.a] Error 2
make[3]: Leaving directory '/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1'
make[2]: *** [Makefile:274: /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/openssh-without-pam/openssh-9.8p1/.built] Error 2
make[2]: Leaving directory '/home/editurn/openwrt/openwrt/feeds/packages/net/openssh'
time: package/feeds/packages/openssh/without-pam/compile#0.20#0.02#0.21
    ERROR: package/feeds/packages/openssh failed to build (build variant: without-pam).
make[1]: *** [package/Makefile:173: package/feeds/packages/openssh/compile] Error 1
make[1]: Leaving directory '/home/editurn/openwrt/openwrt'
````

qosmio, 2024-07-05:
FYI:

Opened a PR for this issue.
https://github.com/openwrt/packages/pull/24516

closed 2024-07-06 by neheb