openwrt/packages #24534: Ruby 3.3.3 ERROR: package/feeds/packages/ruby [host] failed to build.

luojixinhao, 2024-07-08:
OpenWrt version
r26888-e7599942af

OpenWrt release
SNAPSHOT

OpenWrt target/subtarget
ramips/mt7621

Device
Xiaomi Mi Router 3 Pro

Image kind
Self-built image

Steps to reproduce
No response

Actual behaviour
No response

Expected behaviour
No response

Additional info
No response

Diffconfig

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-3-pro=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_IFSTAT_SNMP=y
CONFIG_KERNEL_CGROUP_DEVICE=y
# CONFIG_KERNEL_WERROR is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_LIBSODIUM_MINIMAL=y
# CONFIG_LUCI_CSSTIDY is not set
CONFIG_LUCI_LANG_zh_Hans=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_ENGINE_BUILTIN=y
CONFIG_OPENSSL_ENGINE_BUILTIN_AFALG=y
CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO=y
CONFIG_OPENSSL_OPTIMIZE_SPEED=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
[... 467 lines cut ...]
CONFIG_PERL_NOCOMMENT=y
CONFIG_PERL_THREADS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_SINGBOX_WITH_CLASH_API=y
CONFIG_SINGBOX_WITH_ECH=y
CONFIG_SINGBOX_WITH_GVISOR=y
CONFIG_SINGBOX_WITH_QUIC=y
CONFIG_SINGBOX_WITH_REALITY_SERVER=y
CONFIG_SINGBOX_WITH_UTLS=y
CONFIG_SINGBOX_WITH_WIREGUARD=y
CONFIG_SQLITE3_COLUMN_METADATA=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_RTREE=y
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
CONFIG_PACKAGE_libmbedtls=y

yhl452493373, 2024-07-09:
恢复到3.2就可以了。。

luojixinhao, 2024-07-09:
> 恢复到3.2就可以了。。

目前看来只能如此了

BKPepe, 2024-07-16:
There is missing build log, what exactly failed and without that, we can not help you. We provide support only in English.

closed 2024-07-16 by BKPepe

openips, 2024-07-16:
same here
https://github.com/openwrt/packages/issues/24450