openwrt/packages #24536: jool: Userspace version was updated; now cannot run because it no longer matches kmod version (23.05.3)

mk248, 2024-07-08:
Maintainer: @neheb

Environment: OpenWrt 23.05.3; MT7622 (Belkin RT3200); cortex-a53

Description:

After package jool-tools-netfilter was updated to version 4.1.12.1, Jool crashes upon startup with this error:
```
#service jool start
Error: The kernel module returned error 22: Version mismatch. The userspace client's version is 4.1.12.0,
but the kernel module is 4.1.8.0.
Please update the kernel module.
```
The program enforces at runtime that the kmod and the userspace control program must be the same version.  Suggest rolling back the userspace to 4.1.8.0 so that they do match.

neheb, 2024-07-08:
@mk248 I believe @hauke 's working on a new release. That should fix it.

mk248, 2024-07-09:
Thanks for replying.  I'm working on a deployment and need to solve this
within the next few days.  It looks like 23.05-SNAPSHOT contains the new
kmod so could I use that?

GoetzGoerisch, 2024-07-09:
`23.05-SNAPSHOT` works fine, running this productively.

neheb, 2024-07-20:
23.05.4 was released. Closing.

closed 2024-07-20 by neheb