openwrt/packages #24539: collectd and some dependent packages not available for aarch64_cortex-a53

nextgen-networks, 2024-07-08:
Maintainer: @jow- & @hnyman
Environment: aarch64_cortex-a53, BananaPi BPI-R4, Snapshot (put here arch, model, OpenWrt version)

Description:

Since last week collectd and some dependent packages are not available for at least "aarch64_cortex-a53" architecture:
- collectd-mod-rrdtool
- collectd-mod-iwinfo
- collectd-mod-cpu
- collectd-mod-memory
- collectd-mod-interface
- collectd-mod-load
- collectd-mod-network

as result installation of package luci-app-statistics or update of systems that have that packages installed is not possible without removing these packages.

Would be great to to have that solved.

Thanks for your fantastic and ongoing work!

hnyman, 2024-07-09:
As already discussed on the forum, the reason is the recent version upgrade of postgresql by @dangowrt .
It fails to build in buildbot and thus the dozens of packages depending on it fail too, including collectd.

Reference to 
https://github.com/openwrt/packages/pull/24327#issuecomment-2211704924
and
https://forum.openwrt.org/t/impossible-package-selection-for-luci-app-statistics/203182/16?u=hnyman

As there has been no visible progress regarding fixing postgresql, I will likely revert it's upgrade commit, as it impacts so many other packages.

nextgen-networks, 2024-07-09:
@hnyman Ohh! Thanks for the heads up and link to forum - was not aware of that situation.

hnyman, 2024-07-09:
postgresql upgrade reverted with 3dbefd299a
Collectd should be back for all targets in 1-3 days, depending on the buildbot packages round-robin schedule.

closed 2024-07-09 by hnyman