openwrt/packages #24560: libopenssl3 - Checksum or size mismatch for package libopenssl3

xxxajk, 2024-07-12:
# Issue template (remove lines from top till here)

Maintainer: @miska 
Environment: 
```
{
        "kernel": "5.15.150",
        "hostname": "OpenWrt",
        "system": "ARMv6-compatible processor rev 7 (v6l)",
        "model": "Raspberry Pi Zero W Rev 1.1",
        "board_name": "raspberrypi,model-zero-w",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.3",
                "revision": "r23809-234f1a2efa",
                "target": "bcm27xx/bcm2708",
                "description": "OpenWrt 23.05.3 r23809-234f1a2efa"
        }
}

```

Description:
Did opkg update.
Fails with:
```
 * opkg_install_pkg: Checksum or size mismatch for package libopenssl3. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-samba4.

```

xxxajk, 2024-07-12:
additionally tried manual install from https://downloads.openwrt.org/releases/23.05.3/packages/arm_arm1176jzf-s_vfp/base/libopenssl3_3.0.14-2_arm_arm1176jzf-s_vfp.ipk
still no luck.
    Size: 1.43 MiB
    MD5: 285bdd2e807c499cfc6498d0f362d5b0
    SHA256: a46b9621253424d4f531dc15e9289ed65306eec710151123f25c69daf0b11752

xxxajk, 2024-07-12:
worked with opkg -V4
strange...

Creo2005, 2024-10-26:
The same - opkg_install_pkg: Checksum or size mismatch for package libopenssl3. Either the opkg or the package index are corrupt. Try 'opkg update'.

ronlaws86, 2025-01-10:
Also having this problem. Seems very arbitrary which packages are and aren't affected. Was able to instll samba, a bunch of other stuff fine, then got around to packages htop, nano, btop and got an error. 

Using the flag ` --force-checksum ` will work around the issue and the package seems to install just fine. 

This is in OpenWRT 23.05.5 bmips/bcm6368 in my case.  (DGND3700v1)

Doublefire-Chen, 2025-08-22:
Same issue.
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package libopenssl3. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package strongswan-full.