openwrt/packages #24569: netbird > new version available

cmonty14, 2024-07-13:
Can you please update repository with latest netbird [release 0.28.4](https://github.com/netbirdio/netbird/releases/tag/v0.28.4)

Maintainer: @\oskarirauta
Environment:
mipsel_24kc
OpenWrt 23.05.3

BKPepe, 2024-07-21:
Feel free to submit pull request, if you want to have it updated. :)

wehagy, 2024-07-22:
Just to be sure, `openwrt snapshot` or `openwrt 23.05`? Because my PR is to update the package from the `openwrt snapshot`.

cmonty14, 2024-07-23:
In my case, OpenWrt 23.05

wehagy, 2024-07-23:
@BKPepe removed the fixes from my PR #24627 because won't fix the problem, my update is for the `openwrt snapshot` and the issue is to update the package for `openwrt 23.05`.

wehagy, 2024-07-23:
If you can test @cmonty14, send me your `target`, `subtarget` and `profile`, I can try to build it with `github actions`, and send you the image/ipk link for you to test, if everything is ok, I can open a PR to update in `openwrt 23.05` .

mlsmaycon, 2024-09-12:
@cmonty14 do you have some time to test with @wehagy build?

wehagy, 2025-04-15:
Can be closed as not planned:
- `OpenWrt 23.05` is approaching its end of life.
[image]
- The latest `netbird` version available for backporting is [`0.29.1`](https://github.com/netbirdio/netbird/commit/4c130a02917259fc331f32d95107009786a806f7), for more details, see PR: https://github.com/openwrt/packages/pull/25062#issue-2558413006, the same `golang` version issue occasionally affects `netbird`.
- I occasionally backport to `OpenWrt 24.10`, which is the current release, I recommend updating your `OpenWrt` version.
- Lack of interrest.

closed 2025-05-21 by BKPepe