openwrt/packages #24570: Tailscale is not in the 23.05.3 x64 repository

jphein, 2024-07-13:
tailscale is not in https://downloads.openwrt.org/releases/23.05.3/targets/x86/64/packages/
i've tried over the course of 24 hours. it's not there.

Maintainer: @p-w-p @neheb @mochaaP @1715173329 
Environment: 23.05.3 x64 
OpenWrt 23.05.3 r23809-234f1a2efa / LuCI openwrt-23.05 branch git-24.073.29889-cd7e519

Description: Tailscale is not in the 23.05.3 x64 repository

mochaaP, 2024-07-13:
<https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/packages/tailscale_1.58.2-1_x86_64.ipk>

mochaaP, 2024-07-13:
btw this version is *very* outdated and had some CVEs, although not very applicable for typical OpenWrt usage. (for the first one, since nobody actually uses openwrt as a multi-user system so this is fine; for the second one, forwarding and routing the traffic is done in fw4 instead, and tailscale already had a server side mitigation for this.)

We might want to backport whatever the latest version is, as long as our go toolchain allows. (iirc it should, i did the go1.22 port.)

mochaaP, 2024-07-13:
For unknown reasons, tailscale isn't presented in `Packages.gz`. The update should also fix this.

1715173329, 2024-07-14:
It's not unknown reason, but network issue. ~~I even thought these buildbots are located in China.~~ The package will be available again after new build completed.
Anyway, that update won't apply to 23.05 branch since tailscale devs don't bother to keep compatibility with old go toolchian.

closed 2024-07-16 by BKPepe

qdrop17, 2024-07-17:
The tailscale package build is still failing for x86_64:
https://downloads.openwrt.org/releases/faillogs-23.05/x86_64/packages/tailscale/

This package is missing from the 23.05.3 release (arch x86_64):
https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/packages/

We plan to attempt building it ourselves and submit a PR with a fix, but we'd appreciate any quicker resolution on your end.

Thank you for your efforts.

adilTepe, 2024-08-20:
I was able to use opkg to install tailscale if I use the latest x86-64 **non efi** then running `tailscale update` updated tailscale to latest version 1.70.0. But when I flash the generic x86-64 **efi**, I also experience the described nonexisting package problem.