openwrt/packages #24577: ucode 有使用发方法吗？

uosbox, 2024-07-14:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Format code blocks by wrapping them with pairs of ```
```

brada4, 2024-07-14:
English? Product versions?

BKPepe, 2024-07-16:
No reply, closing.

closed 2024-07-16 by BKPepe