openwrt/packages #24598: Python3:  apt install libreadline-dev  readline/host not found

peterwillcn, 2024-07-16:
Ubuntu 22.04 apt install libreadline-dev 
readline/host
```
The necessary bits to build these optional modules were not found:
readline
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

The following modules found by detect_modules() in setup.py have not
been built, they are *disabled* by configure:
_ctypes_test         _testbuffer          _testcapi
_testclinic          _testimportmultiple   _testinternalcapi
_testmultiphase      _xxtestfuzz

Traceback (most recent call last):
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/./setup.py", line 1621, in <module>
    main()
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/./setup.py", line 1591, in main
    setup(# PyPI Metadata (PEP 301)
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/Lib/distutils/command/build_ext.py", line 343, in run
    self.build_extensions()
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/./setup.py", line 522, in build_extensions
    self.summary(mods_built, mods_disabled)
  File "/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/Python-3.11.7/./setup.py", line 600, in summary
    raise RuntimeError("Failed to build some stdlib modules")
```

SapeNeCo, 2024-07-17:
I had exactly this problem because of the frr package: https://forum.openwrt.org/t/error-no-stdlib-modules-for-python3-11-7/204150/6

crispyoz, 2024-07-17:
I have the same issue since yesterday. Using OpenWrt 23.05.3 build system. Was working fine until yesterday, perhaps the day before.  Restored my back up build system from 7th July, same issue. Issue is triggered when I include nmap or ncat, which have always been included in my build. Selecting these packages causes python to be included. I changed the package to Python 3.11.1 (from 3.11.7), resolved this issue but created others.

/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 pyconfig.h /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/include/python3.11/pyconfig.h
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Modules/config.c /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/config.c
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 ./Modules/config.c.in /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/config.c.in
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Makefile /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/Makefile
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 ./Modules/Setup /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/Setup
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Modules/Setup.bootstrap /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/Setup.bootstrap
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Modules/Setup.stdlib /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/Setup.stdlib
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Modules/Setup.local /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/Setup.local
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Misc/python.pc /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/pkgconfig/python-3.11.pc
/home/chris/openwrt23/staging_dir/host/bin/install -c -m 644 Misc/python-embed.pc /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/pkgconfig/python-3.11-embed.pc
/home/chris/openwrt23/staging_dir/host/bin/install -c ./Modules/makesetup /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/makesetup
/home/chris/openwrt23/staging_dir/host/bin/install -c ./install-sh /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/install-sh
/home/chris/openwrt23/staging_dir/host/bin/install -c python-config.py /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/config-3.11/python-config.py
/home/chris/openwrt23/staging_dir/host/bin/install -c python-config /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/bin/python3.11-config
_PYTHON_PROJECT_BASE=/home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7 _PYTHON_HOST_PLATFORM=linux-mipsel PYTHONPATH=/home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/build/lib.linux-mipsel-3.11:./Lib _PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata__linux_ python3.11 ./setup.py install \
   	--prefix=/usr \
	--install-scripts=/usr/bin \
	--install-platlib=/usr/lib/python3.11/lib-dynload \
	--root=/home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/
running install
running build
running build_ext

The following modules found by detect_modules() in setup.py have not
been built, they are *disabled* by configure:
_ctypes_test         _testbuffer          _testcapi         
_testclinic          _testimportmultiple   _testinternalcapi 
_testmultiphase      _xxtestfuzz                            

running build_scripts
copying and adjusting /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/Tools/scripts/pydoc3 -> build/scripts-3.11
copying and adjusting /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/Tools/scripts/idle3 -> build/scripts-3.11
copying and adjusting /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/Tools/scripts/2to3 -> build/scripts-3.11
changing mode of build/scripts-3.11/pydoc3 from 644 to 755
changing mode of build/scripts-3.11/idle3 from 644 to 755
changing mode of build/scripts-3.11/2to3 from 644 to 755
renaming build/scripts-3.11/pydoc3 to build/scripts-3.11/pydoc3.11
renaming build/scripts-3.11/idle3 to build/scripts-3.11/idle3.11
renaming build/scripts-3.11/2to3 to build/scripts-3.11/2to3-3.11
running install_lib
creating /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/lib-dynload/__pycache__
copying build/lib.linux-mipsel-3.11/__pycache__/_sysconfigdata__linux_.cpython-311.pyc -> /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/lib-dynload/__pycache__
copying build/lib.linux-mipsel-3.11/_sysconfigdata__linux_.py -> /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/lib-dynload
changing mode of /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/lib-dynload/resource.cpython-311.so to 755
changing mode of /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/lib-dynload/ossaudiodev.cpython-311.so to 755
changing mode of /home/chris/openwrt23/build_dir/target-mipsel_24kc_musl/Python-3.11.7/ipkg-install/usr/lib/python3.11/lib-dynload/_statistics.cpython-311.so to 755

SapeNeCo, 2024-07-17:
I went to git packages and found these commits:
[image]
There is an assumption that it is them.

SapeNeCo, 2024-07-17:
YES!!
The problem was in this commit:
https://git.openwrt.org/?p=feed/packages.git;a=commit;h=d257ffe609e6e226d64139bb1b802cba852c6cfb
I downloaded all the packages and modified the python3 files, removing the changes from this commit.

zhanwwan, 2024-07-19:
any fixs?

BKPepe, 2024-07-31:
Ping @neheb , if he has any thoughts.

neheb, 2024-07-31:
Hmm seems I made a comment elsewhere. The issue is probably fixed after @hauke ‘s fix for the ncurses pc file. I can’t reproduce locally to be clear.

garycnew, 2025-06-04:
FYI... Almost a year later, I ran into this same issue on Debian 10 (Buster) and Debian 11 (Bullseye) building Python 3.11.10. Removing "010-no-ncursesw.patch" resolved the issue and successfully completed the build.

neheb, 2025-06-04:
yes, because it's using OS readline.

garycnew, 2025-06-05:
@neheb I suppose my question is why does the patch method using pkgconfig to locate ncurses.pc fail in some build environments?

When attempting to compile Python 3.11.10 on Debian (Bullseye), the config.log shows that the readline package cannot be built as the "Package ncurses was not found in the pkg-config search path."

<CODE>configure:23468: checking how to link readline libs
configure:23474: checking for readline
configure:23481: $PKG_CONFIG --exists --print-errors "readline"
Package ncurses was not found in the pkg-config search path.
Perhaps you should add the directory containing 'ncurses.pc'
to the PKG_CONFIG_PATH environment variable
Package 'ncurses', required by 'readline', not found
configure:23484: $? = 1
configure:23498: $PKG_CONFIG --exists --print-errors "readline"
Package ncurses was not found in the pkg-config search path.
Perhaps you should add the directory containing 'ncurses.pc'
to the PKG_CONFIG_PATH environment variable
Package 'ncurses', required by 'readline', not found
configure:23501: $? = 1
configure:23515: result: no</CODE>

I've ensured that libncurses-dev is installed:

$ sudo dpkg -l | grep -i libncurses | grep -i dev
ii  libncurses-dev:amd64                   6.2+20201114-2+deb11u2           amd64        developer's libraries for ncurses
ii  libncurses5-dev:amd64                  6.2+20201114-2+deb11u2           amd64        transitional package for libncurses-dev
ii  libncursesw5-dev:amd64                 6.2+20201114-2+deb11u2           amd64        transitional package for libncurses-dev

Ncurses.pc is available:

$ sudo dpkg -L libncurses-dev | grep -i ncurses.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/ncurses.pc

The config.log error message recommends including the ncurses.pc directory in the PKG_CONFIG_PATH environmental variable.

neheb, 2025-06-05:
that ncurses is supposed to be in

staging_dir/hostpkg/lib/pkgconfig/ncurses.pc

not the OS.

are you guys building with the SDK?

garycnew, 2025-06-05:
> make package/ncurses/host/compile

No luck.

<CODE>user@a4c806854b33:/openwrt$ make package/ncurses/host/compile
make[2]: Entering directory '/home/user/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/user/openwrt/scripts/config'
 make[1] package/ncurses/host/compile
 make[2] -C package/libs/ncurses host-compile
user@a4c806854b33:/openwrt$ ls -la staging_dir/host/lib/pkgconfig/
total 88
drwxr-xr-x 2 user user 4096 Jun  2 10:02 .
drwxr-xr-x 5 user user 4096 Jun  2 10:03 ..
-rw-r--r-- 1 user user  307 Jun  2 09:34 blkid.pc
-rw-r--r-- 1 user user  328 Jun  2 09:34 com_err.pc
-rw-r--r-- 1 user user  307 Jun  2 09:34 e2p.pc
-rw-r--r-- 1 user user  334 Jun  2 08:29 expat.pc
-rw-r--r-- 1 user user  304 Jun  2 09:34 ext2fs.pc
-rw-r--r-- 1 user user  302 Jun  2 09:59 gmp.pc
-rw-r--r-- 1 user user  337 Jun  2 09:59 gmpxx.pc
-rw-r--r-- 1 user user  351 Jun  2 08:29 libcrypto.pc
-rw-r--r-- 1 user user  747 Jun  2 09:41 libdw.pc
-rw-r--r-- 1 user user  344 Jun  2 09:41 libelf.pc
-rw-r--r-- 1 user user  479 Jun  2 08:15 liblzma.pc
-rw-r--r-- 1 user user  337 Jun  2 08:14 libpkgconf.pc
-rw-r--r-- 1 user user  353 Jun  2 08:29 libssl.pc
-rw-r--r-- 1 user user  365 Jun  2 08:29 libtls.pc
-rw-r--r-- 1 user user  497 Jun  2 08:04 libzstd.pc
-rw-r--r-- 1 user user  329 Jun  2 10:02 mpfr.pc
-rw-r--r-- 1 user user  308 Jun  2 08:29 openssl.pc
-rw-r--r-- 1 user user  330 Jun  2 09:34 ss.pc
-rw-r--r-- 1 user user  343 Jun  2 09:32 uuid.pc
-rw-r--r-- 1 user user  330 Jun  2 08:29 zlib.pc
user@a4c806854b33:/openwrt$ ls -la bin/targets/bcm53xx/generic-glibc/packages/ 
total 2420
drwxr-xr-x 3 user user    4096 Jun  3 02:48 .
drwxr-xr-x 3 user user    4096 Jun  3 02:42 ..
-rw-r--r-- 1 user user  333111 Jun  3 02:42 ldconfig_2.38-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user    3288 Jun  3 02:42 ldd_2.38-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user    7629 Jun  3 02:42 libatomic1_14.3.0-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user 1058319 Jun  3 02:42 libc_2.38-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user   22679 Jun  3 02:42 libgcc1_14.3.0-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user  258262 Jun  3 02:47 libmbedtls21_3.6.3-r1_arm_cortex-a9.ipk
-rw-r--r-- 1 user user    2380 Jun  3 02:42 libpthread_2.38-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user    2675 Jun  3 02:42 librt_2.38-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user  604927 Jun  3 02:42 libstdcpp6_14.3.0-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user   12677 Jun  3 02:42 libthread-db_2.38-r4_arm_cortex-a9.ipk
-rw-r--r-- 1 user user   10431 Jun  3 02:48 mbedtls-util_3.6.3-r1_arm_cortex-a9.ipk
drwxr-xr-x 2 user user    4096 Jun  3 02:42 tmp
-rw-r--r-- 1 user user   78614 Jun  3 02:43 zlib-dev_1.3.1-r1_arm_cortex-a9.ipk
-rw-r--r-- 1 user user   43184 Jun  3 02:43 zlib_1.3.1-r1_arm_cortex-a9.ipk</CODE>

neheb, 2025-06-05:
oh my bad. It's staging_dir/hostpkg/lib/pkgconfig/

garycnew, 2025-06-05:
> oh my bad. It's staging_dir/hostpkg/lib/pkgconfig/

It looks like the ncurses.pc file is there, now. However, it was not there, previously.

<CODE>user@a4c806854b33:~/openwrt$ ls -la staging_dir/hostpkg/lib/pkgconfig/
total 24
drwxr-xr-x 2 user user 4096 Jun  5 18:44 .
drwxr-xr-x 3 user user 4096 Jun  5 18:44 ..
-rw-r--r-- 1 user user  642 Jun  5 18:44 form.pc
-rw-r--r-- 1 user user  642 Jun  5 18:44 menu.pc
-rw-r--r-- 1 user user  634 Jun  5 18:44 ncurses.pc
-rw-r--r-- 1 user user  644 Jun  5 18:44 panel.pc</CODE>

Could this be an order of operations issue?

Thanks, again.

neheb, 2025-06-05:
So there's https://github.com/openwrt/openwrt/blob/main/package/libs/readline/Makefile#L28

But it's not working properly it seems.

garycnew, 2025-06-05:
> So there's https://github.com/openwrt/openwrt/blob/main/package/libs/readline/Makefile#L28
> 
> But it's not working properly it seems.

Hmm... Presently, do you believe the Python3 ncurses patch is failing, because of the libreadline package is not working properly?

neheb, 2025-06-05:
Correct. Edit the python patch and replace readline in the PKG_CONFIG call with ncurses and see what error you get

neheb, 2025-06-06:
also, what does ncurses.pc contain?