openwrt/packages #24633: lua-rs232: Where is the documentation?

Botster, 2024-07-20:
Maintainer: @\cotequeiroz (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)
OPENWRT_BOARD="ath79/nand"
OPENWRT_ARCH="mips_24kc"
OPENWRT_RELEASE="OpenWrt 22.03.5 r20134-5f15225c1e"

Description:

`opkg info lua-rs232` give no upstream URL to help finding the documentation.
The Sources linked at https://openwrt.org/packages/pkgdata/lua-rs232 [https://github.com/openwrt/packages/tree/openwrt-22.03/lang/lua-rs232] contain only a Makefile and Patches.  So, analyzing the source to figure out how to use it is not possible.

Where is the user documentation?

(BTW, I find a glaring absence of user documentation links for all the packages I have so far looked at.)

```
Format code blocks by wrapping them with pairs of ```
```

brada4, 2024-07-21:
I found upstream URL with documentation in the Makefile
https://github.com/srdgame/librs232

Botster, 2024-07-21:
That is very helpful, Andrew. Thank you.

Still, I think the upstream URL should be reported with `opkg info`.  But, that is more of an issue with opkg than it is with lua-rs232. So, I am closing this issue.

closed 2024-07-21 by Botster