openwrt/packages #24668: iperf3: segfault on x86_64

Djfe, 2024-07-25:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: x86_64, Hewlett-Packard HP EliteDesk 800 G1 USDT, OpenWrt GCC 12.3.0 r23899+10-3001d3c5d4
This was an OpenWrt fork: [Gluon v2023.2.3](https://github.com/freifunk-gluon/gluon/tree/v2023.2.3)
Which runs:
```
OPENWRT_REPO=https://github.com/openwrt/openwrt.git
OPENWRT_BRANCH=openwrt-23.05
OPENWRT_COMMIT=3001d3c5d446bf53fc222640d49a4cc79e04a2f3
```

Description:
I just noticed this segfault in dmesg after running a couple of iperf3 commands (I'm not sure which one triggered it or if it could've been triggered by a STRG+C by accident)
```
[459397.930023] iperf3[4064]: segfault at 28 ip 00007ff50794da40 sp 00007ffe5a472880 error 4 in libc.so[7ff507928000+4c000] [459397.930195] Code: 41 5d 41 5e c3 48 8d 35 2e f2 03 00 e9 a1 ff ff ff 90 83 c8 ff c3 83 c8 ff c3 83 c8 ff c3 41 54 55 bd 01 00 00 00 53 48 89 fb <48> 8b 43 28 48 85 c0 74 08 48 ff c5 48 89 c3 eb ef 48 0f bf 43 50 
```

Djfe, 2024-07-25:
I forgot to mention @nbd168 (had to edit the post above)

vincejv, 2024-07-27:
This seems like an upstream issue? I can reproduce it with other linux distros as well

Djfe, 2024-07-27:
how did you reproduce it? Just by running iperf3 or a special command?

vincejv, 2024-07-28:
@Djfe see upstream
https://github.com/esnet/iperf/issues/1694
https://github.com/esnet/iperf/issues/1692
https://github.com/esnet/iperf/issues/1681

> It seems that the server crash because it receives "illegal" message from the client, and somehow it doesn't know to handle it well

I get this seg fault when i abruptly stop the client from performing the iperf test

neheb, 2024-08-27:
Not fixed by https://github.com/openwrt/packages/commit/37ade7efabae535bb4612177bc4cf0a32353b8e9 ?

vincejv, 2024-08-27:
@neheb x86_64 is little endian so that fix isn't applicable

neheb, 2024-08-27:
It relates to JSON code.

Anyway, gdb is needed to figure this out.

trenchcoatjedi, 2024-11-20:
This is still an issue in current snapshots for this architecture.

`[ 1181.356729] iperf3[18191]: segfault at 28 ip 00007fa2adce41f2 sp 00007ffdc4444f40 error 4 in libc.so[7fa2adcb
d000+54000] likely on CPU 1 (core 1, socket 0)                                                                  
[ 1181.370686] Code: e9 a1 ff ff ff 90 b8 ff ff ff ff c3 b8 ff ff ff ff c3 b8 ff ff ff ff c3 41 54 55 bd 01 00 0
0 00 53 eb 07 48 83 c5 01 48 89 c7 <48> 8b 47 28 48 85 c0 75 f0 48 0f bf 47 50 48 89 fb 48 8d 14 80 48          
Segmentation fault 
# uname -a                                                                                        
Linux OpenWrt 6.6.61 #0 SMP Wed Nov 20 08:49:21 2024 x86_64 GNU/Linux 
# cat /etc/openwrt_release                                                                     
DISTRIB_ID='OpenWrt'                                                                                            
DISTRIB_RELEASE='SNAPSHOT'                                                                                      
DISTRIB_REVISION='r28141-52519a59a8'                                                                            
DISTRIB_TARGET='x86/64'                                                                                         
DISTRIB_ARCH='x86_64'                                                                                           
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28141-52519a59a8'                                                        
DISTRIB_TAINTS=''  `

neheb, 2024-11-20:
```
iperf3 -c server -P 4 -t 30
```

is crashing on my mipsel device. Will investigate.

neheb, 2024-11-20:
I'm getting

```
(gdb) bt
#0  0x77f8e06c in freeaddrinfo () from /lib/ld-musl-mipsel-sf.so.1
#1  0x77f05069 in create_socket () from /usr/lib/libiperf.so.0
```

I have a strong feeling https://github.com/esnet/iperf/commit/3da07ae96f5b40f76b75e1ccd4b20267f6a5988e fixes it

neheb, 2024-11-20:
Yep, will push.