openwrt/packages #24687: python3-cryptography is missing from powerpc_464fp repo

GammaC0de, 2024-07-28:
Package: python3-cryptography
Maintainer: Jeffery To <jeffery.to@gmail.com>
Environment: powerpc_464fp 

Description:

```
python3-cryptography is missing from https://downloads.openwrt.org/releases/23.05.4/packages/powerpc_464fp/packages/
```

BKPepe, 2024-07-31:
I looked at https://downloads.openwrt.org/releases/faillogs-23.05/powerpc_464fp/packages/, and I can not see that there is a failing for python3-cryptography. In this case, you need to work harder and help us investigate it more. :-) Output with the error would be sufficient.

GammaC0de, 2024-08-07:
I am not an expert in this field, it would be nice if you'll guide me.

What I have done is: 
```
git checkout openwrt-23.05
make distclean
./scripts/feeds update -a
./scripts/feeds install -a
wget https://mirror-03.infra.openwrt.org/releases/23.05.4/targets/apm821xx/sata/config.buildinfo  -O .config
make defconfig
make tools/install
make toolchain/install
make -j $(nproc) package/python-cryptography/compile
```
Then there was an error suggesting to do `make oldconfig && make prepare`  which I have done and retried to compile again.
After that is seems to compile OK but there is no `.ipk` file inside the `bin` directory.
Also have tried to do `make world`.
Looking at the the attached logs, there seems to be some errors.
[python-cryptography-compile1.zip](https://github.com/user-attachments/files/16532861/python-cryptography-compile1.zip)
[python-cryptography-compile2.zip](https://github.com/user-attachments/files/16532860/python-cryptography-compile2.zip)

GammaC0de, 2024-08-15:
@BKPepe Can you please check this?