openwrt/packages #24689: jool: nat64 incorrect/bad checksum for tcp4 and udp4 Packets on mpc85xx/p1010

stargieg, 2024-07-28:
Maintainer: 
Environment: mpc85xx/p1010 tplink_tl-wdr4900-v1 openwrt-23.05 and master

Description: In NAT64 mode the outgoing TCP4 and UDP4 Packets have incorrect checksum.
ICMP4 Packets have a correct checksum. It is reproducible with [NAT64 for a IPv6-only network (Jool)](https://openwrt.org/docs/guide-user/network/ipv6/nat64)

--WAN Port
```
# tcpdump -vv -n -i wan port 443
tcpdump: listening on wan, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:55:05.781478 IP (tos 0x0, ttl 63, id 38729, offset 0, flags [none], proto TCP (6), length 52) 10.230.6.111.65039 > 10.230.6.97.443: Flags [S], cksum 0x4a3a (incorrect -> 0x6cd6), seq 683981143, win 64998, options [mss 1413,nop,wscale 8,nop,nop,sackOK], length 0
# tcpdump -vv -n -i wan port 53
tcpdump: listening on wan, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:58:56.558108 IP (tos 0x0, ttl 63, id 56188, offset 0, flags [none], proto UDP (17), length 65) 10.230.6.111.61227 > 10.230.6.97.53: [bad udp cksum 0xd34c -> 0xf5e8!] 33906+ A? openwrt.localdomain. (37)
# tcpdump -vv -n -i wan icmp and host 10.230.6.97
tcpdump: listening on wan, link-type EN10MB (Ethernet), snapshot length 262144 bytes
19:01:13.227767 IP (tos 0x0, ttl 63, id 1800, offset 0, flags [none], proto ICMP (1), length 84) 10.230.6.111 > 10.230.6.97: ICMP echo request, id 1378, seq 1, length 64
```

--LAN Port
```
# tcpdump -vv -n -i br-lan port 443
tcpdump: listening on br-lan, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:55:05.781334 IP6 (flowlabel 0x04c88, hlim 64, next-header TCP (6) payload length: 32) fd43:992c:dac1:0:ac01:6e7a:8273:c4ee.20423 > 64:ff9b::ae6:661.443: Flags [S], cksum 0x5963 (correct), seq 683981143, win 64998, options [mss 1413,nop,wscale 8,nop,nop,sackOK], length 0
# tcpdump -vv -n -i br-lan port 53 and host 64:ff9b::ae6:661
tcpdump: listening on br-lan, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:58:56.557985 IP6 (flowlabel 0xf90a4, hlim 64, next-header UDP (17) payload length: 45) fd43:992c:dac1:0:ac01:6e7a:8273:c4ee.58641 > 64:ff9b::ae6:661.53: [udp sum ok] 33906+ A? openwrt.localdomain. (37)
# tcpdump -vv -n -i br-lan icmp6 and host 64:ff9b::ae6:661
tcpdump: listening on br-lan, link-type EN10MB (Ethernet), snapshot length 262144 bytes
19:01:13.227645 IP6 (hlim 64, next-header ICMPv6 (58) payload length: 64) fd43:992c:dac1:0:ac01:6e7a:8273:c4ee > 64:ff9b::ae6:661: [icmp6 sum ok] ICMP6, echo request, id 141, seq 1
```

X86_64 and mvebu/cortexa9 is working fine.

brada4, 2024-08-02:
You need to disable rx and tx using ethtool to get real checksums. Outgoing towards tx offload are not expected to be correct.

stargieg, 2024-08-02:
I don't now what you mean with rx and tx. with ethtool looks like offloading is off.

```
ethtool --show-offload eth0
Features for eth0:
rx-checksumming: off
tx-checksumming: off
	tx-checksum-ipv4: off
	tx-checksum-ip-generic: off [fixed]
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: off
	tx-scatter-gather: off
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp-mangleid-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
generic-segmentation-offload: off
generic-receive-offload: off
large-receive-offload: off [fixed]
rx-vlan-offload: off
tx-vlan-offload: off
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: on [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-tunnel-remcsum-segmentation: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
tx-udp-segmentation: off [fixed]
tx-gso-list: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off [fixed]
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
tls-hw-tx-offload: off [fixed]
tls-hw-rx-offload: off [fixed]
rx-gro-hw: off [fixed]
tls-hw-record: off [fixed]
rx-gro-list: off
macsec-hw-offload: off [fixed]
rx-udp-gro-forwarding: off
hsr-tag-ins-offload: off [fixed]
hsr-tag-rm-offload: off [fixed]
hsr-fwd-offload: off [fixed]
hsr-dup-offload: off [fixed]
```
```
ethtool --show-offload wan
Features for wan:
rx-checksumming: off [fixed]
tx-checksumming: off
	tx-checksum-ipv4: off [fixed]
	tx-checksum-ip-generic: off [fixed]
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: off
	tx-scatter-gather: off [fixed]
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp-mangleid-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
generic-segmentation-offload: off [requested on]
generic-receive-offload: off
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: on [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: on [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-tunnel-remcsum-segmentation: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
tx-udp-segmentation: off [fixed]
tx-gso-list: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
tls-hw-tx-offload: off [fixed]
tls-hw-rx-offload: off [fixed]
rx-gro-hw: off [fixed]
tls-hw-record: off [fixed]
rx-gro-list: off
macsec-hw-offload: off [fixed]
rx-udp-gro-forwarding: off
hsr-tag-ins-offload: off [fixed]
hsr-tag-rm-offload: off [fixed]
hsr-fwd-offload: off [fixed]
hsr-dup-offload: off [fixed]
```

brada4, 2024-08-03:
Could you, please, shorten your post to leave only relevant parts?

stargieg, 2024-08-03:
> Could you, please, shorten your post to leave only relevant parts?

Done.