openwrt/packages #24693: https-dns-proxy: process stuck at 100% (with curl 8.9.0)

AcidSlide, 2024-07-29:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @stangri @robimarko 
Environment: aarch64_cortex-a53, SNAPSHOT (latest as of 2024-07-29)

Description:

A process of `https-dns-proxy` gets stuck at 100% CPU usage.  I have reproduced the error and it seems it is linked to the latest update of curl (with openssl) and/or libcurl4 to 8.9.0.

I have downgraded installed version of curl (with openssl) and libcurl4 to 8.8.0 and issue doesn't happen.  Then I re-installed version 8.9.0 of curl and libcurl4 and the issue came back.  Usually after reboot about a 1-2mins this happens.  Then with a manual restart with `/etc/init.d/http-dns-proxy`, the issue will return after a couple of mins (between 5-15mins).

No issues with curl 8.8.0 (see version details below)
```
curl 8.8.0 (aarch64-openwrt-linux-gnu) libcurl/8.8.0 OpenSSL/3.0.14 nghttp2/1.62.1
Release-Date: 2024-05-22
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
```

Issues with curl 8.9.0 (see version details below)
```
curl 8.9.0 (aarch64-openwrt-linux-gnu) libcurl/8.9.0 OpenSSL/3.0.14 nghttp2/1.62.1
Release-Date: 2024-07-24
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
```

[image]

Captured HTTP-DNS Log right during when the 100% CPU usage was triggered
```
[D] 1722249181.162027 dns_poller.c:114 Starting DNS query
[D] 1722249181.162311 main.c:177 DNS server IP address unchanged (8.8.8.8,8.8.4.4).
[D] 1722249181.162318 dns_poller.c:88 DNS poll interval changed to: 120
[D] 1722249238.749286 main.c:112 Received request for id: B654, len: 68
[D] 1722249238.749323 https_client.c:253 B654: Requesting HTTP/2
[D] 1722249238.749913 https_client.c:207 B654: * TLSv1.3 (OUT), TLS alert, close notify (256):
[D] 1722249238.749956 https_client.c:603 Reserved new io event: 0x7fdf13cd48
[D] 1722249238.750097 https_client.c:79 curl opened socket: 10
[D] 1722249238.750281 https_client.c:603 Reserved new io event: 0x7fdf13cd78
[D] 1722249238.761942 https_client.c:207 B654: * TLSv1.3 (OUT), TLS handshake, Client hello (1):
[D] 1722249238.762131 https_client.c:593 Released used io event: 0x7fdf13cd78
[D] 1722249238.762136 https_client.c:603 Reserved new io event: 0x7fdf13cd78
[D] 1722249238.788913 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, Server hello (2):
[D] 1722249238.789834 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
[D] 1722249238.789881 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, Certificate (11):
[D] 1722249238.794839 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, CERT verify (15):
[D] 1722249238.795665 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, Finished (20):
[D] 1722249238.796016 https_client.c:207 B654: * TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
[D] 1722249238.796261 https_client.c:207 B654: * TLSv1.3 (OUT), TLS handshake, Finished (20):
[D] 1722249238.797149 https_client.c:207 B654: > POST /dns-query HTTP/2
[D] 1722249238.797155 https_client.c:207 B654: > Host: dns.google
[D] 1722249238.797159 https_client.c:207 B654: > User-Agent: https_dns_proxy/0.3
[D] 1722249238.797163 https_client.c:207 B654: > Accept: application/dns-message
[D] 1722249238.797166 https_client.c:207 B654: > Content-Type: application/dns-message
[D] 1722249238.797170 https_client.c:207 B654: > Content-Length: 68
[D] 1722249238.800404 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
[D] 1722249238.800654 https_client.c:207 B654: * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
[D] 1722249239.306673 https_client.c:207 B654: < HTTP/2 200
[D] 1722249239.306725 https_client.c:207 B654: < x-content-type-options: nosniff
[D] 1722249239.306740 https_client.c:207 B654: < strict-transport-security: max-age=31536000; includeSubDomains; preload
[D] 1722249239.306749 https_client.c:207 B654: < date: Mon, 29 Jul 2024 10:34:41 GMT
[D] 1722249239.306757 https_client.c:207 B654: < expires: Mon, 29 Jul 2024 10:34:41 GMT
[D] 1722249239.306791 https_client.c:207 B654: < cache-control: private, max-age=60
[... 134 lines cut ...]
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'
	option procd_trigger_wan6 '0'

config https-dns-proxy
	option bootstrap_dns '9.9.9.9,149.112.112.112'
	option resolver_url 'https://dns.google/dns-query'
	option listen_addr '127.0.0.1'
	option listen_port '5053'
	option user 'nobody'
	option group 'nogroup'
	option verbosity '3'
	option logfile '/tmp/dns-google.log'

config https-dns-proxy
	option bootstrap_dns '8.8.8.8,8.8.4.4'
	option resolver_url 'https://dns.quad9.net/dns-query'
	option listen_addr '127.0.0.1'
	option listen_port '5054'
	option user 'nobody'
	option group 'nogroup'
	option verbosity '3'
	option logfile '/tmp/dns-quad.log'
```

AcidSlide, 2024-07-29:
I was able to reproduce this on an x86 and also on a ramips/mt7621 device (NewfiD2)

Screenshot below is from the x86.. the process got stuck at 100% CPU (on one core) forever until a manual restart to https-dns-proxy is done.  But the issue comes back after a couple of mins.

[image]

stangri, 2024-07-29:
Thank you for the report and investigation @AcidSlide.
Tagging curl maintainer @krant and upstream maintainers @aarond10 @baranyaib90.

AcidSlide, 2024-07-29:
> have you tried forcing HTTP/1.1 use? Does it solve the problem? Does the issue persists if you disable per-instance logging?
> PPS. Can you try curl built against mbedTLS as well?

@stangri I actually enabled the verbose logging per instance to check if I can determine the cause.. but can't find any reason based on the logs.

I'll try testing with HTTP/1.1 and also try to build with a mbedTLS (will test separately) and give feedback once I'm able to do tests.

By the way, is there a config in https-dns-proxy to force HTTP/1.1?

AcidSlide, 2024-07-29:
Test Update:  Still the same issue even if HTTP/1.1 is forced (logging disabled)

Using curl 8.9.0 (with openssl)
```
# curl --version
curl 8.9.0 (x86_64-openwrt-linux-gnu) libcurl/8.9.0 OpenSSL/3.0.14 nghttp2/1.62.1
Release-Date: 2024-07-24
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe
```

[image]

[image]

Matthaiks, 2024-07-30:
curl 8.9.0 with mbedTLS also triggers this issue.
```
curl 8.9.0 (mipsel-openwrt-linux-gnu) libcurl/8.9.0 mbedTLS/3.6.0 nghttp2/1.62.1
Release-Date: 2024-07-24
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
```

AcidSlide, 2024-07-30:
> curl 8.9.0 with mbedTLS also triggers this issue.

Yes.. just tested it also.. can confirm that curl 8.9.0 with mbedTLS also triggering this issue

```
# curl --version
curl 8.9.0 (x86_64-openwrt-linux-gnu) libcurl/8.9.0 mbedTLS/3.6.0 nghttp2/1.62.1
Release-Date: 2024-07-24
Protocols: file ftp ftps http https ipfs ipns mqtt
Features: alt-svc HSTS HTTP2 HTTPS-proxy IPv6 Largefile SSL threadsafe UnixSockets
```

[image]

[image]

BKPepe, 2024-07-30:
I think, we should revert the update, for now.

closed 2024-07-31 by BKPepe

AcidSlide, 2024-08-03:
Confirming `curl 8.9.1` fixed the issue.