openwrt/packages #24699: xz: build broken for mips64/octeon

bluecmd, 2024-07-30:
# Issue template (remove lines from top till here)

Maintainer: @krant 
Environment: mips64_octeonplus

Description:

It seems that liblzma is now failing to build for Octeon. This worked a few days ago, so might be related to [c611da23d8fc62b9c7c78694f45b70c3df069bc5](https://git.openwrt.org/?p=feed/packages.git;a=commit;h=c611da23d8fc62b9c7c78694f45b70c3df069bc5) maybe. A build I made just before that commit works.

See http://downloads.openwrt.org/snapshots/faillogs/mips64_octeonplus/packages/xz/compile.txt

```
checking whether the mips64-openwrt-linux-musl-gcc linker (mips64-openwrt-linux-musl-ld.bfd -m elf) supports shared libraries... /builder/shared-workdir/build/sdk/staging_dir/toolchain-mips64_octeonplus_64_gcc-13.3.0_musl/bin/mips64-openwrt-linux-musl-ld.bfd: unrecognised emulation mode: elf
Supported emulations: elf64btsmip elf32btsmipn32 elf32ltsmipn32 elf32btsmip elf32ltsmip elf64ltsmip
no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... unsupported
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... no
checking whether to build shared libraries... no
checking whether to build static libraries... yes
[..]
cp: cannot stat '/builder/shared-workdir/build/sdk/build_dir/target-mips64_octeonplus_64_musl/xz-5.6.2/ipkg-install/usr/lib/liblzma.so*': No such file or directory
```

It seems that XZ now considers the compiler unable to create shared libraries, compiles, and then the packaging step fails due to no shared files being available to package.

For reference, the successful build done a few days ago (https://github.com/sonix-network/openwrt/actions/runs/10127817414) has the relevant logs like this:

```
checking whether the mips64-openwrt-linux-musl-gcc linker (mips64-openwrt-linux-musl-ld.bfd -m elf64btsmip) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
```

Builds use the snapshot toolchain available at https://downloads.openwrt.org/snapshots/targets/octeon/generic/openwrt-toolchain-octeon-generic_gcc-13.3.0_musl.Linux-x86_64.tar.zst at the time of builds. I have tried a build with current toolchain and feeds set to:

```
src-git packages https://git.openwrt.org/feed/packages.git^eb05355e5cc2d8cca9ce2f4025482704f4be846f
```

and that works, which makes me believe the commit to update xz is the culprit.

bluecmd, 2024-07-30:
Hah! Look at that, https://github.com/tukaani-project/xz/releases says:

> NOTE: 5.2.13, 5.4.7, and 5.6.2 have a build system issue that prevents building of shared libraries on some systems like mips64. See the patch itself for details. The same patch applies to 5.2.13, 5.4.7, and 5.6.2.

The patch is https://github.com/tukaani-project/xz/releases/download/v5.6.2/xz-5213-547-562-libtool.patch

krant, 2024-08-01:
Hey @bluecmd , thanks for the bug-report and investigation, the PR with the fix has been pushed.

closed 2024-08-01 by 1715173329