openwrt/packages #24712: tailscale: missing from 23.05.4 (aarch64_cortex-a53)

tiagoad, 2024-08-01:
Maintainer: @mochaaP 
Environment: 23.05.4 - aarch64_cortex-a53

Description:
The `tailscale` package is missing from `aarch64_cortex-a53` on all 23.05 releases.

There seems to be a build error (https://downloads.openwrt.org/releases/faillogs-23.05/aarch64_cortex-a53/packages/tailscale/compile.txt)

```
Finding targets
verifying github.com/mattn/go-runewidth@v0.0.14/go.mod: checksum mismatch
	downloaded: h1:G7mAYYxgmS0lVkHyy2hEOLQCFB0DlQFTMLWggykrydY=
	go.sum:     h1:Jdepj2loyihRzMpdS35Xk/zdY8IAYHsh153qUoGf23w=

SECURITY ERROR
This download does NOT match an earlier download recorded in go.sum.
The bits may have been replaced on the origin server, or an attacker may
have intercepted the download attempt.

For more information, see 'go help module-auth'.
```

mochaaP, 2024-08-02:
cc @1715173329

1715173329, 2024-08-02:
#24570 I don't really want to repeat it again and again.

1715173329, 2024-08-05:
available now

closed 2024-08-05 by 1715173329