openwrt/packages #24735: uhubctl: should built with __gnu_linux__

bluecmd, 2024-08-03:
Maintainer: @snh 

Description:

Since version 2.5.0 uhubctl supports doing port power operations using sysfs. This is a better way to control the ports as it disconnects devices associated with the port instead of leaving the hanging, as is the case in the old way. This is available in Linux 6.0 and uhubctl falls back to the old process if it can't find the new sysfs files so it should be safe to enable regardless of kernel version.

However, this feature is gated by a define `__gnu_linux__` which I have manually set using `make [..] CPPFLAGS='-D__gnu_linux__'` when building manually to test.

When built correctly `uhubctl --help` presents a sysfs related option:
```
 ./uhubctl --help | grep sysfs
--nosysfs,  -S - do not use the Linux sysfs port disable interface.
```

NOTE: Despite the define being named `__gnu_linux__` it works fine with musl as well. That is what I've been compiling it with.

mvp, 2024-09-01:
I have released uhubctl 2.6.0, which should fix this issue in https://github.com/mvp/uhubctl/commit/169842ae - it added checking for `__linux__` too, which should be present on any Linux platform.

bluecmd, 2024-09-01:
Great! Thanks @mvp ! I'll close this issue then

closed 2024-09-01 by bluecmd