openwrt/packages #24744: px5g-mbedtls only 512 bit key

reinerotto, 2024-08-06:
openwrt 23.05.4
Maintainer: Jo-Philipp Wich
px5g-mbedtls only generates 512 bit keys. Which is not only a general security issue, but also causes errors in certain SSL-Verifications because of too small key length.
Should be changed to 2048 bit length, needs only two simple edits.

brada4, 2024-08-06:
Worth referring to original discussion - security issue is avoided in at least uhttpd as key parameters are passed via parameter.
https://forum.openwrt.org/t/px5g-to-be-improved/206284

reinerotto, 2024-08-06:
Note your special scenario "... in at least uhttpd ..." When using px5g-standalone (which in fact is linked to px5g-mbedtls) only 512bit key generated,. Which i.e. kills nginx, when used directly. 
But I am wondering about not only yours hesitation here: openwrt is so keen about security ...

reinerotto, 2024-08-06:
Have not noticed any.

BKPepe, 2024-10-02:
Package is located in OpenWrt main repository, I dont have permissions to transfer the issue there.
Reference: https://github.com/openwrt/openwrt/tree/main/package/utils/px5g-mbedtls

Your issue should be reported there.

closed 2024-10-02 by BKPepe