openwrt/packages #24747: OLSRD: "Bad Configuration" when selecting "6and4" for IPVersion

LipkeGu, 2024-08-06:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
root@FF-MH-BBD-305:~# /etc/init.d/olsrd stop
root@FF-MH-BBD-305:~# /etc/init.d/olsrd6 stop
Command failed: Not found
root@FF-MH-BBD-305:~# /usr/sbin/olsrd -f "/var/etc/olsrd.conf" -nofork

 *** olsr.org - pre-0.9.9-git_0000000-hash_91d5ae30edfe8ee23769434a3154dcd7 ***
 http://www.olsr.org

Parsing file: "/var/etc/olsrd.conf"
Debug level: 0
Noint set to 1
FIBMetric: flat
TC redundancy 2
NAT threshold 0.75
LQ Algorithm: etx_ff
Smart gateway system: enabled
Smart gateway threshold: 50
Pollrate 0.03
RtTable: 111
RtTableDefault: 112
RtTableTunnel: 113
RtTableTunnelPriority: 100000
RtTableDefaultOlsrPriority: 20000
Link quality level 2
OlsrPort: 698
Willingness: 3
Smart gateway uplink: ipv4
Smart gateway speed: 2500000 uplink/100000 downlink kbit/s
Config line 21: syntax error
Error while reading config file /var/etc/olsrd.conf!
Terminated
root@FF-MH-BBD-305:~#
root@FF-MH-BBD-305:~# cat /var/etc/olsrd.conf

DebugLevel 0
AllowNoInt yes
FIBMetric "flat"
TcRedundancy 2
NatThreshold 0.75
LinkQualityAlgorithm "etx_ff"
SmartGateway yes
SmartGatewayThreshold 50
Pollrate 0.025
RtTable 111
RtTableDefault 112
RtTableTunnel 113
RtTableTunnelPriority 100000
RtTableDefaultOlsrPriority 20000
LinkQualityLevel 2
OlsrPort 698
Willingness 3
SmartGatewayUplink "ipv4"
SmartGatewaySpeed 2500000 100000
IpVersion "6and4" <----------------------------------------------------------- error is here!

Hna4
{
                10.248.3.128 255.255.255.224
}
.....
LoadPlugin "olsrd_txtinfo.so.1.1"
```

LipkeGu, 2024-08-06:
selecting "ipv4" makes olsr happy

GeorgeSapkin, 2025-11-15:
Is this relevant to any recent version of OpenWrt?