openwrt/packages #24761: Error Compiling nginx-njs Module Due to Missing libxml2/libxslt Libraries

QAQBytes, 2024-08-08:
**Package Name**: feeds/packages/nginx

**Maintainer**: @Heil @Ansuel

**Environment**:
                    OS: Ubuntu20.04
                    Architecture: x86_64
                    OpenWrt SDK Version: openwrt-sdk-23.05.0-x86-64_gcc-12.3.0_musl
                    nginx-version:1.25.0
                    nginx-njs version:Latest

**Description**:
```
checking` for execvpe ... found
 + ngx_http_lua_module was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-dav-ext-module
 + ngx_http_dav_ext_module was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-ubus-module
 + ngx_http_ubus_module was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-headers-more
 + ngx_http_headers_more_filter_module was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-brotli
 + ngx_brotli was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-rtmp
 + ngx_rtmp_module was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-ts
 + ngx_http_ts_module was configured
adding module in /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/nginx-njs/nginx
 enabled webcrypto module
 enabled xml module
 enabled zlib module
 + ngx_js_module was configured
checking for PCRE library ... found
checking for PCRE JIT support ... found
checking for OpenSSL library ... found
checking for zlib library ... found
checking for libxslt ... not found
checking for libxslt in /usr/local/ ... not found
checking for libxslt in /usr/pkg/ ... not found
checking for libxslt in /opt/local/ ... not found

./configure: error: the HTTP XSLT module requires the libxml2/libxslt
libraries. You can either do not enable the module or install the libraries.

make[2]: *** [Makefile:685: /home/qaq/openwrt-sdk/build_dir/target-x86_64_musl/nginx-ssl/nginx-1.25.0/.configured_18069b4868d2c2bb49ab28413e96efc5] Error 1
make[2]: Leaving directory '/home/qaq/openwrt-sdk/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/ssl/compile#6.31#3.66#8.70
    ERROR: package/feeds/packages/nginx failed to build (build variant: ssl).
make[1]: *** [package/Makefile:120: package/feeds/packages/nginx/compile] Error 1
make[1]: Leaving directory '/home/qaq/openwrt-sdk'
make: *** [/home/qaq/openwrt-sdk/include/toplevel.mk:225：package/nginx/compile] 错误 2
qaq@debian:~/openwrt-sdk$ make package/nginx/compile V=99cd /
^Cmake[2]: *** [include/scan.mk:103: /home/qaq/openwrt-sdk/tmp/info/.files-packageinfo.stamp] Interrupt
make[1]: *** [/home/qaq/openwrt-sdk/include/toplevel.mk:81: prepare-tmpinfo] Interrupt
make: *** [/home/qaq/openwrt-sdk/include/toplevel.mk:223: package/nginx/compile] 中断

qaq@debian:~/openwrt-sdk$ cd ../
qaq@debian:~$ ls
```
**Steps to Reproduce:**
```
# Extract the package
tar -xf openwrt-sdk-23.05.0-x86-64_gcc-12.3.0_musl.Linux-x86_64.tar.xz

# Rename the extracted directory
mv openwrt-sdk-23.05.0-x86-64_gcc-12.3.0_musl.Linux-x86_64 openwrt-sdk

# Navigate to the openwrt-sdk directory
cd openwrt-sdk

# Update package feeds
./scripts/feeds update packages

# Install all packages
./scripts/feeds install -a -p packages

# Modify nginx Makefile (edit the file as needed) 
#openwrt-sdk/feeds/packages/net/nginx/Makefile

# Modify nginx Config_ssl.in (edit the file as needed)
#openwrt-sdk/feeds/packages/net/nginx/Config_ssl.in

# Compile nginx package with verbose output
make package/nginx/compile V=99
```
[https://github.com/QAQBytes/openwrt-nginx/blob/main/Makefile](url)
[https://github.com/QAQBytes/openwrt-nginx/blob/main/Config_ssl.in](url)

I attempted to modify njs configuration to use libxml2. When running make package/nginx/compile V=99 again, I encountered the following error:

`./configure: error: cannot detect system byte ordering`

I would greatly appreciate any assistance from open-source developers.

Thank you very much

egc112, 2024-09-17:
Try with installing libxslt on your Ubuntu system
`sudo apt-get install libxslt-dev`

P.S. you could also request assistance in the forum

Ansuel, 2024-10-03:
For the future... PLEASE make it clear that the issue is from a downstream change... The NJS library is not supported aka this is a non issue...

Aside from this... https://github.com/openwrt/packages/pull/25082 You are welcome. Now it will be supported.

closed 2024-10-03 by Ansuel