openwrt/packages #24783: sockd Segmentation fault

barl0g, 2024-08-13:
```
root@OpenWrt:~# opkg install sockd
Installing sockd (1.4.1-4) to root...
Downloading http://downloads.openwrt.org/releases/19.07.10/packages/mips_24kc/base/sockd_1.4.1-4_mips_24kc.ipk
Configuring sockd.
root@OpenWrt:~# sockd
Segmentation fault
```

```
# cat /etc/os-release 
NAME="OpenWrt"
VERSION="19.07.10"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.10"
VERSION_ID="19.07.10"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11427-9ce6aa9d8d"
OPENWRT_BOARD="ar71xx/mikrotik"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.10 r11427-9ce6aa9d8d"
```

brada4, 2024-08-15:
You need to repeat with more current release, like 23.04.5 and IF there is a problem fill a report in ..packages repository

barl0g, 2024-08-15:
There is no newer release for my device. What should I do?

brada4, 2024-08-15:
`ubus call system board` - lets confirm it.
Buy a new device if EOL is the case.

barl0g, 2024-08-16:
> `ubus call system board` - lets confirm it. Buy a new device if EOL is the case.
This device: https://openwrt.org/toh/mikrotik/rb941-2nd

The device has not reached EOL. Is there a possibility to backport dante for this release?

brada4, 2024-08-16:
32MB devices are discontinued. There is no essence in your bug report as for a bug in supported releases. You can get some socks proxy out of OEM fw.

esaaprillia, 2024-08-16:
@barl0g 

You can try updating the sockd version and rebuilding it

BKPepe, 2024-08-18:
OpenWrt 19.07 is EoL as said above in comments, you should use supported verison on your device or you should buy a new device. Nothing what we can do here, sorry.

closed 2024-08-18 by BKPepe