openwrt/packages #24856: [avahi-daemon] repead turn on and turn off Wifi Radios, oom happened.

lisir-1024, 2024-08-21:
Describe the bug
Every minute the test turns Wi-Fi Radios on or off, the OOM will appear about 8 hours later. 
the repeat log as follow:

	行   413: Aug 17 15:04:08 avahi-daemon[4039]: Joining mDNS multicast group on interface home-ap-50.IPv6 with address fe80::8278:71ff:fe12:f607.
	行   414: Aug 17 15:04:08 avahi-daemon[4039]: New relevant interface home-ap-50.IPv6 for mDNS.
	行   415: Aug 17 15:04:08 avahi-daemon[4039]: Registering new address record for fe80::8278:71ff:fe12:f607 on home-ap-50.*.
	行  1077: Aug 17 15:04:12 avahi-daemon[4039]: Joining mDNS multicast group on interface home-ap-60.IPv6 with address fe80::2078:71ff:fe12:f608.
	行  1078: Aug 17 15:04:12 avahi-daemon[4039]: New relevant interface home-ap-60.IPv6 for mDNS.
	行  1079: Aug 17 15:04:12 avahi-daemon[4039]: Registering new address record for fe80::2078:71ff:fe12:f608 on home-ap-60.*.
	行  1644: Aug 17 15:04:15 avahi-daemon[4039]: Joining mDNS multicast group on interface home-ap-24.IPv6 with address fe80::8278:71ff:fe12:f606.
	行  1645: Aug 17 15:04:15 avahi-daemon[4039]: New relevant interface home-ap-24.IPv6 for mDNS.
	行  1646: Aug 17 15:04:15 avahi-daemon[4039]: Registering new address record for fe80::8278:71ff:fe12:f606 on home-ap-24.*.
	行  3676: Aug 17 15:07:33 avahi-daemon[4039]: Interface home-ap-50.IPv6 no longer relevant for mDNS.
	行  3677: Aug 17 15:07:33 avahi-daemon[4039]: Leaving mDNS multicast group on interface home-ap-50.IPv6 with address fe80::8278:71ff:fe12:f607.
	行  3680: Aug 17 15:07:33 avahi-daemon[4039]: Withdrawing address record for fe80::8278:71ff:fe12:f607 on home-ap-50.
	行  3774: Aug 17 15:07:35 avahi-daemon[4039]: Interface home-ap-24.IPv6 no longer relevant for mDNS.
	行  3775: Aug 17 15:07:35 avahi-daemon[4039]: Leaving mDNS multicast group on interface home-ap-24.IPv6 with address fe80::8278:71ff:fe12:f606.
	行  3778: Aug 17 15:07:35 avahi-daemon[4039]: Withdrawing address record for fe80::8278:71ff:fe12:f606 on home-ap-24.
	行  3933: Aug 17 15:07:38 avahi-daemon[4039]: Interface home-ap-60.IPv6 no longer relevant for mDNS.
	行  3934: Aug 17 15:07:38 avahi-daemon[4039]: Leaving mDNS multicast group on interface home-ap-60.IPv6 with address fe80::2078:71ff:fe12:f608.
	行  3950: Aug 17 15:07:38 avahi-daemon[4039]: Withdrawing address record for fe80::2078:71ff:fe12:f608 on home-ap-60.
	行  8604: Aug 17 15:11:20 avahi-daemon[4039]: Joining mDNS multicast group on interface home-ap-50.IPv6 with address fe80::8278:71ff:fe12:f607.
	行  8605: Aug 17 15:11:20 avahi-daemon[4039]: New relevant interface home-ap-50.IPv6 for mDNS.
	行  8606: Aug 17 15:11:20 avahi-daemon[4039]: Registering new address record for fe80::8278:71ff:fe12:f607 on home-ap-50.*.
	行  9236: Aug 17 15:11:23 avahi-daemon[4039]: Joining mDNS multicast group on interface home-ap-60.IPv6 with address fe80::2078:71ff:fe12:f608.
	行  9237: Aug 17 15:11:23 avahi-daemon[4039]: New relevant interface home-ap-60.IPv6 for mDNS.
	行  9238: Aug 17 15:11:23 avahi-daemon[4039]: Registering new address record for fe80::2078:71ff:fe12:f608 on home-ap-60.*.
	行  9277: Aug 17 15:11:24 kernel: [104237.061482] mmap: avahi-daemon (4039): VmData 4198400 exceed data ulimit 4194304. Update limits or use boot option ignore_rlimit_data.

After each log: "Withdrawing address record for xxx,",  there are 17 consecutive memory malloc/calloc has not been released in each home-ap-xx. This is the information I can provide
[image]

lisir-1024, 2024-08-21:
[image]
[image]

lisir-1024, 2024-08-21:
[image]
[image]

lisir-1024, 2024-08-21:
https://github.com/avahi/avahi/issues/633

brada4, 2024-08-23:
Software environment like Openwrt architecture, version, diffconfig if any?
Please post text output as  formatted code or attach as txt file.
No memory info like from ps or gdb or /proc seen either.

lisir-1024, 2024-09-04:
close in https://github.com/avahi/avahi/issues/633

closed 2024-09-04 by lisir-1024